pascallanger/DIY-Multiprotocol-TX-Module #942: MT99XX - 17 Sub DRAGON - 6 broken in v1.3.3.33

jakenl, 2024-02-07:
The Eachine Mini Cessna can no longer be connected with v1.3.3.33 (TX: Radiomaster Zorro, EdgeTX 2.9.3)

When I flash in v1.3.3.20 or use the original TX, the plane binds instantly.

Issue can be repeated:
v1.3.3.33 no bind
Flash v1.3.3.20 bind
Flash v1.3.3.33 no bind

pascallanger, 2024-02-08:
This is strange... When I compare the code between 1.3.3.20 and 1.3.3.33 there is no difference on MT99XX.
Have you tried the latest test build? (https://downloads.multi-module.org/latest-test/)
Have you tested other models with v1.3.3.33? What's the module status when running v1.3.3.33?

jakenl, 2024-02-08:
I have now loaded 1.3.3.47 and the plane connects at once! The text says, don't use for flying, so I didn't consider it earlier.

No, I have no other models using that RX. All but one use the Spektrum RX, the remaining one use V761 and they all worked properly.
The model status with 1.3.3.33 was V1.3.3.33 AETR 

and when I click bind: 
V1.3.3.33 Bind... This is only valid for 1-2 seconds and then it goes back to normal

When I disable telemetry, 2 positions down in the list, than the status mentions: 'no telemetry' and the binding + sound keeps going and going without any binding to the plane. I have to manually cancel.

Model status with V1.3.3.47:
No telemetry: status is 'No telemetry'
The normal model status with 1.3.3.47 is V1.3.3.47 AETR 

I flashed back and forth a couple times, but still 33 is bad and 47 is good.

pascallanger, 2024-02-08:
Ok good that it works
You can fly with it as I haven't touched the core for a long time.