pascallanger/DIY-Multiprotocol-TX-Module #944: SG F22 protocol

rdba2k, 2024-02-11:
Hi, Pascal:

Refer to https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page1657#post52092565

I have a working code named as "REALACC_nrf24l01.ino" attached in that post, because I do not know how to add a new protocol to your code base yet. It binds and move the control surfaces as expected. I also compare the xn297 dump from the original TX and the 4-in-1 module, the flags for the corresponding flight mode match.

Thanks.

rdba2k, 2024-02-14:
I have updated REALACC_nrf24l01.ino in 

https://www.rcgroups.com/forums/showpost.php?p=52092565&postcount=24848

I test indoor the vertical mode with TX range check on and outdoors for all modes, it flied perfectly without a glitch. 

The string "REALACC" could be replaced by "SGF22" in the file for "SGF22" protocol. I have asked if anyone can provide his/her TX's dump and make the code more universal.

https://www.rcgroups.com/forums/showthread.php?4187597-SG-F22-RTF-4-Ch-Vertical-Stunt-Plane-Flight-Test-Review/page4#post52114987

Thanks.

phildart007, 2024-02-14:
Hi,
I am really interested. I hope this new protocol will be integrated. I don't have much development experience. I consult the project docs to learn. Super jog

rdba2k, 2024-02-15:
Hi, Pascal:

Could you pls share your modified code? I do not know where to download.

packet[9] is for photo, packet[8] is for both light and flight mode/roll.

10240us is something I guessed. I agree with your correction.

The marking on the TX is actually misleading and the manual is confusing. After watching someone's youtube, pushing up throttle trim is for LED light, pressing down the throttle trim is for photo. I made some modification to my code, but I do not know how to plug the change you made. Attached is my modified code (can not be compiled yet because I do not know the variable "hop"). The main change from my original code is

#define SGF22_FLAG_LIGHT      0x04
#define SGF22_FLAG_VERTICAL   0xC0
#define SGF22_FLAG_6G         0x40
#define SGF22_FLAG_3D         0x00
#define SGF22_FLAG_ROLL       0x08

#define SGF22_FLAG_PHOTO      0x40

and

    if(Channel_data[CH5]<CHANNEL_MIN_COMMAND)
      flight_mode = SGF22_FLAG_3D;  // mode 2 - 3d
    else if(Channel_data[CH5]>CHANNEL_MAX_COMMAND)
      flight_mode = SGF22_FLAG_VERTICAL;  // mode 0 - vertical
    else
      flight_mode = SGF22_FLAG_6G;  // mode 1 - 6g
    packet[8] = flight_mode | GET_FLAG(CH6_SW, SGF22_FLAG_ROLL) | GET_FLAG(CH7_SW, SGF22_FLAG_LIGHT); 
    packet[9] = GET_FLAG(CH8_SW, SGF22_FLAG_PHOTO);  // press the throttle trim, possible for photo

waiting for your code then I can do another test.

Thanks.
[REALACC_nrf24l01.ino.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14299990/REALACC_nrf24l01.ino.txt)

pascallanger, 2024-02-15:
If you download the full repo you'll get all the modified files needed, the main file is here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/SGF22_nrf24l01.ino
The code is now too big for the automated build...

Still on your dump in the file note.txt,  there is a 0x04 flag on packet[9]:
```
all mid sticks + min throttle + mode 0
1B 42 1F 61 04 7F 80 7A 04 **04** 42 10
left aileron
1B 66 1F 61 04 7F 80 00 04 **04** 42 10
right aileron
1B 42 1F 61 04 7F 8D FF 04 **04** 42 10
elevator down
1B BA 1F 61 03 7F FF 7A 04 **00** 42 10
```

rdba2k, 2024-02-15:
Hi, Pascal:

I test indoor the vertical mode with range check on. It flies perfect. I will test outdoor all modes later. So far so good. The LED control is also fixed now.

Thank you!.

rdba2k, 2024-02-15:
BTW, this line need be fixed, adding "packet[9] ="

packet[9] = GET_FLAG(CH8_SW, SGF22_FLAG_PHOTO);				// press down throttle trim for photo

Anyway, my SG F22 does not come with camera, so no test for this.

rdba2k, 2024-02-15:
The pic. of TX is here,

https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page1657

The so-called "throttle trim" actually has nothing to do with the throttle, It gets such a name because it is asides of the throttle. It has 3 ways to activate various functions.

push up - for light, flag in packet[8]
press down - for photo, flag in packet[9]
push down - do not know what is the usage for it.

> 
> Still confused since throttle trim is 0x04 in [8] and 0x40 in [9] where I see in your notes 0x04 in [9]. But let's forget if you are happy with the current flags...

pascallanger, 2024-02-16:
Have you tried to change the value of the tx address to see if it would work? Start by just incrementing rx_tx_addr[3] and see if it binds and controls.

rdba2k, 2024-02-16:
None of below works.

  XN297_SetTXAddr((uint8_t*)"\xC7\x95\x3C\xBB\xA5", 5);
  // XN297_SetTXAddr((uint8_t*)"\xC7\x95\x3C\xBB\xA6", 5); not working

or

    rx_tx_addr[3] = 0x61;
    //rx_tx_addr[3] = 0x62;  not working

Attached is the version which I added video flag, the change is

// packet[8]
...
#define SGF22_FLAG_VIDEO    0x10
//packet[9]
#define SGF22_FLAG_PHOTO		0x40

		packet[8] = SGF22_FLAG_3D						// default
				| GET_FLAG(CH6_SW, SGF22_FLAG_ROLL)		// roll
				| GET_FLAG(CH7_SW, SGF22_FLAG_LIGHT)	// push up throttle trim for light
                                | GET_FLAG(CH9_SW, SGF22_FLAG_VIDEO);   // push down throttle trim for video

		packet[9] = GET_FLAG(CH8_SW, SGF22_FLAG_PHOTO);				// press in throttle trim for photo
 
Thanks.

[SGF22_nrf24l01.ino.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14305362/SGF22_nrf24l01.ino.txt)

pascallanger, 2024-02-16:
Can you try to change the init value from 0x26 to 0x02 and see if it has any effect here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/dc1490b9b08f61f48cfbcbefaf855728965c7026/Multiprotocol/SGF22_nrf24l01.ino#L141

Can you try to change rx_tx_addr[3] to 0x62 (as you've already done) but also do this:
 - Change this line https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/dc1490b9b08f61f48cfbcbefaf855728965c7026/Multiprotocol/SGF22_nrf24l01.ino#L52 to XN297_RFChannel(option);
 - Replace on this line  https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/dc1490b9b08f61f48cfbcbefaf855728965c7026/Multiprotocol/Multi_Protos.ino#L449 OPTION_NONE by OPTION_OPTION
 - Compile, upload and you should now have a new field named option on the radio display. Power up the plane, bind it as usual and then slowly increment option. At one point in time you should get control of the plane, write down the option value and continue to scan until you get the 4 RF frequencies which are currently in use.
If it works just repeat the same with different values of rx_tx_addr[2] and [3], interresting tests could be [2]=0 and [3]=0 then [2]=0 and [3]=1 and so on.

rdba2k, 2024-02-17:
Can you try to change the init value from 0x26 to 0x02 and see if it has any effect here:

Not binding.

rdba2k, 2024-02-18:
It looks like the hopping frequencies depend on [2][3]%5. then select the one from the first group.

rx: RF
[2]=0x00 [3]=0x00 : 12 42 27 57
[2]=0x00 [3]=0x01 : 15 45 30 61
[2]=0x00 [3]=0x02 : 18 49 33 65
[2]=0x00 [3]=0x03 : 21 52 36 68
[2]=0x00 [3]=0x04 : 24 55 39 71
[2]=0x00 [3]=0x05 : 12 42 27 57

[2]=0x01 [3]=0x00 : 15 45 30 61
[2]=0x02 [3]=0x00 : 18 49 33 65

[2]=0x1F [3]=0x61 : 21 52 36 68
[2]=0x1F [3]=0x62 : 24 55 39 71
[2]=0x1F [3]=0x63 : 12 42 27 57

rdba2k, 2024-02-18:
Hi, Pascal:

Attached is the modified version. I test several re-start from the beginning of both airplane and TX, All works.

The main change is

//#define FORCE_SGF22_ORIGINAL_ID
...
	#ifdef FORCE_SGF22_ORIGINAL_ID
		rx_tx_addr[2] = 0x1F;
		rx_tx_addr[3] = 0x61;
		memcpy(hopping_frequency,"\x15\x34\x24\x44", SGF22_RF_NUM_CHANNELS);    //Original dump=>21=0x15,52=0x34,36=0x24,68=0x44
  #else
     rx_tx_addr[2] = RX_num & 0x0F + 0x1F; 
     rx_tx_addr[3] = random(0xfefefefe)&0xFF; 
     uint16_t temp = 0;
     temp =  ( (uint16_t) rx_tx_addr[2] << 8 ) | rx_tx_addr[3];
     switch ( temp % 5 )
     {
        case 0 :
          memcpy(hopping_frequency, "\x0C\x2A\x1B\x39", SGF22_RF_NUM_CHANNELS);  //12 42 27 57
          break;
        case 1 :
          memcpy(hopping_frequency, "\x0F\x2D\x1E\x3D", SGF22_RF_NUM_CHANNELS);  //15 45 30 61
          break;
        case 2 :
          memcpy(hopping_frequency, "\x12\x31\x21\x41", SGF22_RF_NUM_CHANNELS);  //18 49 33 65
          break;
        case 3 :
          memcpy(hopping_frequency, "\x15\x34\x24\x44", SGF22_RF_NUM_CHANNELS);  //21 52 36 68
          break;
        case 4 :
          memcpy(hopping_frequency, "\x18\x37\x27\x47", SGF22_RF_NUM_CHANNELS);  //24 55 39 71
          break;            
     }   
	#endif

[SGF22_nrf24l01.ino.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14324125/SGF22_nrf24l01.ino.txt)

pascallanger, 2024-02-19:
I've pushed the changes. Can you test?

Next, I'm thinking they have more possible RF frequencies by changing this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/5f9ac82ed1f178e4e6caccdb31f81f6645dd3abb/Multiprotocol/SGF22_nrf24l01.ino#L165

Instead of 0x26, try 0x02 and see which RF frequencies you are getting for the 5 first IDs. You can then increment the value by 4 and retry, at least the "sequence" is always 2 + x*4.

Also the "sequence" goes from 02 to 7A by increments of 4 and sometimes it has a flag 0x80 therefore from 82 to FA but I can't tell from the dumps when the switch happens, is this something you can look into?

rdba2k, 2024-02-19:
Changing "packet_sent = packet_count = 0x26"  to "0x02" would not bind. Can you change this to OPTION value so I can go through packet[1]  to see any value can bind.  Either the plane or TX will make a noise when the binding is successful.

I do not see your code randomizes the value of  rx_tx_addr[2] and rx_tx_addr[3].  For example, I did in  my code
     
     rx_tx_addr[2] = RX_num & 0x0F + 0x1F; 
     rx_tx_addr[3] = random(0xfefefefe)&0xFF;

pascallanger, 2024-02-19:
> Changing "packet_sent = packet_count = 0x26" to "0x02" would not bind. Can you change this to OPTION value so I can go through packet[1] to see any value can bind. Either the plane or TX will make a noise when the binding is successful.

Ok but have you tried to scan the RF frequencies as you did before to see if they have changed?

> I do not see your code randomizes the value of rx_tx_addr[2] and rx_tx_addr[3]. For example, I did in my code
> 
> ```
>  rx_tx_addr[2] = RX_num & 0x0F + 0x1F; 
>  rx_tx_addr[3] = random(0xfefefefe)&0xFF; 
> ```

It's not needed. By default, the global id is loaded to rx_tx_addr taking into account RX_num before calling the protocol.

rdba2k, 2024-02-19:
Hi, Pascal:

I test my RX_num from 0 to 4, the code works.

I think to bind, the packet may need match 0x5B0x26...0xAA0x55. Let us wait to see if others can dump their binding packet.

I learn a lot from this process. Thanks again.

pascallanger, 2024-02-20:
Ok let's call this protocol done. Don't hesitate to open issues to work on other protocols (new or enhancement).

closed 2024-02-20 by pascallanger

pascallanger, 2024-03-22:
@rdba2k There is an issue with some IDs. I've been contacted by someone having issues with this protocol. It looks like the RF frequencies are different if [2] or [3] are greater than 0x80. In his case, [2]=E5 and [3]=F2 doesn't work... I've pushed a quick fix [2]&0x7F and [3]&0x7F for now. But it would be great if you can help identify the "upper" table like you did previously.

rdba2k, 2024-04-16:
Hi, Pascal:

Sorry that it takes a while until now I have time to work on SGF22 protocol again. I bought another SG F22 and the address is "2A DE". Could you pls send me the code to change the RF frequency as before. Sorry that I did not keep the old code. Attached is the capture from my second SG F22 TX.
[new 3.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/15003015/new.3.txt)

pascallanger, 2024-04-18:
@rdba2k Here is the post which indicates how to do the frequency scan:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/944#issuecomment-1948052465

Change rx_tx_addr[3] to 0x81 in SGF22_initialize_txid() (as you've already done) but also do this:
 - Change this line https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/dc1490b9b08f61f48cfbcbefaf855728965c7026/Multiprotocol/SGF22_nrf24l01.ino#L52 to XN297_RFChannel(option);
 - Replace on this line  https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/dc1490b9b08f61f48cfbcbefaf855728965c7026/Multiprotocol/Multi_Protos.ino#L449 OPTION_NONE by OPTION_OPTION
 - Compile, upload and you should now have a new field named option on the radio display. Power up the plane, bind it as usual and then slowly increment option. At one point in time you should get control of the plane, write down the option value and continue to scan until you get the 4 RF frequencies which are currently in use.
Just repeat the same with different values of rx_tx_addr[2] and [3], interesting tests could be [2]=0x00 and [3]=0x81 then [2]=0x00 and [3]=0x82 and so on.

rdba2k, 2024-04-19:
The current code has the frequency array of
[2]=0x00 [3]=0x00 : 12 42 27 57
[2]=0x00 [3]=0x01 : 15 45 30 61
[2]=0x00 [3]=0x02 : 18 49 33 65
[2]=0x00 [3]=0x03 : 21 52 36 68
[2]=0x00 [3]=0x04 : 24 55 39 71
[2]=0x00 [3]=0x05 : 12 42 27 57

I test the case of [2][3]
0x00 0x80, 
0x00 0x81, 
0x00 0xB4, 
0x80 0x00, 
0x81 0x00, 
0xB4 0x00, 
the hopping frequencies still fit with the formula of (( (uint16_t) rx_tx_addr[2] << 8 ) | rx_tx_addr[3])%5 in the code.

However, the case of
0x80 0xAA: freq 18 49 33 65
0x81 0xAA: freq 21 52 36 68
the hopping frequencies fit with (  (( (uint16_t) rx_tx_addr[2] << 8 ) | rx_tx_addr[3]) - 1 )%5

Not sure starting at which number the formula need be adjusted.

rdba2k, 2024-04-19:
I test the changing boundary of [2] [3], and I only list the first frequency,

0x80 0x7F: 12
0x80 0x80: 12
0x80 0x81: 15

0x7F 0x80: 12
0x80 0x80: 12
0x81 0x80: 15

0xFE 0xFE: 18
0xFF 0xFF: 24

If the formula is  ( (( (uint16_t) rx_tx_addr[2] << 8 ) | rx_tx_addr[3]) - modifier )%5
It looks like if [2] > 0x7F && [3] > 0x7F, modifier = 1,
otherwise modifier = 0

or the formula is
(( (uint16_t) (rx_tx_addr[2] & 0x7F) << 8 ) | (rx_tx_addr[3] & 0x7F) )%5 ?

rdba2k, 2024-04-23:
Hi, Pascal:

Attached is the modified code, it works with 
[2] <= 0x7F and [3] <= 0x7F, 
[2] > 0x7F and [3] > 0x7F

But things get complicated in other quadrant, I have the case of ([2] [3])
(0x0 0x94) work,
(0x1 0x7F), (0x1, 0x80), (0x1, 0x81) work
(0x7E, 0x80), (0x7E, 0x81) work, (0x7E, 0x82) not work
(0x7F, 0x7F), (0x7F, 0x80) work, (0x7F, 0x81) not work.

The mapping of [2] [3] to the range of 0x7F, 0x7F is already good and simple enough. I will give up now. Thanks.

rdba2k, 2024-04-24:
Hi, Pascal:

Attached is the working version. Main change is

  //rx_tx_addr[2] &=0x7F;
  //rx_tx_addr[3] &=0x7F;
  uint16_t tmp;
  if ( rx_tx_addr[2] > ( 0xFF - rx_tx_addr[3]) ) 
    tmp = 1;
  else 
    tmp = 0;
  uint8_t val = ((( (uint16_t) rx_tx_addr[2] << 8 ) | rx_tx_addr[3]) - tmp )%5;

[SGF22_nrf24l01.ino.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/15084770/SGF22_nrf24l01.ino.txt)

I have verified on those combination of [2] & [3] and they are working.

0x62 0x5B, 0x62 0xAF, 0x7F 0x7F, 0x80 0x80, 0xAA 0x1D, 0xAA 0xB2

pascallanger, 2024-04-24:
I've implemented the change and pushed it on github.
If you could just try a few values just to make sure, thanks.

rdba2k, 2024-04-25:
I test those around the corners and edge of [2][3]
00 00, 00 FF, FF FF, FF EE, FF 00

and those around the slant,
7F 00, 7F 4C, 7F 7F, 7F 80, 7F 9D
80 68, 80 7F, 80 80, 80 9D

they all work.

pascallanger, 2024-04-26:
Thanks for confirming. Hopefully that fully closes this protocol.