pascallanger/DIY-Multiprotocol-TX-Module #946: Protocol KN-WLTOYS vs. heli V950

ivanb123github, 2024-02-15:
Hi, I used the original transmitter for Wltoys NO.V950 in mod 3 with the receiver according to GITHUB https://github.com/donkwt/Arduino_nRF24l01_KN_Receiver?tab=readme-ov-file#arduino_nrf24l01_kn_receiver... I modified it to display the states of the switches - see Annex. Assignment to trim coordinates (Left,Right, X,Y axes) LeXt....RUD ; LeYt....THR; RiXt....AIL; RiYt .... ELE;
TX16s settings: Mode3, internal KN-Wltoys, multi 4 in 1 version 1.3.3.33 AETR. - does not transfer one trim and switch state. Heli model set according to KN-wltoys protocol documentation.
1. Lxt....RUD (trim X right) is not transferred from Tx16s.
2. Wltoys transmitter also has HoverDebugging. status=0, with a faster 3x click it is set to 1 (bit4 buffer[12]). By clicking again, status=0. Tx16 nothing.
Other channels, trims, switches are OK.  
[wltoysRX.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14297709/wltoysRX.zip)
Thank you for the arrangement 
Ivan

pascallanger, 2024-02-15:
When you say bit4 in buffer[12], do you start counting from bit0 or bit1? ie is bit 4 0x08 or 0x10?
Trim rudder  was never needed to fly but I can add it.

ivanb123github, 2024-02-15:
It is 0x08.

pascallanger, 2024-02-15:
The code is updated.
Since the code is now too large, the automated build doesn't work anymore. You need to compile by yourself.

ivanb123github, 2024-02-15:
It works great. Thank you