pascallanger/DIY-Multiprotocol-TX-Module #951: Evaluate Team associated 1/28 scale RTR Transmitter

RickzFPV, 2024-02-23:
I would like to see if i can get some help with getting the protocol for the Team Associated RTR Transmitter.  It is different from the other Associated Transmitters.  I will be ordering a logic analyzer, but it will have to ship from China, and then I will have to learn how to use it.  Maybe someone who is on here already has this transmitter and the equipment with the know how to do this faster than I can.  Any help would be so much appreciated.  

Rick
[image]
[image]
[image]
[image]
[image]

pascallanger, 2024-02-23:
I can guide you to connect a logic analyzer ($10 or so on ebay https://www.ebay.fr/itm/262812296176) to the XN297, basically it's the 3 traces going directly from the XN297 to the controller + ground:
[image]

RickzFPV, 2024-02-26:
I ordered a logic analyzer this morning on ebay gonna take like 10 days to get here before I can do anything with it.  Maybe I can get lucky and someone on here already has the equipment to pull the info needed before I get mine.