pascallanger/DIY-Multiprotocol-TX-Module #952: SG PINECONE FOREST 1604 Car Protocol?

pascallanger, 2024-02-28:
### Discussed in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/discussions/920

<div type='discussions-op-text'>

<sup>Originally posted by **aseitz79** December 18, 2023</sup>
Hi there,

does anybody know if the MPM already contains the protocol to drive the famous SG 1604 models from Pinecone?

I think with the new Radiomaster MT12 (4 in 1 version) a lot of request for suface models support might come up :-)

Thanks in advance,

Alex</div>

pascallanger, 2024-02-28:
Bind
C=42 Enhanced pid=2 S=Y A= 01 03 05 07 09 P(15)= 01 D0 06 00 00 81 00 00 00 69 5F C8 00 00 E8
Packet[0] = packet type, 01 = bind, 08 = normal
Packet[1..5] = normal packets address
Packet[6..8] = 00 unknown
Packet[9] = ESP 00..64..C8 (0..100..200)
Packet[10] = ST_TRIM 00..64..C8 (0..100..200)
Packet[11] = ST_DR 00..64..C8 (0..100..200)
Packet[12] = flags: 0x40 TH.REV, 0x80 ST.REV
Packet[13] = 00 unknown
Packet[14] = sum(Packet[0..13])
RF Channels 45,59,52,67 (42?, 49?, 56?, 64?)
Normal
TX:
C=42 Enhanced pid=1 S=Y A= D0 06 00 00 81 P(15)= 08 64 64 00 00 00 00 00 00 69 5F C8 00 00 60
Packet[0] = packet type, 01 = bind, 08 = normal
Packet[1] = ST  00..64..C8 (0..100..200)
Packet[2] = TH  00..64..C8 (0..100..200)
Packet[3] = CH4 00..64..C8 (0..100..200)
Packet[4] = CH3 00..64..C8 (0..100..200)
Packet[5..8] = 00 unknown
Packet[9] = ESP 00..64..C8 (0..100..200)
Packet[10] = ST_TRIM 00..64..C8 (0..100..200)
Packet[11] = ST_DR 00..64..C8 (0..100..200)
Packet[12] = flags: 0x40 TH.REV, 0x80 ST.REV
Packet[13] = 00 unknown
Packet[14] = sum(Packet[0..13])
RF channel ???
RX:
C=59 Enhanced pid=1 S=Y A= D0 06 00 00 81 P(15)= 10 00 03 FF FF B8 CE 0A 6C 37 FC 31 D7 06 58
C=64 Enhanced pid=1 S=Y A= D0 06 00 00 81 P(15)= 10 00 03 FF FF B8 CF 0A 6C 37 FC 31 D7 8E 58
Packet[0] = packet type, 01 = bind, 08 = normal, 10 = rx
Packet[1..13] = unknown...
Packet[14] = sum(Packet[0..13])
RF channel ???

aseitz79, 2024-04-26:
Any updates on this? When will the protocol be implemented?

aseitz79, 2024-10-07:
Once again in the hope of an answer: Is there any progress here or will this never be implemented?