pascallanger/DIY-Multiprotocol-TX-Module #954: Jumper 4in1 module lights not working

tech5960, 2024-03-13:
Hey all,
The Jumper 4in1 module I have in my Taranis QX7 ACCST transmitter does not work. I cannot bind to my Hubsan H123D Jet quad like I used to be able to. No lights red or green illuminate when I insert the module?

pfeerick, 2024-03-20:
If the lights don't come on... particular the red one... this probably means you have:
1. not turned the MPM on in the radio settings (i.e. External RF => MULTI)
2. the module isn't connected properly (double check the JR micro bay to see there are no bent pins, and that it has clicked in properly)
3. the module isn't working - maybe try reflashing the firmware on it using https://github.com/benlye/flash-multi ? - that may also help idenfity if the issue lies with the transmitter or the module by the time you get here...
4. the module bay isn't working - do you have another module or transmitter to cross check with? Can also check if the second from the bottom (GND) and third from the bottom (VBAT) pins have voltage with a multimeter (but don't short the pins!!)

tech5960, 2024-03-20:
Hi Peter,
Thanks for getting back to me. The model that I built for the purpose of
flying my Hubsan Jet External RF mode is set to MULTI HUBSAN and the
internal RF is off. Status = No telemetry..
The JR pins are all straight so ill try to reload the firmware. I have a
Taranis QX7 ACCST transmitter.
I'll get back to you soon.
Harry Latorre

closed 2024-04-03 by pascallanger