pascallanger/DIY-Multiprotocol-TX-Module #956: Support cloned FLYSKY AFHDS2A protocol

richieg66, 2024-03-16:
Is it possible to add support for cloning the FLYSKY AFHDS2A protocol? I've successfully  cloned Frsky transmitters and that works. It would be great to extend this to use Flysky to enable me to bind  FS-IA6B receivers with multiple transmitters.

Thanks
Richard

pfeerick, 2024-03-20:
Just so you know... "I'm interested" ;)

Rixxxi, 2024-10-21:
I'd be interested in this, too!

pagrey, 2024-10-22:
Try it out, I'd enjoy the feedback. When I have some time I'll update my repo but you can cherry pick the commit, c46d6e2380d31ee255b31eae64500fe7a530f1b3 and it should work fine. Sorry it's not just download and go, I don't manage this project.

Rixxxi, 2024-10-22:
Thanks! When I start compiling something myself, it tends to take ages until I get it running. Don't know if/when I will find the time to give it a try. If I do, I'll let you know!