pascallanger/DIY-Multiprotocol-TX-Module #963: 4 in 1 External Confusion

Slicc98, 2024-04-05:
So I have a Radio Master external 4 in 1 module. My problem is updating it; i have come across three different firmwares available for it:

One marked SFC (land)
One marked AIR (air)
One marked LBT (EU?)

The problem is that I'm using the module for both traxxis cars AND aircraft. The other option not listed is FCC (which is both sfc and air combined), but that option isn't listed. 

Because I wasn't sure, I went looking through the SD card that came with the tx16s... and a firmware IS listed... but it's the lbt!

I purchased the remote and module from the US for use in the US, so why would that be listed? Did he mess something up?

Could I combined the sfc and air firmwares? The firmwares were stored on the sd card  that came with the remote. It was 512Mb. I replaced it with a 128gb card for the remote. I'm not sure which one to pick. 

Help?

pascallanger, 2024-04-05:
For your specific need, you must compile the firmware by yourself and select the protocols you need.
The SD card has nothing to do with the amount of protocols you can load on the module...

Slicc98, 2024-04-05:
> For your specific need, you must compile the firmware by yourself and select the protocols you need.
> The SD card has nothing to do with the amount of protocols you can load on the module...

Are there instructions on how to compile firmware? I'm VERY new to this. As long as there are instructions I'll learn myself. Where are the protocols listed?

pascallanger, 2024-04-06:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md

Slicc98, 2024-04-09:
I got all the way to compiling the firmware. I'm getting error messages on Arduino IDE. All I wanted to do was re-combine the Surface and Air firmware to one package that I can use to flash the external module. It looks like that used to be the standard, but it was separated to save space. 

Perhaps a combined package of surface and air can be readfed in the future? I thought I was doing ok with reading through the whole thing but I have no idea what Arduino IDE even IS.

Slicc98, 2024-04-09:
> I guess you haven't installed the Arduino version 1.8.9 as indicated in the documentation... Compiling is easy but you need to follow the doc at the letter.
> There won't be anymore a combined surface/air build due to space issue. Even for the air, I'll have to start to disable proto soon as it's not going to fit for long...

I have A.IDE 2.3.2 downloaded...do I need _downgrade_ to1.8.9 to do this correctly?

Slicc98, 2024-04-09:
[image]

This is where I am.

pascallanger, 2024-04-09:
Uninstall the arduino 2.xxx and install as 1.8 as instructed. Again follow strictly the doc and it will be smooth.

Slicc98, 2024-04-09:
Same outcome [image]

pascallanger, 2024-04-09:
The version 2 is still at play here... Uninstall everything and delete the Arduino folders in Program files

closed 2025-02-26 by pascallanger