pascallanger/DIY-Multiprotocol-TX-Module #964: new file names are to long

Tiefflieger68, 2024-04-06:
File names in EdgeTX are limited to 28(+3) characters. But the new file names have 31 (eg mm-stm-serial-aetr-air-v1.3.4.0.bin).
If I copy this file at my radio it's not visible in internal file explorer. Of cause, its an easy thing to rename the file, but many users are struggling cause of this (often seen in forum threads).
So, please change the file names to max 28 characters

benlye, 2024-04-06:
That is only true for radios with black and white screens. Radios with colour screens have no trouble with the long file names.

We already have this noted on the [download page](https://downloads.multi-module.org/):

[image]

(I just added Boxer and MT12 to the examples.)

But if you can come up with a file name scheme that includes all the necessary details to differentiate the files and is still meaningful to humans while fitting in 31 characters, feel free to suggest it.

Tiefflieger68, 2024-04-07:
Hello Benlye
thanks for the hints.
My Suggestions:
Yon may short "serial" to "ser" and/or skip "mm-" at the beginning.
Both would be a way

closed 2024-04-07 by Tiefflieger68