pascallanger/DIY-Multiprotocol-TX-Module #965: SLT2 Receiver Binding Issues

Pencilvestr, 2024-04-06:
Opened per your request, continuation from rcgroups post.
https://www.rcgroups.com/forums/showpost.php?p=52330699&postcount=25129

Issue: 
SXC24 Jeep Rubicon with SLT2 receiver will not bind to MT12 with internal 4 in 1 mpm. Using SLT2 protocol. Only knocks the receiver out of bind mode

Hardware:
TX Model: SPMSLT200F
RX Model: SPMXSE2425RX

3v3 USB to serial adapter acquired as requested.

Pencilvestr, 2024-04-07:
Compiled and flashed the serial debug version for for 4 in 1 for the t16s in anticipation. Having trouble getting the serial monitor to output anything however

pascallanger, 2024-04-08:
> Having trouble getting the serial monitor to output anything.

Hummm that's strange since the module does output data right away. Have you used Arduino for the output? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md
Have you tried to compile the code by yourself? This is mandatory for this.

Pencilvestr, 2024-04-08:
Yes I complied the code myself with debug set to serial / only ground protocols. 
I will give it another go and report back. I'll also take a look at what you linked

thanks!

pascallanger, 2024-04-08:
That should work right away with debug serial. The module gives it's ID as soon as you power it on (a model needs to be loaded and the radio fully started ). Try switching RX and TX if it doesn't work for a quick test.

Pencilvestr, 2024-04-10:
Looks like I managed to get it working.
Let me know the next steps please

[image]

jmunderwood, 2024-04-11:
I don't have anything to add here other than to say I am having the same issue with SLT2 receiver that came with an Arrma Granite Grom and Typhon Grom.

pascallanger, 2024-04-12:
@Pencilvestr
Change the address length to 4:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/902419dd3db509c23ec52cd499927ed4712ecac6/Multiprotocol/XN297Dump_nrf24l01.ino#L612C5-L612C21

Change the address to \x7E\xB8\x63\xA9
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/902419dd3db509c23ec52cd499927ed4712ecac6/Multiprotocol/XN297Dump_nrf24l01.ino#L613C36-L613C48

Change the bitrate to XN297DUMP_250K
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/902419dd3db509c23ec52cd499927ed4712ecac6/Multiprotocol/XN297Dump_nrf24l01.ino#L615

Change the packet length to 16
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/902419dd3db509c23ec52cd499927ed4712ecac6/Multiprotocol/XN297Dump_nrf24l01.ino#L616C5-L616C18

Change hopping_frequency_no to 0x50
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/902419dd3db509c23ec52cd499927ed4712ecac6/Multiprotocol/XN297Dump_nrf24l01.ino#L617

Change NRF24L01_WriteReg(NRF24L01_05_RF_CH, option);	//hopping_frequency_no); to NRF24L01_WriteReg(NRF24L01_05_RF_CH, hopping_frequency_no);
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/902419dd3db509c23ec52cd499927ed4712ecac6/Multiprotocol/XN297Dump_nrf24l01.ino#L625

Compile, upload and select the protocol xn297dump/nrf
Make sure your serial console is up and power on the original TX
If all goes well you should get some output, please paste the text content not a picture or upload the output file

Pencilvestr, 2024-04-13:
@pascallanger 
Followed your directions. D/L'd  the latest code with serial monitor enabled and flashed it to the T16s.
Using the following on the t16s
Mode: XN297DU
RF Protocol: NRF
Select RF Channel : 80

[image]

Turned on the Car Transmitter, not getting anything in the serial monitor. Tried a couple times, with moving controls etc
Serial output:

```
Bad frame RX
⸮Multiprotocol version: 1.3.4.7
Module Flash size: 128KB
EEPROM initialized: 0
Protocol selection switch reads as 0
Using bank 0
Module Id: 6550fe30
Init complete
Normal telem 0,1
Protocol selected: 0, sub proto 0, rxnum 0, option 255
TEL_BUF_FULL 25
TEL_BUF_FULL 26
Protocol selected: 63, sub proto 4, rxnum 0, option 80
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=6, Sub=NRF   , Opt=8, FS=0, CHMap=0, rfSw=2
NRF dump, len=16, rf=80, address length=4, bitrate=250K
```

Both SLTv1 and SLTv2 will knock the car out of binding mode without binding to it

Pencilvestr, 2024-04-13:
Messed around with it a little more, able to get some output after pressing the bind button on the t16s once all the other options were selected.

This is just with the car TX turned on not moving any controls. Car RX Off. About 10-15 seconds of output.

```Protocol selected: 63, sub proto 4, rxnum 0, option 82
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=6, Sub=NRF   , Opt=8, FS=0, CHMap=0, rfSw=2
NRF dump, len=16, rf=82, address length=4, bitrate=250K
C: 50 P: F4 71 8D 01 66 E7 7F FF FF FF FF FF FF FF FF FF
C: 50 P: F4 71 8D 01 66 E7 7F FF FF FF FF FF FF FF FF FF
C: 50 P: F4 71 8D 01 66 E7 7F 3E FE D6 7F FD 6D 7F F7 D6
C: 50 P: F4 71 8D 01 66 E7 7E FF FF FF FF F7 FD F9 FB 7D
C: 50 P: F4 71 8D 01 66 E7 7E FF FF DF FF FF FF FB FF B7
C: 50 P: F4 71 8D 01 66 E7 7F FE F7 FF 7F 6B 5A 99 5A CB
C: 50 P: F4 71 8D 01 66 E7 7B BE FF 77 75 F6 DF FF 7E FF
C: 50 P: F4 71 8D 01 66 E7 6F FF 7D D5 7B FB FF DA ED FE
C: 50 P: F4 71 8D 01 66 E7 7B FB FF FF FF FF FF 7B FF D7
C: 50 P: F4 71 8D 01 66 E7 7F EF FF FE BD 7F FB EF FF BB
C: 50 P: F4 71 8D 01 66 E7 7F BF 7F F7 FF F7 FD FF FE FF
C: 50 P: F4 71 8D 01 66 E7 77 FF DF FF FF FF F7 FF EF FD
C: 50 P: F4 71 8D 01 66 E7 7F FF FF DF FD FF FF FF FF 7F
C: 50 P: F4 71 8D 01 66 E7 5F FF FF 7B FF F7 FF FF BF FF
C: 50 P: F4 71 8D 01 66 E7 7F FE FF FF AE FF 7D FB FB F7
C: 50 P: F4 71 8D 01 66 E7 6F DF BF FF FD DB DF FF FF 7E
C: 50 P: F4 71 8D 01 66 E7 7B FF FB 5F FF 5F 7B FF DF E5
C: 50 P: F4 71 8D 01 66 E7 7F FF FF FF FF FF FF FF FF FF
C: 50 P: F4 71 8D 01 66 E7 7F FF FF BD BE FF FF FF F5 FF
C: 50 P: F4 71 8D 01 66 E7 7F FF FF FF FF FF FF FF FF FF
C: 50 P: F4 71 8D 01 66 E7 7F FF FF FF FF FF FF FF FF FF
C: 50 P: F4 71 8D 01 66 E7 7F FF FF FF FF FF FF FF FF FF
C: 50 P: F4 71 8D 01 66 E7 7F FF FF FF FF FF FF FF FF FF
C: 50 P: F4 71 8D 01 66 E7 66 AA A3 DF FF FF FF FF FF FF

```

pascallanger, 2024-04-13:
I don't understand why you have to press the bind button, that has no real effect apart from restarting the protocol...

So the packet content is F4 71 8D 01 with a CRC of 66 E7
F4 71 8D 01 is the address for control packets so now we need to:
 - get the content of the payload and decode it -> that should be easy
 - get the 15 RF frequencies -> not easy

I've updated the code with your original TX ID so download latest code, compile, upload, serial monitor up, power on the original TX:
 - You need to manually scan the RF channel, start at 0 and increment slowly until you get packets (you shouldn't have to validate or hit the bind button but...). The packets should be coming at a constant pace, if yes write down the RF value as ok, if not write as maybe. You should find a total of 15 ok channels. Record your full test and share the text/file. Since CRC is disabled (not checked) that part could be challenging. I will enable the CRC later when I'll have seen the payload length.
 - On a good RF channel, move one control/switch at a time and write down the effect. For example: steering 1st byte move from 00 to FF and byte 5 goes from 00 to 03. Record your full test and share the text/file.

Pencilvestr, 2024-04-13:
Manually scanned each channel and back through them to confirm the results. I also noted certain channel that had repeating patterns of Zeros, which comes out to 15 channels
I also did not have to press the bind button this time. I could select a channel press it then see if I had data coming onto the serial monitor
```
RF Channels Initial

7
10 -- Repeating Zero Pattern
13
14
16 -- Repeating Zero Pattern
17 -- Repeating Zero Pattern
19
20
22 -- Repeating Zero Pattern
23 -- Repeating Zero Pattern
25
26 -- Repeating Zero Pattern
28
29 -- Repeating Zero Pattern
31 -- Repeating Zero Pattern
32
34 -- Repeating Zero Pattern
37
40 -- Repeating Zero Pattern
43
48
49
51 -- Repeating Zero Pattern
52 -- Repeating Zero Pattern
53
54
55 -- Repeating Zero Pattern
56 -- Repeating Zero Pattern
58
59
61
64 -- Repeating Zero Pattern
67
```

Taking a look at Channel 10 I get the following changes when moving Channels one at a time, snippet of outputs
```
Throttle Neutral: 
C: 0A P: 44 00 00 FF 6A A5 D4 80 00 00 00 00 00 00 00 00
C: 0A P: 44 00 00 FF 6A A5 D4 A0 00 00 00 00 00 00 00 00
C: 0A P: 44 00 00 FF 6A A5 D4 80 00 00 00 00 00 00 00 00
C: 0A P: 44 00 00 FF 6A A5 D4 80 00 00 00 00 00 00 00 00
C: 0A P: 44 00 00 FF 6A A5 D4 A0 00 00 00 00 00 00 00 00

Throttle Full:
C: 0A P: 44 38 00 FF 6E 4C 7A 80 08 00 00 01 10 00 20 00
C: 0A P: 44 36 00 FF 6E EE 20 80 00 00 00 00 14 80 00 00
C: 0A P: 44 36 00 FF 6E EE 20 A8 00 08 00 10 00 00 00 10
C: 0A P: 44 38 00 FF 6E 4C 7A 92 04 84 80 C0 80 00 84 02
C: 0A P: 44 38 00 FF 6E 4C 7A A0 00 00 00 00 00 00 00 00
C: 0A P: 44 38 00 FF 6E 4C 7A 80 00 00 00 02 00 00 00 00
C: 0A P: 44 38 00 FF 6E 4C 7A B7 AB 70 00 00 00 00 00 00

Throttle Brake/Reverse: 
C: 0A P: 44 C0 00 FF 62 97 78 80 00 00 00 00 00 00 00 00
C: 0A P: 44 C0 00 FF 62 97 78 A0 00 00 00 00 00 00 00 00
C: 0A P: 44 C0 00 FF 62 97 78 A0 00 00 00 00 00 00 00 00
C: 0A P: 44 C0 00 FF 62 97 78 A0 00 00 00 00 00 00 00 00
C: 0A P: 44 C0 00 FF 62 97 78 80 00 00 00 00 00 00 00 00

Steering Center:
C: 0A P: F8 00 00 FF 69 30 A2 A0 40 00 00 00 00 00 00 00
C: 0A P: F8 00 00 FF 69 30 A2 A0 00 00 20 20 00 00 00 00
C: 0A P: F8 00 00 FF 69 30 A2 A0 40 01 00 00 00 00 80 00
C: 0A P: F8 00 00 FF 69 30 A2 A0 00 00 00 00 00 00 00 00

Steering Left:
C: 0A P: 8A 00 00 FF 68 79 86 A0 00 00 00 00 00 00 00 00
C: 0A P: 8A 00 00 FF 68 79 86 80 00 00 00 00 00 00 00 00
C: 0A P: 8A 00 00 FF 68 79 86 80 00 00 00 00 00 00 00 00
C: 0A P: 8A 00 00 FF 68 79 86 80 00 00 00 00 00 00 00 00

Steering Right:
C: 0A P: 8C 00 00 FF 6B 84 60 A3 4A D3 25 15 99 05 A1 4A
C: 0A P: 8C 00 00 FF 6B 84 60 84 16 89 01 06 44 92 28 16
C: 0A P: 8C 00 00 FF 6B 84 60 A2 83 08 0F FF 7F FF D7 DF
C: 0A P: 8C 00 00 FF 6B 84 60 A4 82 00 04 60 14 00 10 91
C: 0A P: 8C 00 00 FF 6B 84 60 8A 84 16 21 49 42 1A 05 40
```
Looks like I'm on the correct track? As I can see movement on certain values consistent with input, for example the bolded portion is changing 

C: 0A P: **_44 C0 00 FF 62_** 97 78 80 00 00 00 00 00 00 00 00
C: 0A P: **_44 C0 00 FF 62_** 97 78 A0 00 00 00 00 00 00 00 00
C: 0A P: **_44 C0 00 FF 62_** 97 78 A0 00 00 00 00 00 00 00 00
C: 0A P: **_44 C0 00 FF 62 97_** 78 A0 00 00 00 00 00 00 00 00
C: 0A P: **_44 C0 00 FF 62_** 97 78 80 00 00 00 00 00 00 00 00

Thanks again for your help and teaching! Also Please let me know if you would like this data structured in a different way, kind of winging it.

pascallanger, 2024-04-14:
The payload is 5 bytes long so definitively something new.
Steering 68 8A...69 F8...6B 8C -> 08A...1F8...38C => 1F8 center should probably be 1FF but even then the end points are not really symetrical on both ends.
Anyway the payload content is easy. There are 4 channels sent.

The RF channels are good, I know how to get them from the radio ID.

I just need the timing to add the sub protocol. For that I've updated (but untested) the code to show the packets timing so download the latest code, compile, upload, serial monitor up, set RF channel 10, power on the original TX and record the timing. I need a large number of packets to get the average time. So this time please paste or save in a file what you get without filtering.

Pencilvestr, 2024-04-14:
Attached text file of about 4 minutes of output. Not moving any channel.
[Timing_MPM_41424.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/14971842/Timing_MPM_41424.txt)

Just want to say that you are awesome at what you do and I really appreciate the effort in getting this added and walking me through the process.

pascallanger, 2024-04-15:
Test the protocol SLT/V1_4CH available here: https://downloads.multi-module.org/latest-test/

Macgyver46, 2024-04-15:
I was able to bind my mt12 4&1 using the v1_4ch to my granite grom , steering,  throttle, brake and reverse work.

Pencilvestr, 2024-04-15:
Loaded the test build on my MT12...doing the same thing, just knocking the RX out of binding mode. Maybe i'm doing something wrong?

> I was able to bind my mt12 4&1 using the v1_4ch to my granite grom , steering, throttle, brake and reverse work.

What model RX is on those? What steps did you take to bind?

Macgyver46, 2024-04-15:
It's an SLT2 that connects to a spektrum 25a 2&1 esc/reciver.  I just created a new model and selected SLT, THEN SELECTED  the V1_4ch long pressed the power button on the grom , then hit bind on the radio. The grom made a long beep. I had to turn off bind on the radio.   

I'm just experimenting but it works as far as I can tell when the other test builds all failed to bind.

Pencilvestr, 2024-04-15:
No idea..... I can get it to bind to my t16s every single time. Not having any luck at all on the MT12.

Macgyver46, 2024-04-15:
Could the version of edgetx make a difference?  I have version 2.10.0- nightly date : 2023 - 12-22

Pencilvestr, 2024-04-15:
I just tried a every version available for the MT12 no dice. Just knocks the RX out of bind mode but doesn't actually bind

pascallanger, 2024-04-16:
Ok since the RX receives the bind packets it looks like the problem is instead around the module ID.
Can  @Macgyver46 and @Pencilvestr  share your Global IDs using the MultiConfig.lua scripts?
Original radio ID: F4 71 8D 01 -> OK
TX16S ID: 65 50 FE 30 -> OK
MT12_Pen: ?? ?? ?? ?? -> NOK
MT12_Mac: ?? ?? ?? ?? -> OK

@Pencilvestr just to be sure that I'm on the right track, can you put your TX16S ID on the MT12 (using MultiConfig/lua)? If your MT12 works that way, it means that not all the IDs are allowed.

Macgyver46, 2024-04-16:
I'm new to all this and very limited on what I can do.  I thought my successful bind and control with the same hardware would be helpful enough.

pascallanger, 2024-04-16:
@Macgyver46 it's easy, just go to the scripts, run the multiconfig.lua script and share a picture of the screen.

Macgyver46, 2024-04-16:
Global ID: 29 F9 BE 3E

Cyrf ID: C4 A8 25 DF C1 FB  

I hd to add the script to my controller but think I found what you were looking for and another ID #

Pencilvestr, 2024-04-16:
> Ok since the RX receives the bind packets it looks like the problem is instead around the module ID. Can @Macgyver46 and @Pencilvestr share your Global IDs using the MultiConfig.lua scripts? [...]

You nailed it Pascal.

T16S Global ID: 65 50 FE 30  - Binds Fine
MT12 Global ID 57 0B 31 A5  - Would not Bind

Changed the MT12 to match the Global ID of the MT12 and it bound instantly. 

What are the issue of having both Radios have the same global ID?

pascallanger, 2024-04-18:
Original radio ID: F4 71 8D 01 -> OK
TX16S ID: 65 50 FE 30 -> OK
MT12_Mac: 29 F9 BE 3E -> OK
MT12_Pen: 57 0B 31 A5 -> NOK

> What are the issue of having both Radios have the same global ID?

You shouldn't use both TXs at the same time on the same protocol. But changing the ID is not the right method... We need to find out the IDs that are causing issues.

@Pencilvestr I've modified the code and you now have some homework. On your TX16S running the latest code published some minutes ago, bind your RX starting with the RX number 0, then RX number 1..2..3..... up to 31. I'm expecting that for some values of RX number your RX will fail, write these values down.

Pencilvestr, 2024-04-18:
Went through 0-61.
Only channels that refused to bind at all were 8 / 11. 

Of note, some channels seemed to bind quicker and "better". What I mean is when the original TX is bound to the car, the car's RX light will blink until it is bound. Once bound it will be solid. Some of these channels will bind and the controls work, however the RX light will blink, with different intensities. I can go through and identify these channels as well if you'd like.

pascallanger, 2024-04-18:
Flickering usually means packet loss so that's bad. If you could identify the ones with flickering. Was there a lot of them?
I was not anticipating such RF channels issue. we might not get to the bottom of it with this method. The right method would be to connect a logic analyzer on the RX rf chip...
Have you really done 0 to 61 or you meant 31?

Pencilvestr, 2024-04-18:
Yes I did 0-61. 
It goes pretty fast as this RX has a power switch and bind plug and will go into bind mode once switched on.
I'll get you a list of the one that has very noticeable flicker, some were very hard to see

Let me see how hard it would be to connect the a logic analyzer on the RX

pascallanger, 2024-04-18:
Above 31 is not needed as it will have no effect at all, basically using your original ID unchanged...

ProvenRiver764, 2024-04-19:
Got my Arrma Granite Grom connected to my MT12. It refused to bind so I went through receiver IDs till it did. It wouldn't connect at all till I got to 28. 30 seems to be the most stable but the steering is slightly choppy.

pascallanger, 2024-04-19:
@ProvenRiver764  could you share your global id ?

Pencilvestr, 2024-04-19:
> Above 31 is not needed as it will have no effect at all, basically using your original ID unchanged...

0-31 two times in back to back. T16s Results:

No Bind:
8

Slow RX Blink after binding:
10,11,12,13,25,26,27,28,29

Fast RX Blink after binding:
3,5,6,7,9,14,15,17,18,19,21,22,23,24,30,31

Note:
Everything not listed bound without RX flashing. 
Even when bound with RX flashing controls still work. 
Steering channel is a little "twitchy" compared to using stock TX no matter if RX blinking after being bound etc etc

ProvenRiver764, 2024-04-20:
> @ProvenRiver764 could you share your global id ?

Gladly, how do I view it?

Pencilvestr, 2024-04-20:
> > @ProvenRiver764 could you share your global id ?
> 
> Gladly, how do I view it?

Go to scripts -> then multiconfig.lua -> "Global ID"

ProvenRiver764, 2024-04-20:
> > > @ProvenRiver764 could you share your global id ?
> > 
> > 
> > Gladly, how do I view it?
> 
> Go to scripts -> then multiconfig.lua -> "Global ID"

Global ID: 57 1B DD 50

pascallanger, 2025-02-26:
Linked to this issue: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1046
Protocol is now working fine and documentation updated.

closed 2025-02-26 by pascallanger