pascallanger/DIY-Multiprotocol-TX-Module #967: [new protocol] Traxxas legacy 2.4

RobertoD91, 2024-04-17:
Hi, I have an old Traxxas 2228 TX and a 2217 RX.

From this https://traxxas.com/support/Traxxas-TQi-Compatibility-Guide , I understood that it is different from the already implemented Traxxas TQ; indeed, the 2217 receiver doesn't bind to my multi-module (version 1.3.4.0).

I disassembled the radio:

[image]

I assume this is the real transmitter, right?
[image]
[image]

So, should I sniff this connection?
[image]

This is the back of the PCB, for reference:
[image]

Any suggestion?

Pencilvestr, 2024-04-17:
I'd desolder that RF lid and grab some pictures of what is underneath

pascallanger, 2024-04-18:
If you have a saleae compatible logic analyzer (less than $10 on ebay), hook it up to these 8 pins:
[image]
Don't forget to connect GND as well.

RobertoD91, 2024-04-22:
> I'd desolder that RF lid and grab some pictures of what is underneath

Can't desolder, but i found this: https://fccid.io/XVE-SA09181/External-Photos/Modular-Pictures-1202329

[image]

pascallanger, 2024-04-22:
We don't see anything anyway... Connect a logic analyzer on the 8 pins I've pointed out earlier.

pascallanger, 2024-04-25:
Please use logic 1.2.14 instead of 2.xx
It looks like the RF module was not plugged to the motherboard and therefore the CPU went in a loop not finding it was looking for. It means that you must connect the RF module. Out of 4 signals MOSI/MISO/SS/CLK, 2 signals MISO/SS are missing but may be due to the fact that the RF module was not present on the capture. MOSI is connected on D2 and CLK on D4.
For the dumps, make sure the TX has all the trims centered and D/R set to max range.
You need to do different dumps for each action in a separate file:
 - power on not touching anything
 - power on in bind mode with a RX in bind mode
 - power on, CH1 mid, min, mid, max, mid
 - power on, CH2 mid, min, mid, max, mid
 - power on, CH3 mid, min, mid, max, mid
 - any other features or buttons

pascallanger, 2024-04-25:
I can't open your files as I get the following error: 
[image]
Are you really using logic 1.2.14 ? https://support.saleae.com/logic-software/legacy-software/older-software-releases#id-1.2.14-download

pascallanger, 2024-04-25:
Just do the power on for now to see if all is ok before losing time on others

pascallanger, 2024-04-25:
Far more better. On the file poweron: CH0=CLK, CH3=MOSI, CH5=SS and CH7=MISO.
I don't have the time to look at it further but that's a good start.

pascallanger, 2024-04-26:
Normal packets
-----------------
SOP 3C 37 CC 91 E2 F8 CC 91 . Is it constant? Set before reading the CYRF ID D8 AA 59 E6 44 FB so it could be constant.
RF channels alternates between 0B and 34. Where are they coming from ? I'm wondering if 0B could be the main RF channel and 34 (04 bind channel xor 30) is the channel that the RX uses to find the main channel. The TX listens on both channels and read the RSSI right after the packet is transmitted.
Timing is CH34 (where the sticks are read) then 7ms later on CH0B then 13ms later on CH34 for an overall 20ms loop.
Payload 16 bytes
02|D8|AA|59|E6|A2|C6|05|D9|05|90|07|90|07|0A|78
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Type|ID0|ID1|ID2|ID3|UNK1|CH1L|CH1H|CH2L|CH2H|CH3L|CH3H|CH4L|CH4H|UNK2|XOR

Type: 02 normal packet?
ID0..3: CYRF ID
UNK1: unknown value A2? Bind state=2 done?
CH1H:CH1L: channel 1 value 042D..05C6..0733 -> looks like PPM values with a center on 5DC=1500 but the low part being truncated to 1069 and high part 1843. Any end points being set on the TX? Is this the throttle channel?
CH2H:CH2L: channel 2 value 036E..05D9..0840 -> looks like PPM but with extended range this time center 5DC=1500 low 878 and high 2112.
...
UNK2: unknown value 0A? Looks like the first RF channel -1: 0B-1=0A.
XOR: XOR of all payload values from Type to UNK2 stored in XOR

Bind packets
--------------
RF channel 04
Timing every 10ms
Payload 8 bytes
2A|D8|AA|59|E6|A0|03|44
-- | -- | -- | -- | -- | -- | -- | --
Type|ID0|ID1|ID2|ID3|UNK1|UNK2|XOR

Type: 2A bind packet?
ID0..3: CYRF ID
UNK1: unknown value A0? Bind state 0?
UNK2: unknown value 03? Looks like the first RF channel -1: 04-1=03.
XOR: XOR of all payload values from Type to UNK2 stored in XOR

RX sends:
02|A8|03|A9
-- | -- | -- | --
Type|UNK1|UNK2|XOR

Type: 02 normal packet?
UNK1: unknown value A8? Ack bind state 0 with a xor 8?
UNK2: unknown value 03? Looks like the first RF channel -1: 04-1=03.
XOR: XOR of all payload values from Type to UNK2 stored in XOR

After receiving from the RX switches to:
2A|D8|AA|59|E6|A1|92|05|D9|05|90|07|90|07|03|0E
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Type|ID0|ID1|ID2|ID3|UNK1|CH1L|CH1H|CH2L|CH2H|CH3L|CH3H|CH4L|CH4H|UNK2|XOR

Type: 2A bind packet?
ID0..3: CYRF ID
UNK1: unknown value A1? Bind state 1?
CH1H:CH1L: channel 1 value
CH2H:CH2L: channel 2 value
...
UNK2: unknown value 03? Looks like the first RF channel -1: 04-1=03.
XOR: XOR of all payload values from Type to UNK2 stored in XOR

RX sends:
02|A9|03|A8
-- | -- | -- | --
Type|UNK1|UNK2|XOR

Type: 02 normal packet?
UNK1: unknown value A9? Ack bind state 1 with a xor 8?
UNK2: unknown value 03? Looks like the first RF channel -1: 04-1=03.
XOR: XOR of all payload values from Type to UNK2 stored in XOR

After receiving from the RX switches to:
02|D8|AA|59|E6|A2|92|05|D9|05|90|07|90|07|03|25
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Type|ID0|ID1|ID2|ID3|UNK1|CH1L|CH1H|CH2L|CH2H|CH3L|CH3H|CH4L|CH4H|UNK2|XOR

Type: 02 normal packet?
ID0..3: CYRF ID
UNK1: unknown value A2? Bind state=2 done?
CH1H:CH1L: channel 1 value
CH2H:CH2L: channel 2 value
...
UNK2: unknown value 03? Looks like the first RF channel -1: 04-1=03.
XOR: XOR of all payload values from Type to UNK2 stored in XOR

After some time switches to normal packets alternating on RF channels 0B and 34
Do we need to do all the bind states or just sending the initial bind packets is enough?

pascallanger, 2024-04-27:
Please try again, stupid mistake...

pascallanger, 2024-04-29:
I don't understand why it's not working... Please test v1.3.4.8

pascallanger, 2024-05-02:
Please test again. If it doesn't work then I have no idea why...

pascallanger, 2024-05-03:
I don't get it...
You are not trying to bind right? Just control an already bound RX to your original TX, correct?
The RX doesn't do anything? The blinking pattern stays the same when you power on the multi module?
Have you tried to control with all the 4 channels set to their middle positions in the mixers?

pascallanger, 2024-05-04:
I'm getting really annoyed, never spent so much time...
Can you try again with the latest code?

Is it possible to connect to the SPI on the RX side?

pascallanger, 2024-05-04:
> It works! 🥳 All four channel

Wow that's not too early. But now I need to understand why it works... We'll need to do some testing.

rOOOOOOOOn, 2024-12-03:
Hi, I tried to bind my new Traxxas Mini Maxx to my MT12 using v1.3.4.12. I did not manage to get it to bind either with TQ1 nor TQ2, gave up after 15minutes and installed a ELRs receiver in it. So I know have a Traxxas TQ Remote and a TQ Receiver (#6519) laying around, if there is interest I could send it over (preferably Europe) for free, so please let me know.
[image]
[image]

rOOOOOOOOn, 2024-12-15:
Ok as for my previous comment, I was just me not being able to read. v1.3.4.12 with that 6519 reciever and TQ2 protocol work just fine, I just was not used to how you bind a TRAXXAS receiver.

Sorry and thanks for all your work.