pascallanger/DIY-Multiprotocol-TX-Module #977: Hitec telemetry not fully received

pierrotm777, 2024-05-14:
I open this issue for help to complet the hitec telemetry.
Actually, my Tx16s MkII 4IN1 receive only  TRSS, TQly and RxBt.
I should like to have all other features as rpm, current, voltage, temp, and fuel, etc..
For my test, I use:
- Radiomaster Tx16s MkII 4IN1
- Hitec optima 7 receiver
- Hitec telemetry station HTS-SS
- Hitec RPM sensor
- Hitec Fuel.

I have a logic analyser.
Thanks for your help,
Pierre

pascallanger, 2024-05-14:
Now that you give the sensors you are using, this is strange as it should be working at least the RPM but let's check what the module is receiving.

pierrotm777, 2024-05-14:
How

pierrotm777, 2024-05-14:
Reopen sorry

pascallanger, 2024-05-15:
Frame 11 contains the RXBatt
RPM and FUEL should be in frame 15
Frame 1B contains the Vario information which is far from the RPM and Fuel you are talking about...
I've looked at both hitec_v2-1 and hitec2. There is only a response on frame 11.
Are you using real sensors or playing around?
The sensors I've implemented on OpenTX/EdgeTX are at the end of the hitec protocol file with the frames content: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/2bdbd7088c0846b08be1d47e889d0cb85cf533ff/Multiprotocol/Hitec_cc2500.ino#L405
If the frames arrive then the radio should display them.

pierrotm777, 2024-05-15:
I use real sensor for these tests.
But I use optima as subprotocol and not opt_fw.
As see if that changes something with opt_fw.

pascallanger, 2024-05-15:
So Optima is the one you want

pascallanger, 2024-05-15:
Do you see the i2c frames of your sensors?

pierrotm777, 2024-05-15:
Curiously yes, I but they are empty.

pascallanger, 2024-05-15:
If empty then they won't display... Verify the connections, that's all I can say.

pierrotm777, 2024-05-15:
My connections are correct.

pascallanger, 2024-05-15:
Then may be the telemetry station is faulty or the sensors...
We can check what the module is receiving but I fear that it will be the same as what you have on i2c.
With an external multi module, download the latest sources, select only a few protocols in _config.h and compile with debug option set to native for an irangex module or ftdi for the others, upload, open the serial monitor and you should see the debug prompts.

pierrotm777, 2024-05-15:
Can i use your [mm-stm-xn297dump-usbdebug-v1.3.4.0.bin](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/download/v1.3.4.0/mm-stm-xn297dump-usbdebug-v1.3.4.0.bin) or i need to compile my own version ?

pascallanger, 2024-05-15:
> I have a diy external module for another radio, but it is not compatible with my tx16s external box. I need to work for adapt it, add power dc converter etc... I see that as soon as possible

Debug is only available on stm32 modules

pascallanger, 2024-05-15:
> Can i use your [mm-stm-xn297dump-usbdebug-v1.3.4.0.bin](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/download/v1.3.4.0/mm-stm-xn297dump-usbdebug-v1.3.4.0.bin) or i need to compile my own version ?

I can't remember which protocols are included in the xn297dump version. The best is to compile by yourself.

pierrotm777, 2024-05-15:
Ok, I use a blackpill board as module.
This is my Diy board, here a bluepill but i use a blackpill.
[image]

I have build a board for my external box.
[image]
Witch pin i must to use for Rx (B10) and Tx(A3).

pascallanger, 2024-05-15:
It might just be easier to add a FTDI on your TX16S internal module...

pierrotm777, 2024-05-15:
Do you think that this _config.h file would be ok for my blackpill on my external tx16s ?
[Config.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/15324047/Config.zip)

pascallanger, 2024-05-15:
From a quick review it looks good.

pierrotm777, 2024-05-15:
Ok, i shall try to update and adapt my module tomorrow.

Thanks for your help,

Pierre

pierrotm777, 2024-05-16:
Sorry Pascal,
I was very busy this day.
What I must define into my config file for build a FTDI debug version.
Tx to S-port pin and Rx to PPM_OUT pin.
I upload always my firmware with a FTDI module on pins a9/a10 and +5v/gnd.
I suppose that the debug uses these pins?

pascallanger, 2024-05-16:
It's not in the config file but in Arduino. Set Debug Option to FTDI.

pierrotm777, 2024-05-19:
Hello Pascal,

Sorry again for this delay.
I have some issue for have a firmware witch bind my hitec receiver.
Do i need to uncomment these two lines ?
[image]

I send you again my last config file.
[_Config.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/15368765/_Config.zip)

For upload the firmware, i hold the BOOT button and power the module and i use my FTDI module for upload the firmware.
I connect well the A3 pin to S-Port and B10 to PPM_OUT pin.

But no bind :-(

pierrotm777, 2024-05-19:
I have tried also to use the flash_multi tool for upload my firmware with same result, no bind.
I have also tried the mm-stm-xn297dump-ftdidebug-v1.3.4.0.bin firmware.
No bind.

pascallanger, 2024-05-19:
What does the LED ?

pierrotm777, 2024-05-19:
The LED BLINK each 500ms i think.
I have tested my module with my DIY handset and my LED stay well to ON.
So my DIY handset see well that HITEC protocol is on the MPM module.
But on my TX16S, the LED blink always.

pierrotm777, 2024-05-19:
It's i invert my serial connection, my led is now ON and i can bind :-)
I continue now.

pierrotm777, 2024-05-19:
Bind is now ok, i can move a servo on my channel 3, but i haven't telemetry.
My stm32 haven't inverted serial port.
I have commented all telemetry defines:
//#define INVERT_TELEMETRY

//#define INVERT_TELEMETRY_TX
 
//#define MULTI_TELEMETRY

//#define MULTI_SYNC

pascallanger, 2024-05-19:
And the debug display, this is what's interresting for now...

pierrotm777, 2024-05-19:
[debug1_hitec.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/15369154/debug1_hitec.txt)
This is my first debug file with RPM and FUEL sensors

pierrotm777, 2024-05-19:
My wiring for a bind ok is that:
[image]
But i don't receive telemetry  TRSS, TQly and RxBt as i can with the internal MPM module.

pierrotm777, 2024-05-19:
I receive also these infos:
Module Flash size: 128KB
EEPROM initialized: 0
Protocol selection switch reads as 0
Module Id: 3700705c
Init complete
Protocol selected: 0, sub proto 0, rxnum 0, option 255
Protocol selected: 39, sub proto 0, rxnum 0, option 0
Proto=Hitec, nbr_sub=3, Sub=Optima, Opt=2, FS=0, CHMap=0, rfSw=1

pierrotm777, 2024-05-20:
Hello,
Do you need other debug?
If I can help, say me how to do.
Pierre

pascallanger, 2024-05-21:
For the telemetry to work on the radio, you have to invert B10 with something like a simple transistor B=B10, C=SPORT, E=GND.
Exactly like on I2C, the data in your telemetry is full of zeros just the RXBatt value in frame 11.
Anyway, with what you get the radio should display plenty of sensors. You must do the sensors detection while the RX is powered and connected to the station.
The decoding of the frames is happening here in EdgeTX: https://github.com/EdgeTX/edgetx/blob/53c794da8cd3dcbc6d49597f642799a3c6d9d49b/radio/src/telemetry/hitec.cpp#L220
Since you had some telemetry values like RSSI and more displayed, it looks like everything works as expected. It is a matter of doing the sensors discovery with the station plugged (no need for sensors to be connected).
Are the telemetry frames coming at a good pace? You seem to miss quite a few which is most likely a bad tuning or too close from the RX.

pierrotm777, 2024-05-21:
The station and sensors seems well connected. I think to a Rx data input default.
I shall try to find another receiver for my tests.
Thanks for your help.

pascallanger, 2024-05-21:
The RX and station are well connected since you get the frames 11 and up, The RX alone send frame 10 only, this is where if you do the sensors detection at that time it won't work. Just that the sensors connected to it don't send anything. RPM being at zero is normal, you need to move a magnet in front of it to get values. I'm not sure what the fuel sensor is supposed to send by default but 0 looks like a good default value too.

pierrotm777, 2024-05-21:
I shall try to simulate a blade and look at if the rpm sensor return something.

pascallanger, 2024-05-21:
Yes you need to do that or you have 0 rpm...
But even with everything at zero, you should see on the radio plenty of sensors. At least that was the case on OpenTX, may be a bug has been introduced in EdgeTX but that would be strange since you have the RX batt from what you said. All I can think of is that you've done the sensors discovery before having the station correctly plugged.

pierrotm777, 2024-05-21:
I have some news.
If i use my DIY handset with a DIY MPM module (STM32 128K) i have this debug file.
[debug2_hitec.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/15387353/debug2_hitec.zip)
As you can see, the frame seems not to 0 .
I do a new test with my TX16S and the same MPM module .

pierrotm777, 2024-05-21:
Same results with my DIY handset.

pierrotm777, 2024-05-21:
I don't understand why, but if i use again my internal module, i receive no all the telemetry (20 sensors).
But RPM stay to 0.
Very curious, may be the method for bind?
I don't know !

pierrotm777, 2024-05-21:
Please read :
I don't understand why, but if i use again my internal module, **i receive now all the telemetry (20 sensors**).

pierrotm777, 2024-05-21:
But now, i haven't a debug possibility :-( !

pascallanger, 2024-05-21:
I don't know what you want to debug on the module... The module does what it is supposed to do. The protocol is working as expected and forward the received telemetry to the radio to be decoded and displayed. So having debug is kind of useless really... I'm still unsure what you are looking for here... By the way, The TX16S internal module can run debug easily using a FTDI as mentioned earlier.
If you get no reading from your sensors it means that either you are not activating them (for example RPM, use a magnet in and out quickly and the rpm will appear) or the sensors are broken or there is a sensor cable issue or the station has an issue... But in the end it's not an issue with Multi or the radio.

pierrotm777, 2024-05-21:
My final goal is to build a telemetry simulator module for Hitec. But I didn't know if I had a handset issue or a problem with my telemetry simulator. 
I know now that my problem is with my simulator module which don't return the good frames.

pascallanger, 2024-05-21:
Ok closing this thread then.

closed 2024-05-21 by pascallanger

pierrotm777, 2024-05-21:
Yes, thanks again Pascal