pascallanger/DIY-Multiprotocol-TX-Module #978: Qidi560/FX9630 Bind issue

Kfreel2019, 2024-05-15:
Hello. There seems to be an issue with binding the Qidi560 to OpenTx using FX9630 protocol Bind seems semi successful as most controls respond normally. All lights flash rapidly, there has been no success in assigning a switch to control the lights.  With gyro off, plane idle and no input from Tx servos will randomly move at irregular intervals.  I have seen lights behave properly only when toggling Bind on tooff by pressing and holding "enter" button after achieving a new bind

I am a complete novice with RC in this aspect, I have a Jumper TLite if that's relevant

pascallanger, 2024-05-15:
Can you do the same servo test with the original transmitter? Just to be sure that it's not a behavior of the plane to move its servo on its own time to time.
I don't really understand what you are doing to get the lights to behave properly... Can you give all the steps like 1. power on the plane LEDs are blinking, 2. launch a bind on the radio LEDs stop blinking...

Kfreel2019, 2024-05-15:
I tried email response with videos but didn't work.
I connected the stock TX and had jo abnormal behaviors over a period of 5 minutes. Process to see lights work normally- tx on with bind ch selected ,plug in battery,  then select bind in reciever menu.(plane binds controls function properly lights flashing abnormally) at this point if you select bind the lights will act appropriately for approximately 1 sec before going dark and the bind refreshing. Note that when losing connection to the stock TX light go off immediately. 

If its of any help if I could find and ship a stock Rx I will do so, though there is no listing kn BG or Ali for one at the moment.

pascallanger, 2024-05-15:
I've finally purchased this plane. One more... I just have too many I shouldn't have done it but... You are welcome if you want to make a small donation for it instead of your proposition of shipping the RX ;-)
So let's wait for the delivery end of next week.

pascallanger, 2024-05-16:
I've received the QIDI-560 TX.
I'm totally puzzled as there is a bug with the light function... From what I see if you long press once the "Function key" to change the LEDs pattern then the trims won't work correctly anymore. To test the bug, power on the radio and plane, change the trim on roll/aileron, you should see the effect of the trim on the ailerons, set back the roll/aileron trim to center, now long press the "Function button" until it beeps, try to change the trim on roll/aileron, you should see the trim effect on rudder now despite trying to adjust ailerons... Same goes with the rudder trim which you can't set anymore once you longed press that button.
Basically the long press on the light mess up the trims. And adjusting the rudder trim (and may be the elevator trim) will most likely affect the LEDs...
So now the thing is that I'm setting all the trims all the time and therefore it has a side effect of playing with the LEDs again and again. On their sides, they only send the trim that you are touching to the plane. So if you don't touch the trims nothing happens, you might just see something when playing with the trims.
Each time you press the rudder trim, it's like long pressing 2 times the function button which I don't know what it does on the LEDs.
Can you test my assumption on the trims bug before I start to implement something?

Basically, if I disable the trims for this plane, things should start to work. The DR channel of the QIDI 550 is the acrobatic feature (short press on function key) on this plane which is launched each time you toggle the associated switch.

Kfreel2019, 2024-05-16:
Yes, I will give it a go in about a half hour. Just got home from work and need to wash up.

pascallanger, 2024-05-16:
And I need to eat... I've already spent more than 3 hours in a raw wondering how I am going to deal with this bug...

Kfreel2019, 2024-05-16:
Yes I observe the same action. The rudder trims after turning the light off almost seem to aft as a rudder button. If you do a short press it will move and return to it's position. I tried setting the trims to switch's on my Jumper Lite- when toggling them the flash pattern of the lites change.

pascallanger, 2024-05-17:
Please test v1.3.4.9 available here: https://downloads.multi-module.org/latest-test/
I've added a new subprotocol under FX called Q560, check the channels assignment: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-q560---2

Kfreel2019, 2024-05-17:
I will test when I get home from work(will be about 9 hours).  Other users have reported instability issues so TBH I haven't even put it in the air yet. 

Also, thank you for working on this!

Kfreel2019, 2024-05-18:
So after struggling to figure out updating the firmware, (I did say I'm a novice, right?) It all looks good to go. Leds go on off, gyro switchs work, only had to reverse rudder.  I will be able to test fly on Sunday.  Thanks Pascal!

closed 2024-06-04 by pascallanger

jwebber23, 2024-07-12:
Hi, i looked at the link and found version 1.3.4.10 but only had the version taer rather than the aetr that i normally have.  Will there be a test build in aetr or do i need to wait for the stable release update?

pascallanger, 2024-07-12:
Check again in the file it's there.... It's even the test version 1.3.4.11 since this issue has been completed

jwebber23, 2024-07-13:
Hi, 
Sorry, yes it is there.  I was using the search in Windows 11 file browser and it was "helpfully" limiting the search results. 

I have tried it and it all looks good.  I will have a fly tomorrow.

Thanks :-)

jwebber23, 2024-07-14:
All worked as expected.  Thanks again for your help.

mkfrcp, 2024-08-24:
Hi, I try to bind TX16S (4 in 1 mm) and my Qidi 560. I upgraded mm to 1.3.4.11 (TAER) and set FX560 protocol without success. As I've never use bind on channel before, I tried both bind on channel (CH16) an manual bind. It does'nt work at all. Of course, the original transmitter binds immediatly to the plane. Any idea?
Thks in advance for your help :)

pascallanger, 2024-08-24:
The protocol is FX and sub protocol Q560: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-q560---3
Hitting the bind button or checking the bind on channel feature and assigning a switch to ch16 should get the bind going. 
What's the module status on the radio? 
Are you trims nearly centered?

mkfrcp, 2024-08-24:
The 4in1 module works fine with HoTT ou Frsky receivers.
Model memory is totally new (trims are centered).
When I hit the bind switch affected to CH16, "Binding" is shortly displayed on the screen but the plane is not responding.

pascallanger, 2024-08-24:
You need to maintain the bind switch pressed for a few seconds not just a short pulse. But I don't think it's the issue here. 
Hott and frsky are using the cc2500 RF component. Do you have any planes using the nrf24l01?

mkfrcp, 2024-08-24:
I tried to maintain the switch a few seconds. No effect. Unfortunately, I don't have any plane using the nrf24l01 chip. The QIDI TX PCB is not equipped with a nrf24l01 but a CB2401. Based on the datasheet of this component, it is probably a clone of nrf24l01. Is there another test to do ?

mkfrcp, 2024-08-27:
Is there a way to test that the nrf24l01 chip built into the Radiomaster multimodule is working properly?

pascallanger, 2024-08-27:
CB2401 is a PA (power amplifier). They are using a XN297 core in the all in one MCU + RF chip.
No idea how to test the NRF without a model or putting it in debug mode...
Have you tried to bind in low power mode?

mkfrcp, 2024-08-27:
True !
Yes, I tried to bind in low power mode too. Without success.
I'll try to find a model and let you know.
Thks for your help.

alex97barth, 2024-12-12:
> Please test v1.3.4.9 available here: https://downloads.multi-module.org/latest-test/ I've added a new subprotocol under FX called Q560, check the channels assignment: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-q560---2 It should address the lights. I have no idea if it had a link with the servo movements you've seen. I've also been reported that in 6G the plane would continuously do aerobatics, did you experience the same? I can't see why it would be the case since you need to change CH5 for the flag to be set/unset...

I am having the same issue: continually doing aerobatics in 6G mode. Channel 5 set to toggle flight modes. Channel 7 set to LED channel 6 has no efffect on flight modes.

pascallanger, 2024-12-12:
I'm flying with this plane every WE and all the members of my club have one too flying with Multi. No issues so this must be something you are doing.

alex97barth, 2024-12-15:
> I'm flying with this plane every WE and all the members of my club have one too flying with Multi. No issues so this must be something you are doing.

Thanks for letting me know. I will try again and check in. Has anyone had success with an external sbus receiver?

MeTeeZee, 2025-01-23:
> Please test v1.3.4.9 available here: https://downloads.multi-module.org/latest-test/ I've added a new subprotocol under FX called Q560, check the channels assignment: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-q560---2 

Hello, thank you for the work you do for this.  I was wondering when you might expect that the q560 sub_protocol will form part of the "latest stable" release. Many thanks, T

pascallanger, 2025-01-23:
> Hello, thank you for the work you do for this. I was wondering when you might expect that the q560 sub_protocol will form part of the "latest stable" release. Many thanks, T
When I'll be done with some of the protocols I'm working on right now.