pascallanger/DIY-Multiprotocol-TX-Module #979: DSM: More than 12 channels possible?

JenusL, 2024-05-16:
Hi!
Just wondering why there is a limit of 12 channels with the DSM protocol?
Is it a hardware or software limit?
What would be required to support more channels?

mawzthefinn, 2024-05-28:
12ch is the old limit on DSM2/X channel ranges without X-Plus support being implemented.

There is now a 14ch mode in DSMX, but it supports exactly 1 expensive receiver, X-Plus is supported by another 4 (all very expensive, only one in current production)

JenusL, 2024-06-02:
Ok yeah I use one of the cheaper 20 channel receivers that only have serial output (SPM4650 and SPM4651T).
Any idea what special things they use? Is that X-Plus?

mawzthefinn, 2024-06-06:
>14ch requires X-Plus support in the transmitter, as that's how they encode the extra 8ch into the 12ch format (XPlus is 10+8ch or 12+8ch mapping, depending on specific DSMX channel mapping selected).

DSMX is actually limited to 14ch (2 x 7 frame channels), they get the extra channels by encoding them in reduced resolution and variable latency within the 7th channel of each frame (so 6 full res channels per frame + 1 'channel' with multiple low-resolution channels packed in, with full res channel sets alternating so each full res channel is in every second frame and low res channels can be as rare as every 8th frame).

JenusL, 2026-08-16:
I'm super happy to see this being fixed in #1138 and #1179.
Thank you!

closed 2026-08-16 by JenusL