pascallanger/DIY-Multiprotocol-TX-Module #983: Ali Express Mini Heli

Githubbachelor, 2024-06-02:
Hallo I am looking for the protokoll for the Ali Express little Helis for MULTI-Modul 4-1 Jumper. Does anybody know if there is one build? I have the latest version from Multimodul Firmware. Maybe if not can you build that, so that I can fly the Helis with my FRSKY Horus 12S.
Thanks very much beforehand

pascallanger, 2024-06-02:
Aliexpress is selling dozens of helps with different manufacturers. Your request is point less... The latest build contains all the supported protocols.

closed 2024-06-02 by pascallanger

Githubbachelor, 2024-06-02:
Ok but do you know which Protokoll i must use?

pascallanger, 2024-06-02:
Your question doesn't have any sense. You still don't indicate which model... Each model has nearly a different protocol... Have you looked by yourself in the protocol details as the models are indicated ?

Githubbachelor, 2024-06-02:
C186 Pro B105 2,4G RTF RC Hubschrauber 4 Ppropellers 6 Achsen Elektronische Gyroskop für Stabilisierung Fernbedienung Hobbies Spielzeug

Githubbachelor, 2024-06-02:
C159 EC135 Skaliert 2,4G 4Ch RC Hubschrauber für Erwachsene Professionelle Gyro Stabilisiert Einem Klick Rund Flug/Nehmen Off/landung

pascallanger, 2024-06-02:
All the RC era  helis are supported. Please read the documentation to find the protocol.

Githubbachelor, 2024-06-02:
I feel so sorry. But I cannot found? Cause my english and Know how ist Not good enough. Can you send me a link? Please

Githubbachelor, 2024-06-02:
I need the Name of the Protokoll and where can I find it and Dokumentation

Githubbachelor, 2024-06-02:
Found it. Thanks!

Githubbachelor, 2024-06-03:
You did a phantastic work. Thank you Pascal