pascallanger/DIY-Multiprotocol-TX-Module #985: Hoot protocol, how to define SUMD output

pierrotm777, 2024-06-04:
Hello,

I use the Hott protocol with a Graupner GR-24 Pro receiver.
With a true Graupner handset, it's possible to define the channel 8 as SUMD signal.
`SUMD (Signal cumulé numérique) au canal 6 (GR-24PRO canal 8)
Si le signal cumulé numérique est activé au niveau du canal 6 le
signal cumulé est émis (au lieu d'un servo) avec 8 canaux à partir de
cette prise. Le récepteur HoTT configuré en tant que SUMD génère
en permanence un signal cumulé numérique à partir de 8 signaux de
commande de l'émetteur et le prépare au niveau du raccord de
servo défini spécifiquement pour le récepteur. Un tel signal est utilisé
pour des équipements électroniques dans le cadre des nouveaux
systèmes Flybarless et alimentations Power.`
This feature don't exist for this protocol.
Should it be possible to add it ?

Thanks,

Pierre

pascallanger, 2024-06-04:
I would need a dump of the TX with the feature disabled/enabled.

pierrotm777, 2024-06-04:
Ok, i need to compile with the debug option as we have done with the hitec protocol ?
I see that with a guy how has a Graupner transmitter and send you a dump file asap.
Thanks,
Pierre

pascallanger, 2024-06-04:
Humm that's not that as simple as you write it down.
2 options:
 1. Easy for me/Moderate for you: open the Graupner TX and connect a logic analyzer on the CC2500, do a dump and share
 2. Moderate to Hard for me/Easy for you: develop a RX on Multi to receive the HoTT packets and ouput them on debug for analyze

pierrotm777, 2024-06-04:
In fact, I understand just that now.
I think another way is possible by programing directly the receiver with a serial interface.
You can close this issue if you want.
But thanks for your quick answer.

pierrotm777, 2024-06-04:
How yes, a Rx option would be may be the solution for me

pierrotm777, 2024-06-04:
With that solution, my Tx16s would use zRx Hott protocol and I should bind it with an original Graupner handset?

pascallanger, 2024-06-04:
yes this is solution 2 how to, with debug enabled and post the output here

pierrotm777, 2024-06-04:
Ok

pierrotm777, 2024-06-06:
Hello Pascal,
I had speak about my goal to a friend.
I see if it's hansed has the sumd design.
Say me please when you Rx hott will be ready for this test.
Thanks again,
Pierre

mha1, 2024-07-18:
just use the HoTT LUA

pierrotm777, 2024-07-18:
Thanks for this share 👍

mha1, 2024-07-18:
Howto assuming you are using OpenTX or EdgeTX (pictures show EdgeTX on a TX16s connected to a GR16 receiver):
- download MultiLuaScripts.zip (link above) and extract its content to your SD Card. Make Sure you find "Graupner HoTT.lua" in folder SCRIPTS/TOOLS
- select protocol MULTI/HoTT as your interal or external MPM protocol. Bind a Graupner receiver and make sure it is connected and delivering telemetry
-  press the SYS key and click on Graupner HoTT to start the Graupner HoTT LUA script

[image]

- use the Page> key to go to the last receiver configuration page
- use the scroll wheel to move the cursor ( >) to the last line and press Enter (press scroll wheel) 
- edit CH OUT TYPE to SUMD

[image]

Check the Graupner manual for the receiver you are using as the receiver config pages look different depending on the receiver.

closed 2024-07-19 by pascallanger

pierrotm777, 2024-07-20:
I have downloaded the Graupner Hott script and i haven't the same script than you.
I can only use channel 8 for SUMD output but i should like to use the channel 10.

pascallanger, 2024-07-20:
The full menu is sent by the RX itself and displayed on the radio. That's why the display can be very different. If it only proposes sumd on ch8 then that's what it supports. You might look at upgrading the RX fw version.

pierrotm777, 2024-07-20:
Ok, thanks for these infos.
I have already updated my receiver.
I keep the only channel 8 witch run well as sumd output for me.

mha1, 2024-07-20:
> The full menu is sent by the RX itself and displayed on the radio. That's why the display can be very different. If it only proposes sumd on ch8 then that's what it supports. You might look at upgrading the RX fw version but I doubt it will make a difference as this might be a hardware limitation on it.

Yes, check the Graupner receiver manuals. All Graupner receivers I know can only use a specific channel for SBUS or SUMD output. Which cannel it is depends on the receiver type. Check the manuals.

pierrotm777, 2024-07-20:
Ok, I see that.
Thanks

mha1, 2024-07-20:
Also the menus look different depending on the receiver type. But if you know the original Graupner system you sure know all this too. If not, just read the Graupner manuals.