pascallanger/DIY-Multiprotocol-TX-Module #987: FrSkyX2/D16 bug: binding an XM receiver won't work anymore since 1.3.3.33

egonl, 2024-06-06:
While binding a FrSky XM or XM+ receiver works fine in 1.3.3.20, I can't get a successful bind in 1.3.3.33 or 1.3.4.0.

Steps to repeat:
- Flash 1.3.3.33 or 1.3.4.0
- Power up the XM while holding the bind button. The LED stays solid RED
- Press Bind on the transmitter 
 
Equipment used:
- FrSky XM ACCST 2.1.2 FCC
- Taranis Q X7 on OpenTX 2.3.15
- iRangeX IRX4+ with Multiprotocol 1.3.3.33  or 1.3.4.0

rogerq, 2024-08-22:
I confirm this issue exists even on EU LBT module.
I suspect the commit that updated bind sequence to support newer receivers like Archer series broke XM+

commit  a41123deb26 "FrSky Archer Bind"

frankiearzu, 2024-10-03:
Same issue, bind with older version of the MM firmware, but not with the latest one.. I agree with @rogerq that the point where is broke is on the "FrSky Archer Bind" changes.