pascallanger/DIY-Multiprotocol-TX-Module #992: DSMR Telemetry inconsistent refresh for telemetry

jslatten, 2024-07-03:
**The Problem**

I'm currently facing an issue with telemetry with my recently purchased Radiomaster MT12 4in1 transmitter and Spektrum SR315 DSMR receiver. I'm running the latest release of EdgeTX 2.10.1 and have updated the Multimod firmware to V1.3.4.0 (I've tried both AETR and TAER channel mapping if it matters. Currently using firmware with AETR channel map).

The first issue is related to extremely inconsistent telemetry updates from the connected Spektrum Firma Smart ESC (120a - 4s Max) when connected from my Radiomaster MT12 radio to the Spektrum SR315 Receiver.  Sometimes I'll receive refreshed telemetry data every several seconds and other times I'll only get updated telemetry once every 1-2 minutes. Sometimes I'll get no telemetry data at all which typically requires powering the vehicle and transmitter off and on several times until I eventually start to receive telemetry data from the Smart ESC again. Every once in a while (like 1 out of 30) I'll start to receive telemetry  when I first power on the rc car and transmitter, but it's very rare to just work correctly. Once working it will sometimes continue to update/refresh on a semi-regular basis for a while, but other times it will stop working after a little bit of time. While it's working the telemetry is usable, but it would be great if it refreshed a little quicker and more consistently. For the brief period it is receiving telemetry the refresh rate of telemetry is all over the place. It can range from once every 2-5 seconds to once every 10-30 seconds to very sporadic updates to eventually no updates at all. I'm within 1-5 feet from the vehicle/receiver for all of these tests. 

**What am I expecting?** 

I'm looking to receive consistently quick and regular updates of the data for the following sources coming from the Spektrum Smart ESC: Erpm, EVIN, TFET, ECUR, ETHR, and A2. On the other hand there are several other telemetry sources from the SR315 receiver or from the RM MT12 that seem to be consistently available and refresh quite often i.e TRSS, FDEa, FLss, Hold. I believe I'm receiving those every 1 second or possibly more often than that. I first posted this issue over in the EdgeTX project, but they are saying that EdgeTX receives, processes, and displays the telemetry directly from the 4in1 Multi-module radio and shouldn't impact the telemetry reception or refresh rate.

I also have a Spektrum DX6C transmitter that has full telemetry and I've ran a number of tests with the Spektrum transmitter and it seems to have a live stream of all of the ESC telemetry that I'm interested in and it's being refreshed more than once per second it seems. Prior to switching to the RM MT12 transmitter I haven't experienced any issues with receiving telemetry on the Spektrum DX6C transmitter so I don't think it's the Spektrum receiver or my Spektrum ESC.

jslatten, 2024-07-12:
Is there any additional information I can provide to help troubleshoot this issue or any suggestions?

jslatten, 2024-08-13:
It's been a month now without any comments, acknowledgement, or assignment...Is this project dead or did I submit the issue incorrectly? 

Over the past month, I've continued to face the telemetry reception issues. I've read on forums that people are experiencing issues receiving telemetry when the Radiomaster MT12 is very close to the receiver i.e within 1-5 ft or around 30-153cm. I'm wondering if my inconsistent telemetry is really just this issue...I think some people have speculated that the transmitter is getting overloaded from the very close Spektrum SR315 receiver, which results in not receiving & displaying telemetry properly or in some cases intermittent. This seems to make sense as most of my testing is very close to my receiver.

Any help is appreciated! Please let me know if I've done something incorrectly I can resubmit. Thank you!

pascallanger, 2024-08-13:
Sorry I'm on vacation and don't have the time and hardware to look into it.
What you can do is to test different releases and even intermediate builds if possible since I think I saw another report saying that the other IDs were ok at some point and then stopped working. If you could figure out when it broke that will help the troubleshooting. 
For the telemetry, is it only on short ranges or also at a distance?

ken830, 2024-10-13:
I'm experiencing the same. I find low-power mode helps telemetry come through. Putting my hand around the multi-module's antenna also seems to help. Is it just a consequence of the transmitter operating at much higher power than the Spektrum and drowning out the telemetry signal somehow?

pascallanger, 2025-01-29:
Please try the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.26 with your DSMR receiver including multiple receiver numbers.