pascallanger/DIY-Multiprotocol-TX-Module #996: Protocol "MouldKg", order no. 90 is not selectable but should be supported

schuhmer, 2024-07-28:
Hardware: Radiomaster TX16 Mark 2, internal 4in1 Module
Software: EdgeTX v2.10.1
MPM firmware: 1.3.4.0 AETR
No external TX Module installed

Hello, I have flashed all 3 possible MPM firmware versions to select the protocol "MouldKg" within the "Model Setup" -> "Internal RF" -> "Mode"="Multi", protocol "MouldKg" not included within pull down menu.
Tested firmware versions from github (by copying with USB to SD card within TX16Mark2 and flashed from "SYS Menu" -> "SD Card"):
"mm-stm-serial-aetr-air-v1.3.4.0.bin"
"mm-stm-serial-aetr-lbt-v1.3.4.0.bin"
"mm-stm-serial-aetr-sfc-v1.3.4.0.bin"
"Radio Setup" -> "Country code" = "Europe" or "America" doesn't change the behavior of the missing protocol.

Because of my other requirement, to have as much as possible protocols installed I build the firmware by myself. Downloaded source of "v1.3.4.0", installed Arduino IDE 1.8.9 and up to date java. Must comment out HOTT protocol and HOTT telemetrie to fit the firmware into the STM32 of the internal MPM 4in1. No Errors, but "MouldKg" not included within pull down menu. To check if it's not an issue with the comment out of the HOTT stuff, I build another MPM firmware with HOTT not comment out but all "..._RX..." protocol commented out to fit the firmware within the STM32. Also no errors during build, but "MouldKg" not included within pull down menu.

I will attach one of the used "_config.h" renamed as "_config.h.gif". Lines with "//KASC"" on it's end are sometimes commented out to fit the firmware into the STM32.

Btw. it looks like comment out "//#define INVERT_TELEMETRY_TX" doesn't save any amount of flash space, like written in the comment above the statement within "_config.h".

In any case, thanks a lot for your work!
[image]

schuhmer, 2024-07-28:
Hello again, I updated the Arduino IDE from the suggested one (1.8.9) to the last as working reported one 2.1.1 (issue #895 https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/895). Compile also with no errors, but bin files are total equal to the compiled one with 1.8.9. So Protocol "MouldKg" also not included within pull down menu.

**I downgrade the 4in1 Module firmware to version 1.3.2.86 (download and installed the compiled version direct from github "multi-stm-serial-aetr-v1.3.2.86.bin"). Then I can select Protocol "MouldKg" successful within pull down menu.** But that version is not a solution for me.
I also tested newer versions for the 4in1 Module, also direct download of the bin files from github. **But it seems that on version 1.3.3.0 and higher there isn't Protocol "MouldKg" not included within pull down menu.**

I found no indication in any release notes or a issue that this portocol is no more supported. And in version 1.3.4.0 the define "#define	MOULDKG_NRF24L01_INO" is not comment out.

I give up with my investigation and will look here in a few days/weeks for any helpful comment/solution to that issue.

pascallanger, 2024-07-28:
Ok I'll have a look. The good/bad versions might give me a good troubleshooting staying point.

pascallanger, 2024-07-29:
It's an issue in EdgeTX not adding it to the menu for some unknown reason.

schuhmer, 2024-07-29:
Hello pascallanger, thanks a lot for your investigation. It's amazing that I was able to select it with MPM module version 1.3.2.86 within EdgeTX 2.10.1. Maybe a define mismatch. 
I figured out that the oppression of all these protocols (except the SCANNER one) are added during change of EdgeTX form verion 2.8.5 to 2.9.0. I think they are added because the MPM firmware doesn't fully support these protocols at thich early times.

Should I create an issue on the EdgeTX project or will you do this, and who should the close this issue here?

(And again, many thanks for the verry good support of up to 4 Mould King bricks. I can control the toys of my children in a way the Mould King analog transmitter newer support it (bad 4 wheel drive mode))

pascallanger, 2024-07-30:
Prior to v1.3.3.0, edgetx was using it's internal protocol list instead of asking the module which protocols are available like OpenTX was doing. This was due to the touch menu introduced at that time. In 1.3.3.0, I've added a special function which sends the list of available protocols with all the parameters to fill the menus.
By loading v1.3.2.86, you just see the edgetx internal list missing all the new protocols and sub protocols...
The removed protocols are the ones which were disabled at the time in the automated builds due to flash space issue.  Someone must have think at the time that it was a good idea to hide them from the menu despite the fact that you could do a manual build and include them...
An issue is already opened. My 2.11 edgetx build works fine with the proposed change.

closed 2024-07-30 by pascallanger