pascallanger/DIY-Multiprotocol-TX-Module #1002: Irangex nano pl18 edgetx 2.10.3

swemiklan, 2024-08-21:
Flasht edgetx  to pl18  ( no hardware mod ) installd new nano module and tryed to bind. no luck.
Edgetx show ( no multi telemetry detected ) in model page and radio information.
Flash module wiht 1.3.3.33 managede after some trubble dfu mode/ usb i can read firmware of module trou flashmulti 6,1
still no luck binding, sam text in edgetx.  Now i am not able to read module wiht flashmulti, module led are solid green and red slow blink.   Elrs module works fine.
Can i erase module in flashmulti when its not responding/ able to read module.  Other suggestions. 
Bangood Note:
"The STM32 MCU have change to ARM, fireware is ARM CORTEX-M
The new stock will be ARM, Existing inventory is not affected"
Mikael

pascallanger, 2024-08-21:
Which module is it?

swemiklan, 2024-08-22:
IRangeX IRX4 LITE CC2500 NRF24L01+ A7105 CYRF6936 4 I 1 Multiprotokoll TX-modul för Frsky X-lite  from  bangood.

swemiklan, 2024-08-23:
Now module is back in order Thank to Benlye.
Module is not recogniced by tx  ( no multi telemetry detected ).
How do i continue troubleshoot multimodule  (serial conection ) ?
Elrs module is working in same adaptor.

swemiklan, 2024-08-23:
[image]

[image]
[image]

pascallanger, 2024-08-23:
What's the module LEDs doing ?

swemiklan, 2024-08-24:
Solid green and red blink.

swemiklan, 2024-08-24:
Does multimodule need more or other placement of serialport?
Nano adaptor has 3 wires and i have done conectivity test trou adaptor to inside module all good.
Nano adaptor works in radio wiht elrs module and radio works wiht jr adaptor and irangex plus module.

pascallanger, 2024-08-24:
You are not saying how the module blinks...
The module needs 4 wires to work: Ppm, telem, gnd, batt.

swemiklan, 2024-08-24:
Its green solid and red blink.
Ok now we getting som where, schematic on nano  and jr adapters i see 3 connections serialport to the right in adaptor where do fourth (ppm) go?
How can i compere jr bay pinout and nano pinout?
I need to trace out connetion to pl18.
I will start to google  Pinout,   THANKS ALOT  Mikael

swemiklan, 2024-08-24:
PPM from pl18  = found    ( compared wiht fjr2 )
Now where to put it in MPM lite pinout ?
https://www.rcgroups.com/forums/attachmentNew.php?attachmentid=11016912  is all i found, In jr its equivalent toNr5 Pxx out on Nano/lite pinout?    Shoud i solder wire to pin nr 5 in nano adapter to try out?

swemiklan, 2024-08-24:
So i tryed soldering wire from pin ppm/ RX in tx to pin 5  Pxx out in MPM and no difference.
[image] This is Frm301 Connection to Pl18
Adaptor has 3 wires from sport, gnd and +5v to serialport, gnd and vcc in Mpm originaly and now + my wire now.
Is there a need for more wires in adaptor maybe between Head/tx to Pxx in also ?

swemiklan, 2024-08-24:
[image]
Found this at last.
point 1  Is  rouled out.
point 2 FW on radio and module seems correct , inverson setting in use probleby not different mpm works.
point 3 Foulty module hmmm how do i solve that issue?

closed 2024-08-25 by pascallanger