pascallanger/DIY-Multiprotocol-TX-Module #1004: add sub protocol "PTF22S" to "SGF22" protocol

rdba2k, 2024-08-30:
Hi, Pascal:

I have a working code for this airplane called "Park10 F22S". 

https://www.aliexpress.us/item/3256806205368814.html?spm=a2g0o.productlist.main.41.59ab1f30pjQ57C&algo_pvid=5ff483cc-b089-45ed-a8a4-3524f5d06869&algo_exp_id=5ff483cc-b089-45ed-a8a4-3524f5d06869-20&pdp_npi=4%40dis%21USD%21101.49%2130.40%21%21%21716.25%21214.56%21%40210313e917250608200872694ef2a6%2112000037001686288%21sea%21US%213301939029%21X&curPageLogUid=twJxl0Q62NOv&utparam-url=scene%3Asearch%7Cquery_from%3A

Basically the only difference from SGF22 is the binding channel

#define PTF22S_BIND_RF_CHANNEL	10

And the first byte when binding is

		packet[ 0] = 0x61;	// 0x5B;

the first byte of the data packet is

		packet[0] = 0x21; // 0x1B;

I have test fly the code and it is working. Attached is the code I directly modify on 
the original SGF22 code. Hope you can combine this code and add a sub protocol to it, let us call it PTF22S (PT=ParkTen).

Thanks.

Roger

[SGF22_nrf24l01.ino.txt](https://github.com/user-attachments/files/16822765/SGF22_nrf24l01.ino.txt)

rdba2k, 2024-09-01:
I have creates the pull requests, but not sure I did it correctly. Here are the files I made changes to add "PTF22S" sub protocol.

[Multi_Protos.ino.txt](https://github.com/user-attachments/files/16826652/Multi_Protos.ino.txt)
[Multi.txt](https://github.com/user-attachments/files/16826654/Multi.txt)
[SGF22_nrf24l01.ino.txt](https://github.com/user-attachments/files/16826655/SGF22_nrf24l01.ino.txt)
[Multiprotocol.h.txt](https://github.com/user-attachments/files/16826656/Multiprotocol.h.txt)
[_Config.h.txt](https://github.com/user-attachments/files/16826657/_Config.h.txt)

pascallanger, 2024-09-02:
Please test v1.3.4.12 https://downloads.multi-module.org/latest-test/

rdba2k, 2024-09-03:
The flight tests of v.1.3.4.12 on both SG F22 and Park 10 F22S protocol are good.

Thanks.

pascallanger, 2024-09-03:
Great

closed 2024-09-03 by pascallanger