pascallanger/DIY-Multiprotocol-TX-Module #1011: Arduino Pro Mini build - Unable to put into PPM bind

eltesco1, 2024-09-05:
Hi There,

I have built a CC2500 only module for FRsky protocols. I have built the device and triple checked all connections. Device is in PPM mode switch position 1.
Device is on the bench, when i hold down the bind button (pin 13) and power on, i get no status LED (pin 13 via resistor) until i release the button and then all i get is a slow blink (No PPM signal i think). No Matter what i do i cannot get it into bind mode. It also seems i cannot get it into bank select (position 15) red just continues the slow blink.

I can turn the device to serial mode (get faster flashing LED (no serial input i think). However my radio is not yet compatible with the serial function.

Any ideas for tests i can do to find out why?

pascallanger, 2024-09-05:
When you hold down the bind button you short cut the LED which is why you don't get any information until you stop pressing it.
If the LED blinking pattern indicates no PPM or serial then the module won't do anything, no transmission is allowed and therefore binding can't happen.
The bank selection is only available if you have defined more than 1 bank in _config.h. The default builds and the config file by default have only 1 bank defined.

pascallanger, 2025-02-26:
No more requests so closing

closed 2025-02-26 by pascallanger