pascallanger/DIY-Multiprotocol-TX-Module #1013: Traxxas ECM-2.5 Support

inventor7777, 2024-09-06:
Hi!

I have a TRX-4m I’d like to bind to my MT12 4-in-1 but I cannot get it to bind. The ECM 2.5 is Traxxas’ integrated RX/ESC for the TRX-4m. 

It binds perfectly to my TQ & TQi transmitters, but no go with the Traxxas protocol on my radio. When binding is initiated, the light on the ECM instantly stops blinking red and goes to solid red, but it never successfully binds. (The light blinks red slowly in bind mode, solid red when not connected, quick blinking red when not connected for more than a couple seconds, green when connected)

Here is an internal photo of the ECM (it has a waterproof coating from the factory).
[image]

The FCC link - https://fccid.io/XVE-TRX1039

Thank you!

pascallanger, 2024-09-06:
Upgrade your module to the latest test version and retry: https://downloads.multi-module.org/latest-test/

inventor7777, 2024-09-06:
@pascallanger which file do I update to? I believe I read about the firmware files being split up into different parts, so which one do I use for my radio to keep my current protocols? Thanks!

inventor7777, 2024-09-06:
Or how do I find out which protocols are included in each version?

inventor7777, 2024-09-06:
I currently use a mixture of RadioLink Surface, AFDHS2A, and FrSky D8. (And Traxxas, obviously)  Are those all included in the Surface build?

pascallanger, 2024-09-06:
Try it and you will see what the radio display

inventor7777, 2024-09-06:
@pascallanger The ECM works perfectly with the surface build! And all my used protocols are all included which is great 😊 

Thank you!

closed 2024-09-06 by inventor7777