pascallanger/DIY-Multiprotocol-TX-Module #1014: AFHDS3 protocol, and TX/RX donation, how?

zelogik, 2024-09-06:
Hello,

I know that it's a old issue, but I try...

@pascallanger, I have two noble nb4, and some FGr4s, and many "clone" from dasmikro (not available anymore).
A working RadioMaster MT12 with 4in1 and EdgeTX could be fantastic with a AFHDS3 protocol.

Are you interested, and could be considered like a "bounty ;)" if I offer to you a noble nb4 and a original FGr4s to check/implement AFHDS3?
Ie: If it's compatible with a multiprotocol TX, you can keep both TX and RX :-D
Ie2: Just find a way to send to you (ie: I'm from France)

Edit: I just seen a post about DSM (surface), I have an old spektrum dx3r + receiver (SR3100/SR300 clone and maybe original), I can put in the box... #972 

below pictures of AFHDS3 compatible module from dasmikro:
[image]
[image]

Regards

_Originally posted by @zelogik in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/231#issuecomment-2329157713_

zelogik, 2024-09-06:
Could you send me an email to zelogik.spam "sur" gmail, because yearly user without account on rcgroups and need 10msg minimum to be able to send/read PMs :D
In french: Je te répondrais sur ma "vraie" email. merci

zelogik, 2024-09-06:
A quick (but this time effective) research on internet, and seem like the rx/tx is based on the Semtech SX1280, https://www.farnell.com/datasheets/2720464.pdf

I don't think the 4-in-1 module is compatible... but i'm not expert here.

zelogik, 2024-09-06:
Image from mouser:
[image]

Look like exactly the same than all afhds3 receiver, and could you confirm or not, the compatibility with CYRF6936/CC2500/A7105/NRF24L01.

I could send you the noble nb4, RX/TX, but i think there is no hope :-/

zelogik, 2024-09-06:
And now... ELRS is based on that transceiver? ...
[image]

pascallanger, 2024-09-06:
It depends which feature they have turned on. So until I have the hardware I can't tell anything.

zelogik, 2024-09-06:
So check your mail :-D

tmur11, 2025-11-16:
Was there ever any progress on this?

Gavitron, 2025-12-28:
I am hoping to learn if @pascallanger ever received an NB4 from @zelogik, but not for AFHDS3 support, but instead, for learning if the NB4 can support the multiprotocol tool?  there is an OEM NB4 module for FHSS support, and the FCC Internal photos make it appear to have an internal USART:  https://fcc.report/FCC-ID/N4ZRM005/5347675  I have an NB4, but not the RM005 module, and would love to try to get that port working with my multiprotocol devices just like they do on my Taranis.