pascallanger/DIY-Multiprotocol-TX-Module #1015: carisma protocol, is it possible?

gr3yh47, 2024-09-09:
i have a radiomaster mt12 multiprotocol that i want to bind to a carisma gt24. is it possible? i cannot determine protocol

attached is a picture of the chip. below are links to the (quite useless) product pages for the aio i have, a standalone rx with the same protocol, and the tx that the car comes with. 

TX: https://carisma-shop.com/products/kd-propo-ctx-2000-2-4ghz-wheel-transmitter?_pos=1&_sid=404a3a6a2&_ss=r#
AIO: https://carisma-shop.com/products/gt24-esc-rx-module-n-version?_pos=3&_sid=2272fdc39&_ss=r#
Same protocol standalone rx: https://carisma-shop.com/products/sca-1e-receiver?_pos=5&_sid=2083c9a22&_ss=r#

edit: someone found these links, they think regarding the same chip. maybe it helps?
https://github.com/jnk0le/RFM7x-lib
https://docs.ai-thinker.com/_media/2.4g/docs/bk2425_datasheet_v1.3.pdf

gr3yh47, 2024-09-10:
@pascallanger how can i do spi dumps? i need specialized equipment? I'm a developer/techie/fpv hobbiest so i have a skillset but i've never really messed with anything as low level as the protocol itself. 

also i have a transmitter and aio (with blown FET but the rx still works) that i could send you depending on how expensive it is to ship, want to direct msg me your shipping address and i'll see what it would cost me?

tonyai320, 2024-12-24:
Thanks for asking that I encounter same problem

pascallanger, 2025-02-26:
Let me know if you are still interested.

pascallanger, 2025-02-28:
Purchase a $10 8 channel analyzer on ebay, solder it on the spi bus of the rf chip and do some captures. That's what required, if you can't I need a TX + RX set...

gr3yh47, 2025-03-01:
can you ping me in may? i have a bit of a maelstrom in various life situations right now. i hope to have more bandwidth in a couple months and would be happy to do this.

wojciechgawel, 2025-04-20:
Hello! I have like 3 transmitters and would like to get rid of them and drive the cars on 4in1 module if possible. I can send one of transmitters but need the address. Otherwise I need some more instructions where to solder in and what analyser to use. I have never done this before. I can send photos of the PCBA.

pascallanger, 2025-04-22:
The transmitter alone is not enough, I need a RX or model to go with it.

wojciechgawel, 2025-04-25:
Binding and so on...
OK - I can send both - It is integrated RX+ESC plus 4 wire servo controller.
How can I send it ?