pascallanger/DIY-Multiprotocol-TX-Module #1016: Frsky  D8 and Graupner Hott not works on Ethos 1.5.15

fdapport, 2024-09-12:
Module hardware: IRX4 Lite modulo 4-In-1 2.4G Multi
Firmware Module: mm-stm-serial-reta-air-v1.3.4.0
Radio hardware: Frsky X20S
Firmware radio: Ethos 1.5.15 

it work great  with Hitec receivers and V761 receiver. 
I am not able to set Frsky D8  receiver ( V8FR-II) and Graupner receiver (GR16), see attached picture.
What protocol I need to select? I have try Frsky D8 and Graupner Hott but not works. Could please help to set it. Thanks, 
Franco

[image]

pascallanger, 2024-09-12:
Both receivers should work well with the latest firmware and these protocols:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#hott---57
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#frskyv---25
No idea how Ethos works,..

fdapport, 2024-09-14:
I have try as well with a taranis X9d+ 2019 and the Miltiporotocol module, Openx Tx version 2.3.15, multiprotocol version 1.3.4.0. same result, both receivers NOT Working with frskD8 and Hott protocol. Could somebody help me. Thanks

[image]

[image]
[image]

pascallanger, 2024-09-14:
Have you done the frequency tuning as indicated in the protocols ? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#frskyd---3

fdapport, 2024-09-14:
I have try a couple of values with any signal/connection,  if you thinks this could resolve the problem I need to do more values and see if a could bind it.

pascallanger, 2024-09-14:
As indicated, you need to try to bind with 0, -40, +40.
After I don't know your frsky Rx , since it's a V8 so I would have say it's frsky v protocol but...

pascallanger, 2024-09-15:
Also, what are the module LEDs doing before, during and after bind?

fdapport, 2024-09-17:
Finaly I the V8FR-II works with Frsky-V protocol as sugested!!! Thanks... Hott is not working also with fine tune procedure , do you know if I ned to load some LUA script to alow Hott works? Please let me know...

pascallanger, 2024-09-17:
There is no lua to run for the bind. Hott should work, try different values of freq tunning and try at different distances from touching the antennas to a couple of meters. The RX listens to the tx and the tx to Rx.

fdapport, 2024-09-17:
what about the receiver firmware, could be relevant for the bind process? Thanks for your help.

pascallanger, 2025-02-26:
ETHOS issue, nothing to do with Multi.

closed 2025-02-26 by pascallanger