pascallanger/DIY-Multiprotocol-TX-Module #1017: Flysky AFHDS2A RX fails to bind with an original Flysky TX

MirlynG, 2024-09-23:
I have a 4 in 1 RM TX16s and an RM Pocket with  a 4 in 1 module. I have set up the wireless trainer facility successfully in both directions i.e TX16s to Pocket and Pocket to TX16s by swapping the 4 in 1 module around. Therefore, I know it works and the setup is right.
I have now been trying to bind 4 different Flysky AFHDS-2A 2/3 channel surface/pistol grip transmitters as the student radio for days, but have found it impossible. I have tried every combination of powering on and pressing bind buttons but have never achieved a bind.

The following is copied from the FlySky AFHDS-2A RX protocol page:
"Uses the MULTI-Module as a DSM receiver instead of a transmitter.  Compatible with any radio transmitting AFHDS 2A"
Apart from the obvious incorrect reference to DSM, I am wondering if in fact this not right.

I have found a closed issue going back to January 2021, and tried two earlier versions of  the firmware going back to 1.3.2.86 (1.3.2.8 was suggested in the post but is not available) Unfortunately, no joy. i am beginning to think that this will only work with an FS

I have a disabled grandson who loves the rc cars and boats I have, but cannot use a stick radio. He is ok with a pistol grip but keeps crashing them!! Hence I would love to be able to take over before he wrecks any more. What are the possibilites of getting this to work with the likes of an FS-GT2E transmitter which is an AFHDS-2A unit, or am I just doing something wrong?

Be great to hear back from you. Thanks

MirlynG, 2024-09-24:
Would a DX3 work in auto mode despite being dsmr?

pagrey, 2024-09-29:
I tested using a Flysky FS-i6 using AFHDS2A and an iRX4+ module running FS2A_RX and that works fine. I set the FS-i6 in bind mode then selected bind on the EdgeTX radio and it immediately connected. Be sure the radios are not too close. Unfortunately, I only have an FS-GT2B which use AFHDS so I can't test with the exact same radio.

MirlynG, 2024-10-05:
I think the issue lies in the fact that the air and surface protocols are in fact different, despite both being AFHDS 2a. I don't think there is going to be a reolution to this any time soon. Thanks for the input nonetheless.

pascallanger, 2025-02-26:
I think this is the new Flysky AFHDS2A BS protocol which is not supported yet and has its own issue already.

closed 2025-02-26 by pascallanger