pascallanger/DIY-Multiprotocol-TX-Module #1018: add sub protocol "J20" to "SGF22" protocol

rdba2k, 2024-09-25:
The subprotocol is for this airplane, KF700 J20.

https://www.aliexpress.us/item/3256807449221710.html?gatewayAdapt=vnm2usa4itemAdapt

I test flight the code and it is verified. Attached are the codes.

I would like have some changes in this web page

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#SGF22---97

Add the following for clarity: 

SGF22 & F22S: Mode -100% = 3D, 0% = 6G, 100% = Vertical 
J20: Mode -100% = Gyro off, 0% = Horizontal, 100% = Vertical. CH8 - Invert

Thanks.

[_Config.h.txt](https://github.com/user-attachments/files/17123670/_Config.h.txt)
[Multi.txt](https://github.com/user-attachments/files/17123671/Multi.txt)
[Multi_Protos.ino.txt](https://github.com/user-attachments/files/17123672/Multi_Protos.ino.txt)
[Multiprotocol.h.txt](https://github.com/user-attachments/files/17123673/Multiprotocol.h.txt)
[SGF22_nrf24l01.ino.txt](https://github.com/user-attachments/files/17123674/SGF22_nrf24l01.ino.txt)

pascallanger, 2024-09-26:
Merged into the main code. Please test to see if everything works as expected.
Thanks

closed 2024-09-26 by pascallanger

rdba2k, 2024-09-26:
Test flight of all 3 models under SGF22, all work as expected.

Sorry still some minor change in 
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#SGF22---97

Take out vertical mode from ParkTen F22S since it only has 3D/6G for this model, such as

Sub_protocol F22S
Model: ParkTen F22S
F22S: Mode -100% = 3D, 0% = 6G

pascallanger, 2024-09-27:
Done