pascallanger/DIY-Multiprotocol-TX-Module #1026: FrSky XM+ X2 not binding with latest firmware

frankiearzu, 2024-09-28:
RX:  FrSky XM+ FCC 2.1.2
TX: TX16 MM v1.3.4.0
It does not bind with FrSky X2 protocol.

It was corrected in #543, but broke on later releases.

==========================================
Works Correctly after that fix:  #543     MM Firmware: 1.3.2.77
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commit/04d08c6d67ca42ef2365a0b5568bc33e17954cef

This broke it again when adding ARCHER Bind:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commit/a41123deb26fa29c1b316be28c203d1abe05f6af

ParkerEde, 2024-10-02:
there is already an issue:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/987

frankiearzu, 2024-10-03:
Clossing it since is a duplicate

closed 2024-10-03 by frankiearzu