pascallanger/DIY-Multiprotocol-TX-Module #1035: Use Protronik ptr 6a as trainer by RX mode

pierrotm777, 2024-10-06:
Hello,
I should like use a Protronik ptr 6a transmitter as trainer solution because this transmitter accept modes 1 to 2 by switch a button on the bottom side. It's protocol is FHSS.
Is it possible to add it, or this protocol isn't compatible with the mpm system?
Thanks,
Pierre

pascallanger, 2024-10-09:
I need a set of TX + RX to be sent to my attention for reverse.

pierrotm777, 2024-10-09:
Ok, I see that ASAP.

pascallanger, 2025-02-26:
Looks like it won't happen.

closed 2025-02-26 by pascallanger

pierrotm777, 2025-02-26:
Sorry, I haven't bought this handset.
I buy one and send It as soon as possible .

pierrotm777, 2025-02-28:
Hello Pascal, I shall receive one Tx Rx Ptr6a in March. Can you give me one address?

pierrotm777, 2025-03-01:
Hello Pascal, I have received my Protronik handset. How can I send you it ?