pascallanger/DIY-Multiprotocol-TX-Module #1042: QIDI 560 LED Channel issue

Solman79, 2024-11-13:
Hi, ive downloaded the latest test version and i have bound the QIDI 560 to the Q560 protocol. I cant seem to get the LED channel to work. On the original TX you held the button in for 3 seconds to turn the LEDs off, but im struggling to figure out how to get this to work on my Jumper T Lite

pascallanger, 2024-11-13:
Are you using the correct channels:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-q560---3

Solman79, 2024-11-13:
Hi, yes, I used that data from that exact page to set it up. I feel as
though it is something really simple. I've tried putting that channel on
different switch types, but still can't get the LEDs to turn off.

Solman79, 2024-11-13:
I tried it again tonight and created a new model just to be sure, but still nothing using Channel 7 listed under Q560. I tried some other channels as well, but nothing seemed to control the LEDs.

Solman79, 2024-11-15:
> Are you using the correct channels: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-q560---3

Tried again tonight, bound it to the original transmitter and it works, still can't get it to work on the MPM transmitter using channel 7

pascallanger, 2025-01-19:
Jus tried on my QIDI 560, assigning a switch to CH7 enables me to turn on/off the lights.

lazyzero, 2025-01-22:
I tried today to get the LEDs work with version 1.3.4.20 with switch that goes from -100 to 100 but the lights are not going off.

Solman79, 2025-01-22:
Yes, I've still got that same issue as well. I'm on the latest test build
and cannot get it to work on a -100 to 100 switch either.

pascallanger, 2025-01-23:
My guess is that they have finally seen their bug that the light flag was overwriting the trims and corrected it in a new version of the protocol.
None of you have an external module by any chance?
Anyway, I've modified the code to set an unused flag to see if that would trigger the light:  [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.21
Can you try CH7 and report?

closed 2025-01-23 by pascallanger