pascallanger/DIY-Multiprotocol-TX-Module #1043: in Wizard for WING: Switch picked for dual rates is not shown in summary screen or programmed in model

ror1948, 2024-11-13:
I have updated the program in the Lua file "3.Wing.lua" attached so this fault does not occur.

Changes below:

-- WAS at line 452
-- updateInputLine(defaultChannel_0_AIL, 0, expoVal, 100, "SC" .. CHAR_UP)
-- updateInputLine(defaultChannel_0_AIL, 1, expoVal, 75 , "SC-")
-- updateInputLine(defaultChannel_0_AIL, 2, expoVal, 50 , "SC" .. CHAR_DOWN)
-- updateInputLine(defaultChannel_0_ELE, 0, expoVal, 100, "SC" .. CHAR_UP)
-- updateInputLine(defaultChannel_0_ELE, 1, expoVal, 75 , "SC-")
-- updateInputLine(defaultChannel_0_ELE, 2, expoVal, 50 , "SC" .. CHAR_DOWN)

-- IS at line 452
updateInputLine(defaultChannel_0_AIL, 0, expoVal, 100 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_UP)
updateInputLine(defaultChannel_0_AIL, 1, expoVal, 75 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value].."-")
updateInputLine(defaultChannel_0_AIL, 2, expoVal, 50 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_DOWN)
updateInputLine(defaultChannel_0_ELE, 0, expoVal, 100 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_UP)
updateInputLine(defaultChannel_0_ELE, 1, expoVal, 75 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value].."-")
updateInputLine(defaultChannel_0_ELE, 2, expoVal, 50 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_DOWN)

-- added these lines
at line 385
--dual rate sw
if ElevronFields.is_dual_rate.value == 1 then
drawNextLine("Dual Rate SW", nil, ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value])
end

[updated_3.Wing.zip](https://github.com/user-attachments/files/17733297/updated_3.Wing.zip)

pascallanger, 2024-11-13:
Not sure what it has to do with Multi...

ror1948, 2024-11-13:
sorry please ignore

closed 2024-11-13 by ror1948