pascallanger/DIY-Multiprotocol-TX-Module #1046: Servo is not working properly with SLT2

AlexGigov, 2024-12-13:
Hello, 
I have a strange servo behaviour on Losi Micro-B using Radiomaster Mt12 with an external Radiomaster 4in1 module with 1.3.4.12 test build firmware. The RX and TX bind together, but the servo is not smooth. The best I can describe it is as if the packet rate is not constant and there are micro cuts during the movement. I do not see a problem with the throttle. So far I have tried to change the inputs, frequency tuning and tested with different channel order but the issue persists.
With the original Spectrum SLT2 TX everything works smoothly.

Any idea what could solve the issue? 

Thanks

Pencilvestr, 2025-01-01:
I think someone needs to send in an SLT2 RX/TX to fix these issues.

You can look at the original post I made for SLT2. I think its as good as it gets for the time being until someone donates

CDRaff, 2025-01-30:
How can I send one? Would you like one of the newer orange button ones that controls DSC too?

pascallanger, 2025-01-30:
Which protocol are you using for SLT2? There is no SLT2 protocol... Are you talking about the protocol SLT and subprotocol V1_4CH?
If that's the case, I need an original TX + RX (or car or whatever) to be sent to my attention in France.

pascallanger, 2025-02-21:
I could purchase a car set to get to the bottom of this issue but I need a little help as they all seem quite expensive...

pascallanger, 2025-02-21:
So if you are ok to do some donation(s) let me know.

Gaz-B, 2025-02-24:
> I’ve encountered the exact same problem with my Mojave GROM, which comes with a 2-in-1 receiver with ‘SLT’ symbol on it. My MT 12 tx successfully bound with the receiver with ‘SLT’-‘V1_4CH’ (fw v1.3.4.31 and only successful with this protocol), but the steering is not as smooth as when it was bound with the original tx. Hope this issue could be solved in the future. Thanks a lot.

I don't know if you've tried but the Gyro DSC ?. Can you adjust this with the MT12 ?.

Gaz-B, 2025-02-24:
> I could purchase a car set to get to the bottom of this issue but I need a little help as they all seem quite expensive...

There are some TX RX combos removed from cars at quite reasonable prices. Would having one of these deffo get the issue resolved. https://www.ebay.co.uk/itm/276839217240?_trkparms=amclksrc%3DITM%26aid%3D777008%26algo%3DPERSONAL.TOPIC%26ao%3D1%26asc%3D20240131095853%26meid%3D4e19c61b902d4371940bfe19f82cd9d0%26pid%3D101949%26rk%3D1%26rkt%3D1%26itm%3D276839217240%26pmt%3D0%26noa%3D1%26pg%3D4375194%26algv%3DRecentlyViewedItemsV2WithMLRPboosterAndUpdatedFeatures_BP%26brand%3DSpektrum&_trksid=p4375194.c101949.m162918&_trkparms=parentrq%3A395b4d521950a6f110e5030cfffff0de%7Cpageci%3A920f46c4-f2e2-11ef-8a13-ba9f12008757%7Ciid%3A1%7Cvlpname%3Avlp_homepage

Gaz-B, 2025-02-24:
> I’ve encountered the exact same problem with my Mojave GROM, which comes with a 2-in-1 receiver with ‘SLT’ symbol on it. My MT 12 tx successfully bound with the receiver with ‘SLT’-‘V1_4CH’ (fw [...]

Now you have it working are you able to adjust the gyro DSC ?.

Pencilvestr, 2025-02-24:
I'll donate for a car / just the RX/TX set if either will work. Let me know.

pascallanger, 2025-02-24:
I've found a TX + Rx set for 30€ + 5€ shipping on the local second hand market: https://www.leboncoin.fr/ad/modelisme/2910782924
Are you ok to donate for the purchase?

What does gyro DSC mean? Is that something you can set from the radio?

Pencilvestr, 2025-02-24:
Donated 15usd, hopefully someone else can help cover the rest

Gaz-B, 2025-02-24:
DSC. Digital Stability Control. You alter the gyro gain by putting full right steering and then pressing the TX power button. I don't get my car until tomorrow so can start playing then but as it's only a 2ch radio I'm starting to wonder if getting full functions with another radio is even achievable.

Gaz-B, 2025-02-24:
This is the full manual to the Brushed unit in your link they don't seen to do one for the brushless but the radio function DSC ETC all seem to be identical. https://www.horizonhobby.com/on/demandware.static/-/Sites-horizon-master/default/dw8aae40b5/Manuals/SPMXSER1025_MANUAL_EN.pdf
Spektrum Firma 25A Brushed Smart ESC Spektrum Firma 25A gebürsteter - Horizon Hobby<https://www.horizonhobby.com/on/demandware.static/-/Sites-horizon-master/default/dw8aae40b5/Manuals/SPMXSER1025_MANUAL_EN.pdf>
4 Waterproof Component Your new Horizon Hobby® Spektrum™ 25A Brushed Waterproof 2-in-1 ESC and Dual Protocol Receiver has been designed and built to allow you to operate the product in many “wet conditions”,
www.horizonhobby.com

yinglong, 2025-02-25:
> Now you have it working are you able to adjust the gyro DSC ?
Haven’t dug into that yet. But my guess is that the dsc should be implemented as a dedicated channel(CH 3?).

Gaz-B, 2025-02-25:
[yinglong] The Transmitter and receiver is two channel only. This I believe is why changing DSC profiles is done with full lock and pressing the power button. I was expecting the same as you then started reading in more detail.

pascallanger, 2025-02-25:
I don't see anything about DSC in the documentation you linked... I'm not even sure which RX/ESC I'm going to get if I purchase the set I linked earlier...

Gaz-B, 2025-02-25:
> I don't see anything about DSC in the documentation you linked... I'm not even sure which RX/ESC I'm going to get if I purchase the set I linked earlier...

My mistake that seems to be a Brushless only feature. I should have sent this manual.  https://www.arrma-rc.com/on/demandware.static/-/Sites-horizon-master/default/dw4bd747f2/Manuals/ARA2306_MANUAL_ML.pdf

pascallanger, 2025-02-25:
Can you try the latest [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.36?

pascallanger, 2025-02-25:
They indicate that the SLT2 transmitter is different being SLT2 DSC on this brushless version. So no idea what changes they have made...
The TX sends 4 channels even if they say 2. So may be the gyro mode is on one of the other 2 channels. To be tested.

Gaz-B, 2025-02-25:
> Can you try the latest [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.36?

😁😁👌👍👏 Great success Held the power button for 3 secs and instant auto bind. Steering seems smooth I never saw what the previous versions were like as I only just picked up the car new and fitted a battery plug. DSC is on ch4 I've put it on a knob ranging -35 off to 15 full.  It seems the stock DSC command of right steering and power press sets the output value of CH4 as it can be done with the car off.

pascallanger, 2025-02-25:
Can you test with the latest release to compare ? If everything runs smooth on 1.3.4.36, the issue is really the TX ID vs RF frequencies. In this version you are locked down to only the ID of @Pencilvestr 's radio. Let's try to see if we can get more than 1 ID working to prevent accidents of controlling other cars...

Gaz-B, 2025-02-25:
If I use 1.3.4.31 With my ID 50 17 6F 22 No bind with ID 65 50 FE 30 it binds but choppy steering. With 1.3.4.37 with my original ID 50 17 6F 22  still no bind.

CDRaff, 2025-02-25:
Feels rock solid to me now. I was on the 1.3.4.25 prerelease and it felt kinda wonky, a little jittery and a little laggy; now it feels rock solid. Tested on the SPMXSE2625RX found in the brushed Mojave Grom, the SPMXSBER1025G found in the brushless Typhon Grom BLX, the SPMXSE4510RX found in the Fury 223s, and the SPMXSE2825RX found in the Losi Micro-B.

@Gaz-B great job sussing out the DSC settings. I had been goofing with it, and found that little spot on channel 4 that would enable it, but I was too dumb to think to change the channel min/max to fine tune it. Works on both the SPMXSBER1025G and the SPMXSE4510RX. 

Also will be donating this coming week; we had previously discussed me trying to get a esc and receiver to you, but my wife got sick with a gallbladder thing so it side tracked me for a bit, but we're all back in order now and I'll be able to send a chunk of change off to you to help you get some gear. Thanks a ton for all the hard work you are putting in.

pascallanger, 2025-02-25:
@CDRaff Can you try all the receiver numbers from 0 to 63 with this new [test version ](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip)1.3.4.37 and let me know which ones are working or not?
You will need to rebind for each receiver number, check the LED for flickering and steering smoothness.

CDRaff, 2025-02-25:
Just edited my comment, but it was based off the 1.3.4.36 not .37. Will test .37 and will expand beyond the first few receiver numbers. When testing on .36 I had gone to receiver 04 and found that they were all working no matter what receiver I was using.

CDRaff, 2025-02-25:
Ok tested receivers 00-60 on the SPMXSE2825RX(Losi Micro-B) and on the SPMXSBER1025G(Typhon Grom BLX 223s) and both work well on every receiver. Every once in a while I would hit a receiver number that would initially act wonky when first bound, but power cycling the car fixed it every time so that may have just been caused because I was binding over and over again. 

I'll test the SPMXSE2625RX and the SPMXSE4510RX tonight when I get some free time.

pascallanger, 2025-02-25:
That looks promising! I think it will work fine on the other receivers. Instead, can you try the receiver numbers on v1.3.4.38 ?

CDRaff, 2025-02-25:
Yes I can. As soon as I have some free time I'll jump on it.

Gaz-B, 2025-02-25:
I just tried all 60 the ones that weren't happy were 14,25,36,57, Very sluggish and jittery steering.

pascallanger, 2025-02-25:
> I just tried all 60 the ones that weren't happy were 14,25,36,57, Very sluggish and jittery steering.

Which version ? Have you power cycled the model like @CDRaff indicated earlier?

pascallanger, 2025-02-25:
Great 👍 Can you try the receiver numbers on v1.3.4.38?

Gaz-B, 2025-02-25:
> [@Gaz-B](https://github.com/Gaz-B) Great 👍 Can you try the receiver numbers on v1.3.4.38?

What exactly am I looking for as on retest 37 seems faultless.

Gaz-B, 2025-02-25:
v1.3.4.38 All 60 working a couple stumbled on first bind power cycle sorted them. I did notice on some flickered the blue led for a couple of secs then sorted themselves before I could try a power cycle. Stumbles were on 25,28,30,44.

pascallanger, 2025-02-25:
> > [@Gaz-B](https://github.com/Gaz-B) Great 👍 Can you try the receiver numbers on v1.3.4.38?
> 
> What exactly am I looking for as on retest 37 seems faultless.

I'm trying to figure out the IDs versus frequencies. Right now, you've confirmed that I can generate 64 good pairs. I usually don't stop there and try to get more. If I implement only these 64 IDs, you have 1 chance in 64 to have a conflict if someone uses a multi module nearby and that gets worse if there are more of them around you.
I don't have a RX so I can't test myself... I usually spend endless hours trying this kind of things so that you don't have to worry about conflicts. Here I'm trying to reduce the number of tests by changing some parameters which I think would work but I need the confirmation.
But if you think that's useless, I stop here. I don't have a SLT2 receiver and probably won't own one so up to you.

CDRaff, 2025-02-25:
Tested 0-63 on firmware 1.3.4.3.8 on both the SPMXSE2825RX(Losi Micro-B) and the SPMXSBER1025G(Typhon Grom BLX 223s) and both were working well with all errors being sorted after power cycling. There was a bit of a longer hangup on receiver 19 when binding to the SPMXSBER1025G, but it eventually sorted itself out.

Gaz-B, 2025-02-25:
I don't mind testing it's just I couldn't see any change so had no idea if it was helping. I'm so used to having to bind the Losi promoto with every battery this now seems well sorted 🤣.

pascallanger, 2025-02-25:
Ok hopefully last try with v1.3.4.39 which is using partly your module ID. If you both @CDRaff and @Gaz-B and may be others try once more receiver numbers between 0 to 63. If it works I call it done.

The Losi Promoto is on my list to address soon as I do have access to one nearby.

pascallanger, 2025-02-25:
@Gaz-B, I would like to update the documentation with your findings on the DSC settings.
Do you have the values for the different modes? Or should I just indicate "CH4 is used for DSC settings: -35% off to +15% full"

Gaz-B, 2025-02-25:
> [@Gaz-B](https://github.com/Gaz-B), I would like to update the documentation with your findings on the DSC settings. Do you have the values for the different modes? Or should I just indicate "CH4 is used for DSC settings: -35% off to +15% full"

Those are the figures that appear to be right. Not easy to be sure but that seems to cover off to full gyro action. I have no means to measure what the CH4 output pulses are on the stock radio.

Gaz-B, 2025-02-25:
> The Losi Promoto is on my list to address soon as I do have access to one nearby. That is exciting news I know a few are doing like me and rebinding with every battery.

pascallanger, 2025-02-25:
Oh you mean that it's proportional from -35% to 15% ? The documentation you shared earlier kind of indicate that there was only 4/5 steps.

Gaz-B, 2025-02-25:
> Oh you mean that it's proportional from -35% to 15% ? The documentation you shared earlier kind of indicate that there was only 4/5 steps.

Yes the stock radio is changed by full right steering and push the power button with with each press cycles from 1 to 4 beeps. 1 being off and 4 being max. My guess is they set  ch4 to preset outputs the receiver doesn't even need to be on. When the car is linked it just sees the set position.

Gaz-B, 2025-02-26:
1.3.4.39 all 63 slots working three needed a power cycle and were fine after.

Gaz-B, 2025-02-26:
> Tested all 64 receivers on v1.3.4.39 on both SPMXSE2825RX(Losi Micro-B) and SPMXSBER1025G(Typhon Grom BLX 223s) and everything is still working fine.
> 
> Also as for the DSC control note I think what your planned "CH4 is used for DSC settings: -35% off to +15% full" is good. Maybe include the pulse width length in microseconds(from 1321µs to 1577µs) as well?

Do those figures seem right on your BLX grom too ?. I only picked mine up today so this testing is all it's done so far.

CDRaff, 2025-02-26:
Yes they seem like the right numbers on both the Typhon Grom 223s and the Fury 223s.

pascallanger, 2025-02-26:
Ok, I've cleaned up the code and updated the documentation. Let's close this issue.

closed 2025-02-26 by pascallanger

tschaerni, 2026-08-29:
It seems this issue is resolved, but I just bound my Radiomaster 4in1 module (running the latest stable version v1.3.4.63) to one of my Arrma Mini Kratons (SPMXSBER1040G) and I also get jittery servo movement (even once hat jittery throttle, were the motor would pulse).
I've tested about 4-5 different receiver IDs and even tested it on a different Mini Kraton. Even playing around with frequency tuning didn't show any improvement. Same goes with powercycling.
Smooth servo movement with the SLT2 transmitter, but jittery with the 4in1 module (as someone here already mentioned, it feels like package loss)

I also tested it on my Arrma Mojave Grom 223S BLX (SPMXSBER1025G) and I've observed the same behaviour.

Any ideas?