pascallanger/DIY-Multiprotocol-TX-Module #1048: New protocol for JJRC H36 (yet another)

Tikju, 2024-12-18:
Hello Pascal,

I just purchased a brand new JJRC H36 from Aliexpress and was hoping to bind it using the IRangeX IRX4+ with either the MJQx -> E010 or the E010 R5 protocols. Unfortunately, neither of them worked (despite trying different frequency fine tuning settings). I am thus assuming that the current version of the H36 comes with yet another new procotol that is not supported by Multi Module, yet. The board on my drone reads JR-NH010R9.

I also found this discussion on rcgroups that suggests the same: [https://www.rcgroups.com/forums/showthread.php?4554145-New-version-of-the-JJRC-H36-%28E010-or-NH010%29](url)

Would you be willing to attempt reverse-engineering this new protocol? What kind of information / support would you need from my side!

Thanks a lot in advance and best regards!
Tobias

Tikju, 2025-01-19:
[H36_dump.txt](https://github.com/user-attachments/files/18468178/H36_dump.txt)

Small update: I just made an XN297 Dump with the IRX4+ (see attachment). Would this be sufficient for figuring out the protocol?

pascallanger, 2025-01-19:
This is a good start but I need more dumps to try to understand this protocol.
First use the latest XN297Dump protocol version that you can find in the test build https://downloads.multi-module.org/latest-test/
Please run the following dumps in Auto mode:
 - Capture the original TX in bind mode
 - Capture the original TX in normal mode
   - at the end when identify sticks appear, move slowly (really slowly) the following channels A E T R and then press the feature buttons in a specific order that you'll write down and paste here.

Tikju, 2025-01-19:
Thanks a lot for these instructions!

Here are the logs for the different channels and modes:
[H36 Dump.zip](https://github.com/user-attachments/files/18470265/H36.Dump.zip)

If something is missing or I can test further things, let me know!

Tikju, 2025-01-27:
Cool, thanks for looking into it!

You need to go from 0 to +100% throttle (TX beeps once) then to -100% (beeps again, drone flashes to indicate that it's bound).

pascallanger, 2025-01-27:
XN297, address length=5, bitrate=1M, Scrambled

Bind
---
RF channel: 50
Timing: 2065µs,3085µs,10285µs...
Address: CC 6C 47 90 53

Packet
P(13): 2E BC 00 11 00 36 3A 31 2B 00 00 00 ED
P[0] = 2E ? TXID or just start?
P[1] = most likely a crc
P[2] = TXID[0]?
P[3] = TXID[1]?
P[4] = TXID[2]?
P[5] = RF[0]=54
P[6] = RF[1]=58
P[7] = RF[2]=49
P[8] = RF[3]=43
P[9] = 00 ?
P[10] = 00 ?
P[11] = 00 ?
P[12] = ED ? bind ?

Normal
---
RF channel: 43, 49, 54, 58
Timing:
- each channel is sent 4 times 1830µs,3085µs,3085µs
- switch from one channel to the next 18500µs, so 10500µs between end of channel n and channel n+1

Address: CC 6C 47 90 53 => same as bind

Packet
P(13): 2E 54 00 11 00 7E 80 80 80 00 20 20 60
P[0] = 2E ? TXID or just start?
P[1] = most likely a crc
P[2] = TXID[0]?
P[3] = TXID[1]?
P[4] = TXID[2]?
P[5] =  
P[6] = 
P[7] = 
P[8] = 
P[9] = 
P[10] = 
P[11] = Trim E 02..20..3E
P[12] =

pascallanger, 2025-01-27:
I'm confused by how you name the sticks...
Left<->right is aileron, correct?
Azimuth is rudder, correct?
Forward<->backward is elevator, correct?

Tikju, 2025-01-27:
Yes, right. I didn't know how they are typically called for drones. :)

pascallanger, 2025-01-27:
I'm trying to figure out the crc right now...

pascallanger, 2025-01-28:
Ok I've got the crc figured out... Took me a lot of hours (4+) for something in the end relatively easy...
From there I can code the protocol.

pascallanger, 2025-01-28:
Please try the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) 1.3.4.25 protocol H36, documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#h36---103

If you set the receiver number to 0 (default) it will use your original TX ID as a test to make sure everything works fine (some channels might be reversed, I didn't pay attention...). But if you increase the RX number it will use your radio ID.
Can you try different values of RX number and see if it works?

Tikju, 2025-01-28:
Wow, good work! Everything works absolutely perfectly out of the box. I tried a few different RX numbers between 0 and 63 and was able to rebind with all of them.

Also, all the channels are correct, I did have to reverse anything. Regarding the speed mode, I am assuming that it always uses the faster one now and I could reduce the channel weights myself in Edge TX if I wanted the drone to be slower, right?

In any case: Thanks a lot for your effort!

pascallanger, 2025-01-28:
> Regarding the speed mode, I am assuming that it always uses the faster one now and I could reduce the channel weights myself in Edge TX if I wanted the drone to be slower, right?

Yes... I usually set the rate to high with all the models since you can control everything on the radio side like adding expo or dual rate which is better than just low rate.

> Thanks a lot for your effort!

If you appreciate my work and want to offer me a beer, here is the page: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Donations.md

closed 2025-02-26 by pascallanger