pascallanger/DIY-Multiprotocol-TX-Module #1049: bind irx4 lite with multiplex m-link receiver

Franz-Royer, 2024-12-19:
I can't bind an rx7dr light m_link receiver with the irx4 lite multi.
Firmware ethos 1.5.19 EU on frsky x20.
Firmware irx4 lite is 1.3.4.0
Luascript is installed on the transmitter, latest version.
The module works, I can bind the Spektrum receiver. I can also bind the Goosky s2 without any problems.
I have already reset the M-Link receiver.
no matter how I try it always flashes the same

Can anyone help me?

Lg Franz

pascallanger, 2025-01-19:
I know many people using MLink so it should work...
The radio and RX must be really close for the bind to complete: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#mlink---78

pascallanger, 2025-02-26:
Looks like that it is solved...

closed 2025-02-26 by pascallanger