pascallanger/DIY-Multiprotocol-TX-Module #1050: V761 can

chang-le-z, 2024-12-22:


chang-le-z, 2024-12-22:
Hardware: JUMPER T14
Version : edgetx 2.10.5
Multiprotocol Module Firmware V. 1.3.4.0
Receiver: volantex bf109 v761-11

Hi,
I just found a function called ''FIND MY AIRCRAFT(using beeper to find plane)'' on the latest volantex rc plane.It works fine on the volantex's remote,but can't do it with Multiprotocol Module.And I think i have already selected the correct protocol（v761-ch4）,because the other functions worked very well except the BEEPER =( .

Can anyone help me?

Thanks.=p

pascallanger, 2024-12-26:
How do you activate the better on the original radio? Is there a dedicated button?

chang-le-z, 2024-12-26:
[image]
[image]
This is the picture of the instructions.  Maybe this can help you to understand it.

chang-le-z, 2024-12-26:
It's the same button for the return and the beeper function,but activate the return function don't need to hold for 3 seconds.
I've tried to active the beeper as the same way(press the return button and hold for 3 seconds) on my radio . Didn't work...

pascallanger, 2025-01-01:
I have a friend that has one of these new plane with a beeper. He has thrown away the original radio but may be I'll be able to find the flag by trials... More to come mid/end of next week.

chang-le-z, 2025-01-02:
Okay! Thank you soooooo much!!!!!! =*

pascallanger, 2025-01-07:
Please try the test version v1.3.4.13 https://downloads.multi-module.org/latest-test/ , the beeper is on CH10: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-4ch---1

chang-le-z, 2025-01-08:
Never mind.I just downloaded the file from github.Thank you for your continued help.

chang-le-z, 2025-01-09:
Ok,it works! But I found a new problem. If i use the 4ch mode to enable telemetry,i should choose "Receiver No."→"Range" otherwise the connection will be unstable(the plane often lose control).Toprc has no such problem...

chang-le-z, 2025-01-09:
Just now,I choose the “low power” mode,the problem is also solved.Sounds like the low power mode is “inverted”？

chang-le-z, 2025-01-09:
My fault.The same problem occured on the original radio.Probably plane's reason.

closed 2025-01-09 by chang-le-z