pascallanger/DIY-Multiprotocol-TX-Module #1051: Q560 issue

mkfrcp, 2024-12-30:
Hello, I finally found the time to try version 1.3.4.12 but I still can't bind my TX16S transmitter with the Maule. 
I've also tried another receiver combo but it doesn't work either.
I see that several users have succeeded in binding their transmitter (4in1 module).
Here's the configuration in my transmitter.
So I'm probably missing something. Can anyone help me solve this problem?
What is the precise configuration to be entered in the 4in1 menu?
What is the configuration in the 4in1 menu?
Do I need to move the sticks to bind?

Thanks for your help
[image]

pascallanger, 2024-12-30:
Config looks good... You have to bind manually through the interface since you aren't using bind on channel.
I see that you have a TAER firmware. Have you configured the channels with the same order?

mkfrcp, 2024-12-30:
I also tried with bind on channel but it doesn't work either. I have been
flying my planes for 2 years with this TAER configuration and I have never
encountered a problem with HoTT, spektrum or Multiplex receivers for
example. Is there a particular operation to be carried out? Do you have
another idea that could explain the problem with two Maule's receivers? I
would like to point out that at no time does the LED of the receiver stop
flashing. It's as if the transmitter was turned off while when I turn on
the Maule's transmitter, the binding is done in two seconds.

mkfrcp, 2024-12-31:
I tried to use bind on channel. No success on both receivers. Any idea?

pascallanger, 2024-12-31:
Have you checked on rcgroups to see if there is a new version of the TX/RX for this plane?
Just for a trial, bind in low power mode and see if that changes anything?

mkfrcp, 2024-12-31:
I'll look on rcgroups but after spending so much time, I've come to the same conclusion. There is likely to be a new version of TX/RX.

I've already tried it in low power mode but it made no difference.

Thanks for your help.

closed 2025-01-19 by pascallanger