pascallanger/DIY-Multiprotocol-TX-Module #1054: Eachine EAT15 car support?

yarda, 2025-01-04:
In the recent firmware 1.3.4.0 the EAT14 support was added (XK/cars). Any chance to add support for the quite popular EAT15 car (also known as UDIRC UD1603)?

pascallanger, 2025-01-19:
Without access to one of these cars I can't really do anything...
2 options:
 - you send me the original TX and car
 - you sponsor the purchase of one car

pascallanger, 2025-01-20:
You can use your multi module to scan and possibly listen packets over the air using a debug firmware and the XN297Dump protocol like on this other issue: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1048#issuecomment-2601004104

yarda, 2025-01-20:
Thanks for the tip, I didn't know it.

pascallanger, 2025-01-20:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_XN297Ldump.md

yarda, 2025-02-10:
I had trouble to flash the debug firmware from the release [mm-stm-xn297dump-ftdidebug-v1.3.4.0.bin](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/download/v1.3.4.0/mm-stm-xn297dump-ftdidebug-v1.3.4.0.bin). On the radio it shows error:
```
NEEDS FILE
opentx-inv
```
The other firmwares from the release flashed and worked OK, including the usbdebug variant. Unfortunately, my module doesn't have connected USB lines to the STM32 and there is built-in CP2102 bridge, so I need the FTDI debug firmware variant.

I will try to compile it myself

benlye, 2025-02-10:
@yarda we currently don't have a way to compile an FTDI debug image which can be flashed from the radio. I'm working on it but have a few issues to resolve in the build process first.

For now you have two options:
1. Use the file attached to [this comment](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1069#issuecomment-2642516360), which can be flashed from the radio
2. Assuming you have an external module - it sounds like you do - upload the firmware using the [Flash Multi](https://github.com/benlye/flash-multi) tool

yarda, 2025-02-10:
> [@yarda](https://github.com/yarda) we currently don't have a way to compile an FTDI debug image which can be flashed from the radio. I'm working on it but have a few issues to resolve in the build process first.
> 
I was tempted to flash the ftdi debug firmware from the release through the USB adaptor, but I discarded this idea due to the failed firmware check in the radio. Maybe it could be also mentioned in the readme.

yarda, 2025-02-12:
The protocol seems pretty easy, channel 45, 250k:
```
P: 08 WW TT SS LL 00 00 00 00 GG EE DD PP 00 CC

WW - steering wheel, 00 - min, 64 - idle, C8 - full
TT - throttle/break , 00 - min, 64 - idle, C8 - full
SS - speed mode, 00 - min speed, 64 - middle speed, C8 - full speed
LL - light switch, 00 - idle, C8 - pressed
GG - gyro, 00 - min, C8 - max
EE - steering trim, 00 - min, C8 - max
DD - steering dual rate, 00 - min, C8 - max
PP - reverse switches, 00 - all normal, 40 - throttle reversed, 80 - steering reversed
CC - checksum, just sum bytes 0 - 13

Example packet:
P: 08 64 64 00 00 00 00 00 00 00 34 4F 00 00 53
```

I need to capture the bind sequence,  I had some trouble with it, maybe it is (I am not 100 % sure):
```
250K C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
```

It seems it has also telemetry built-in and probably returns car battery status to the remote, but currently I have no idea how to reverse it.

yarda, 2025-02-12:
I don't know whether it sticks to channel 45 or can autoselect different channel e.g. in case it's used/noisy, I tried several times and every time I got channel 45 (I will also recheck). It seems there is no hopping.

yarda, 2025-02-12:
The model is autobound each time the model and remote is switched on.

yarda, 2025-02-12:
> I need to capture the bind sequence, I had some trouble with it, maybe it is (I am not 100 % sure):
> 
> ```
> 250K C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
> ```
> 
If power on the car (not the TX) I dumped nothing. If I power on the TX (not the car), I also dumped nothing. If I explicitly listen on channel 45 and power both the car and the TX I got the following:
```
Channel=45,0x2D
RX:     0us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
```

yarda, 2025-02-13:
It seems the channel can change, after dozens dumps (steps are: power on the car and TX, control the car, power off the car), I got the following:
```
Proto=XN297DP, nbr_sub=7, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52

Packet detected: bitrate=Using NRF
250K C=52 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(15)= 08 64 64 00 00 00 00 00 00 00 34 4F 00 00 53
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45
RX on channel: 45, Time:     0us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
Trying RF channel: ,46,47,48,49,50,51,52,53,54,55,56,57,58
RX on channel: 58, Time:     0us P: 08 64 64 00 00 00 00 00 00 00 34 4F 00 00 53
Trying RF channel: ,59
RX on channel: 59, Time:     0us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
RX on channel: 59, Time: 11597us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
RX on channel: 59, Time: 343156us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
RX on channel: 59, Time: 26351us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
Trying RF channel: ,60,61,62,63,64,65,66
RX on channel: 66, Time:     0us P: 08 64 64 00 00 00 00 00 00 00 34 4F 00 00 53
Trying RF channel: ,67
RX on channel: 67, Time:     0us P: 08 64 64 00 00 00 00 00 00 00 34 4F 00 00 53
RX on channel: 67, Time: 11676us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
Trying RF channel: ,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84

5 RF channels identified: 45[1] 58[1] 59[4] 66[1] 67[2]
Keeping only RF channels with more than 2 packets: 59* 67
--------------------------------
Identifying RF channels order.
Time between CH:59 and CH:59
Time: 47887us
Time between CH:59 and CH:67
Time: 240789us

Channel order:
59:     0us
59: 47887us
67: 240789us

--------------------------------
Identifying Sticks and features.
P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
P: 08 64 64 00 00 00 00 00 00 00 34 4F 00 00 53
P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
```
I am not sure what the channel 67 is, noise, interference (remotes too close)? But it's now probably on channel 59.

pascallanger, 2025-02-13:
> The protocol seems pretty easy

Correction the packet structure is usually easy. The difficult part is usually the ID and associated RF frequencies(y), bind negotiation and sometimes timing.

> ```> P: 08 WW TT SS LL 00 00 00 00 GG EE DD PP 00 CC
> > WW - steering wheel, 00 - min, 64 - idle, C8 - full
> TT - throttle/break , 00 - min, 64 - idle, C8 - full
> SS - speed mode switch, 00 - min speed, 64 - middle speed, C8 - full speed
> LL - light button, 00 - idle (depressed), C8 - pressed
> GG - gyro, 00 - min, C8 - max
> EE - steering trim, 00 - min, C8 - max
> DD - steering dual rate, 00 - min, C8 - max
> PP - reverse switches, 00 - all normal, 40 - throttle reversed, 80 - steering reversed
> CC - checksum, just sum bytes 0 - 13
> > Example packet:
> P: 08 64 64 00 00 00 00 00 00 00 34 4F 00 00 53
> ```

That's a good start.

> I need to capture the bind sequence, I had some trouble with it, maybe it is (I am not 100 % sure):
> ```
> 250K C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
> ```

It's not a bind packet...

> It seems it has also telemetry built-in and probably returns car battery status to the remote, but currently I have no idea how to reverse it.

You are not sharing any dumps so can't see anything of what's really going on or help...

> I don't know whether it sticks to channel 45 or can autoselect different channel e.g. in case it's used/noisy, I tried several times and every time I got channel 45 (I will also recheck). It seems there is no hopping.

It can't autoselect... There must be a negotiation or pre established frequencies that are switched to in case of interferences.

>The model is autobound each time the model and remote is switched on.

That's a wild statement. I'm wondering how you can be sure of that...

> If I powered on the car (not the TX) I dumped nothing. If I powered on the TX (not the car), I also dumped nothing.

You are not saying what you do to "dump". But I can tell you that either the TX or the car is sending something.

> If I explicitly listen on channel 45 and power both the car and the TX I got the following:
>Channel=45,0x2D RX:     0us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
>So IMHO the address is F6 96 01 00 81.

That's not a bind packet or anything useful

> It seems the channel can change, after dozens dumps (steps are: power on the car and TX, control the car, power off the car), I got the following:

The frequency change might be due to a detected interference

> ```
> RX on channel: 59, Time:     0us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
> RX on channel: 59, Time: 11597us P: 08 64 64 00 00 00 00 00 00 00 34 50 00 00 54
> RX on channel: 59, Time: [...]

First time you are sharing what you are seeing. The output timing on channel 59 is not consistent and a really low packet rate. So something is not right.

pascallanger, 2025-02-13:
First, few questions:
- What does the documentation state about the bind/pairing process? Any specific action(s) on the TX or the car?
- How long does it take for the link to be established between the TX and the car?
- Does the TX report that the car is powered on? LED pattern?

Ok so let's capture the bind packet first to see the content:
1. Turn on only the original TX
2. Set the XN297DP protocol and sub protocol 250K
3. Scan manually all the frequencies from 0 to 84 waiting a ten seconds or more on each channel.
4. If something appears on a channel let it run and post the result here to see the packets and timing. In any case, packets appearing or not continue to scan manually all the channels from 0 to 84.
5. Redo step 2 to 4 with the original TX off and the car on.

yarda, 2025-02-13:
Thanks for the reply and tips.

> First time you are sharing what you are seeing. The output timing on channel 59 is not consistent and a really low packet rate. So something is not right.

I guess channel 59 can be interference, I did the dump when the remotes were cca. 50 cm far apart. If they were cca. 2 meters far I was seeing only channel 45. I will retry multiple times to be sure here. I don't have anything more at the moment, all the dumps were similar as the one I provided and didn't reveal anything more (apart the sticks mapping).

> First, few questions:
> 
>     * What does the documentation state about the bind/pairing process? Any specific action(s) on the TX or the car?
> 
Manual quote: After the car power is turned on, the transmitter and the car will carry out automatic frequency matching. When the indicator light of the transmitter and the car lights are always on, the frequency matching is successful. In the process of frequency matching, the car lights will flash rapidly. At this time, the gyroscope is being calibrated automatically. Please place the car on the level ground to complete this operation. After calibration, the car lights return to normal.

Full manual for the reference (it's for UDIR 1601, but IMHO it's the same car, I will recheck with the EAT-15 original manual when at home):
https://udirc.com/wp-content/uploads/UD1601-guide-V2.pdf

>      
>     * How long does it take for the link to be established between the TX and the car?
> 
cca. less than 2 seconds, I will measure it precisely
      
>     * Does the TX report that the car is powered on? LED pattern?
> 
yes and also the car battery status
> 
> Ok so let's capture the bind packet first to see the content:
> 
>     1. Turn on only the original TX
> 
>     2. Set the XN297DP protocol and sub protocol 250K
> 
>     3. Scan manually all the frequencies from 0 to 84 waiting a ten seconds or more on each channel.
>
OK, thanks for the tip, I will try the manual scan, that's the only thing I haven't done yet.

pascallanger, 2025-02-13:
> Thanks for the reply and tips.
>
>
> I guess channel 59 can be interference, I did the dump when the remotes were cca. 50 cm far apart. If they were cca. 2 meters far I was seeing only channel 45. I [...]

There are things you didn't share like the output of channel 45 with the car telemetry.
The time between packets is always displayed and this information is fully missing apart from the last post.
On the only shared dump you can see that none of the channels have a large number of RX which is "strange". I can't compare with how the packets are normally being sent when the car is also on since it's not shared.

> > * How long does it take for the link to be established between the TX and the car?
> cca. less than 2 seconds, I will measure it precisely
That could mean that the bind packets are sent every second or 2 seconds. Which is why you miss them with a quick scan.

> OK, thanks for the tip, I will try the manual scan, that's the only thing I haven't done yet.

Please post dumps of the **slow** manual scan of the tx on only, car on only, and both car and tx on. That will give hopefully a picture of what's going on.

yarda, 2025-02-15:
It's the first time I am using the MULTI-Module dump feature and I am still learning, so please be patient :)

I like the project, so I sent a small PayPal donation according to my possibilities, get a beer or two on me :)

I can confirm the EAT-15 manual is nearly the same to what I linked. In the EAT-15 printed manual some pages are reordered and formatting differs, but the information there is the same, including the text about the frequency matching which is literally the same. The radio and HW seems to be the same on all cars labeled UD1601-UD1608  / PRO variant (and probably also the SG1603-SG1604 variants and maybe others). The only visible difference on the cars is probably the different body shell and different shock tower assembly, but this can be easily cross replaced.

I measured time of the bind process. It usually takes cca. 1 - 2 seconds. The transmitter knows if the car is switched on and in range and if the battery level on the car is OK, this is signaled on the transmitter.

Dump of the manual channel scan, at least 10 seconds on the channel (usually more), transmitter (TRX) on, car off, not bound:
```
Protocol selected: 63, sub proto 0, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=7, Sub=250Kbps, Opt=8, FS=0, CHMap=0, rfSw=2
Channel=0,0x00
Channel=1,0x01
Channel=0,0x00
Channel=1,0x01
Channel=2,0x02
Channel=3,0x03
Channel=4,0x04
Channel=5,0x05
Channel=6,0x06
Channel=7,0x07
Channel=8,0x08
Channel=9,0x09
Channel=10,0x0A
Channel=11,0x0B
Channel=12,0x0C
RX:     0us C=12 Bad CRC
Channel=13,0x0D
Channel=14,0x0E
Channel=15,0x0F
Channel=16,0x10
Channel=17,0x11
Channel=18,0x12
RX:     0us C=18 Bad CRC
Channel=19,0x13
Channel=20,0x14
Channel=21,0x15
RX:     0us C=21 Bad CRC
Channel=22,0x16
Channel=23,0x17
RX:     0us C=23 Bad CRC
Channel=24,0x18
Channel=25,0x19
Channel=26,0x1A
Channel=27,0x1B
Channel=28,0x1C
Channel=29,0x1D
Channel=30,0x1E
Channel=31,0x1F
Channel=32,0x20
Channel=33,0x21
Channel=34,0x22
Channel=35,0x23
RX:     0us C=35 Bad CRC
Channel=36,0x24
Channel=37,0x25
Channel=38,0x26
Channel=39,0x27
Channel=40,0x28
Channel=41,0x29
Channel=42,0x2A
RX:     0us C=42 Bad CRC
RX: 111843732us C=42 Bad CRC
RX: 304134us C=42 Bad CRC
RX: 112974381us C=42 Bad CRC
RX: 695590us C=42 Bad CRC
RX: 113361919us C=42 Bad CRC
RX: 1469221us C=42 Bad CRC
[... 1155 lines cut ...]
51:     0us
67:  4767us
52:  4895us
51:  4896us
58: 31346us
59: 279032us

--------------------------------
Identifying Sticks and features.
P: 08 64 54 00 00 00 00 00 00 00 32 A4 00 00 96
P: 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
P: 08 64 50 00 00 00 00 00 00 00 31 A4 00 00 91
P: 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
P: 08 64 4D 00 00 00 00 00 00 00 32 A4 00 00 8F
P: 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
P: 08 64 54 00 00 00 00 00 00 00 32 A4 00 00 96
P: 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
P: 08 64 50 00 00 00 00 00 00 00 31 A4 00 00 91
P: 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
P: 08 64 50 00 00 00 00 00 00 00 31 A4 00 00 91
P: 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
```
So it's still the same channels. Is it possible it's doing some sort of very slow FHSS? According to the way the control sticks and CRC are coded I expect it to be something very simple.

I will redo the manual scan with the TRX off / car on ond Sunday. Then I can manually scans the channels 42, 44, 45, 51, 52, 58, 59, 66, 67 and while listening on each channel try to bind the car to get the full binding sequence. Am I right? Or should I do different thing?

pascallanger, 2025-02-16:
> It's the first time I am using the MULTI-Module dump feature and I am still learning, so please be patient :)

Sorry if I sounded pressing to get the data. It's just that you need to share as much details as possible to get to the bottom of it.

> I like the project, so I sent a small PayPal donation according to my possibilities, get a beer or two on me :)

Thanks!

> If I understand it correctly the following is the bind request with the TRX address F6 96 01 00 81? RX: 1225578us C=44 Enhanced pid=3 S=Y A= 01 03 05 07 09 P(15)= 01 F6 96 01 00 81 00 00 00 00 32 A4 [...]

The TX initiates the bind by sending it's ID and at the same time sending data. I think the car might not reply to the TX bind packets but instead to the normal packets by sending telemetry back which indicates to the TX that the car is there and therefore exits bind.

The timing and RF channels are still a bit fuzzy, I need to dive in the data to see.

The bad CRCs and the channels receiving low number of packets are usually harmonics or crosstalk. It's kind of usual with the NRF24L01 and noise we need to deal with... If you have a spectrum analyzer, sure check which channels are really in use and if you can get channels usage overtime that could be perfect.

>I will redo the manual scan with the TRX off / car on on Sunday. Then I can manually scan the channels 42, 44, 45, 51, 52, 58, 59, 66, 67 and while listening on each channel try to bind the car to get the full binding sequence. Am I right? Or should I do different thing?

TX off and car on is not going to give anything since the car seems to only respond to the TX.
Instead, I would really like to see a dump in auto and manual of the TX on and car on.

pascallanger, 2025-02-16:
Using XN297 250K Enhanced/Scrambled

Bind packets
---
Bind packets alternate with normal packets
Timing ?
RF channels ?
Address = 01 03 05 07 09
P(15)= 01 F6 96 01 00 81 00 00 00 00 32 A4 00 00 E5
P[0] = 01 = bind packet
P[1] = TXID[0]
P[2] = TXID[1]
P[3] = TXID[2]
P[4] = TXID[3]
P[5] = TXID[4]
P[6] = 00 ?
P[7] = 00 ?
P[8] = 00 ?
P[9] = Gyro 00..C8
P[10] = ST_Trim 00..C8
P[11] = ST_DR 00..C8
P[12] = 00 | 40 TH_REV | 80 ST_REV
P[13] = 00 ?
P[14] = sum P[0..13]

Normal packets
---
250K C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(15)= 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92

P[0] = 08 = data packet
P[1] = ST 00..64..C8
P[2] = TH 00..64..C8
P[3] = Rate 00 low/64 mid/C8 high
P[4] = Light 00/C8
P[5] = 00 ?
P[6] = 00 ?
P[7] = 00 ?
P[8] = 00 ?
P[9] = Gyro 00..C8
P[10] = ST_Trim 00..C8
P[11] = ST_DR 00..C8
P[12] = 00 | 40 TH_REV | 80 ST_REV
P[13] = 00 ?
P[14] = sum P[0..13]

Telemetry
---

pascallanger, 2025-02-16:
Ok I knew I've seen these packets before... That's the Pinecone protocol which I've failed to reverse the first time... May be with your additional TX/RX and dumps we will be able to get to the bottom of it.
So you know the name of the true manufacturer now.
The start of the code is here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Pinecone_ccnrf.ino
If you check in the code the RF channels for the other TX were the same: 
hopping_frequency[0] = 45;
hopping_frequency[1] = 59;
hopping_frequency[2] = 52;
hopping_frequency[3] = 67;

If I remember I was able to bind the car, get answers from the car (telemetry) but never found how to control it = the LED on the car kept blinking.

pascallanger, 2025-02-16:
Can you open your tx and check if you have a separated RF chip? The pinecone tx is an all in one so no way to spy the spi bus...

yarda, 2025-02-16:
It has the FCC ID, so here are the full docs (I like the US FCC guys :):
https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=N&application_id=kIJBJn9P5bCCXXJ0BkObag%3D%3D&fcc_id=2AYZQUDUA31

As the applicant name there is written: UDIRC TECHNOLOGY CO., LTD.

According to the internal photos, it's a single chip.

More interesting information from the docs:
```
Serial models:
UD8xx, SG-8xx, UD10xx, SG-10xx, UD12xx, SG-12xx, UD1
4xx, SG-14xx, UD16xx, SG-16xx, UD18xx, SG-18xx, UD24x
x, SG-24xx, UD28xx, SG-28xx, UD32xx, SG-32xx, UD35xx,
SG-35xx, UD52xx, SG-52xx, UD65xx, SG-65xx, UD76xx,S
G-76xx（X=0-9）
Model difference: All model’s the function, software and electric circuit are the
same, only with a product color and model named different.
Test sample model: UD1601
```

Regarding the RF:
```
4 channels, GFSK
Carrier Frequency of Channels (MHz)
1: 2445
2: 2452
3: 2459
4: 2467
```
I guess the channel numbers maps to the last two frequency digits, which means channels: 45, 52, 59, 67.

They also have nice spectrum screenshots for all used channels, so probably no need for me to duplicate their work :).

But maybe I could use the SDR to find the channels sequence order.

yarda, 2025-02-16:
More dumps:

Manual channel 45, while listening, car switched on, TRX switched on:

```
Channel=45,0x2D
RX:     0us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(15)= 08 64 64 00 00 00 00 00 00 00 32 A4 00 00 A6
RX:  6679us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(15)= 10 00 05 FF FF B7 00 00 00 00 00 00 00 00 00
RX: 26575us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  7679us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 16030us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  9608us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX: 46512us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 24528us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  7515us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX:  8095us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  7962us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  9396us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  6668us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX:  8922us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX: 15673us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 15768us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  7697us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  9936us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX:  8098us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX:  9615us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  8430us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX: 24969us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX: 15316us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  7815us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX:  9821us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX: 17716us C=45 Bad CRC
RX: 55054702us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(15)= 10 00 05 FF FF B7 00 00 00 00 00 00 00 00 00
RX: 25181us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  8413us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  7601us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  7972us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 17793us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX: 17726us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX:  6664us C=45 Bad CRC
RX: 55166025us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  7071us C=45 Bad CRC
RX: 55181369us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX: 15679us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  7701us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  9703us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  8322us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX: 16054us C=45 Bad CRC
RX: 55238689us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX: 15316us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  7899us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  9739us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX: 24344us C=45 Bad CRC
RX: 55303727us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX: 31321us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  6767us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX:  9424us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  6670us C=45 Bad CRC
RX: 55384505us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 13610us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(15)= 08 64 64 00 00 00 00 00 00 00 32 A4 00 00 A6
RX: 19371us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 17338us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX: 19200us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 25792us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  9460us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
RX: 37800us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX: 39634us C=45 Enhanced pid=3 S=Y A= F6 96 01 00 81 P(0)=
RX:  7406us C=45 Enhanced pid=2 S=Y A= F6 96 01 00 81 P(0)=
RX:  9931us C=45 Enhanced pid=1 S=Y A= F6 96 01 00 81 P(0)=
RX:  8123us C=45 Enhanced pid=0 S=Y A= F6 96 01 00 81 P(0)=
[... 1120 lines cut ...]
Identifying RF channels order.
Time between CH:45 and CH:44
Time: 32697us
Time: 60512us
Time: 16021us
Time:  8390us
Time: 31239us
Time: 34012us
Time:  9040us
Time:  7433us
Time: 65717us
Time: 40640us
Time:  8157us
Time between CH:45 and CH:45
Time:  5132us

Channel order:
45:     0us
45:  5132us

--------------------------------
Identifying Sticks and features.
```

Identifying Sticks and features returned nothing.

yarda, 2025-02-16:
Correction, from the photos it seems there are three chips near the antenna.

yarda, 2025-02-16:
The TRX from the FCC photos has three more trimmers, maybe it's for three more channels that is unused on my model and hence the 00 there.

yarda, 2025-02-16:
> Correction, from the photos it seems there are three chips near the antenna.

Correction of the correction, my TRX has single chip: PAN186CV. Not sure what are the other two chips on the FCC photos.

yarda, 2025-02-16:
It's probably PANCHIP, datasheet in chinese:
https://www.chipqry.com/datasheet/pan186cv

Machine translated datasheet:
https://www.onlinedoctranslator.com/app/gettranslateddocument/pan186cv_cn.zh-CN.en.pdf

The datasheet doesn't seem much useful.

yarda, 2025-02-16:
With the SDR I can do 16 MHz bandwidth PHY GFSK dump, unfortunately, this beast has over 22 MHz bandwidth, so I am unable to capture the whole hop sequence in realtime :).

yarda, 2025-02-16:
I probably got the difference of the TRX shown on the FCC side. In the initial version they certified they probably used 4wd version of the TRX, it has 4wd mode switch and status LEDs for the modes (front drive, rear drive, front/rear drive, front/rear drive reversed) and in addition to the dual rate trimmer it has also front steering and rear steering trimmers. These are probably causing the most zeroes in the payload in my dumps.

yarda, 2025-02-16:
And the PAN186CV doesn't have enough ADCs, so they used external ADCs for all the trimmers, hence the two more ICs on the FCC photos.

yarda, 2025-02-17:
> With the SDR I can do 16 MHz bandwidth PHY GFSK dump, unfortunately, this beast has over 22 MHz bandwidth, so I am unable to capture the whole hop sequence in the realtime :).

If necessary I could probably borrow SDR with wider bandwidh from the local university or I have HF SDR with cca. 30 MHz bandwidth and I could homebrew 2.4 GHz downconverter for it, but both approaches would require some time.

yarda, 2025-02-17:
> If necessary I could probably borrow SDR with wider bandwidh from the local university or I have HF SDR with cca. 30 MHz bandwidth and I could homebrew 2.4 GHz downconverter for it, but both [...]

Or maybe I could sniff 2-3 successive channels per time. I suppose the hop sequence is static with the constant delay between hops, so from multiple sniffs I should be able to verify the hop sequence. I will try to play with it during Tuesday.

pascallanger, 2025-02-17:
The scans with both the TX and RX on seems to show a few things:
 - for whatever reason, my scanning process seems to catch mostly the empty P(0) packets
   - it could be because for the xn297 emulation I have no idea of the size of the incoming packet so I'm receiving 32 bytes all the time. If the TX/RX responds right away behind the P(0), I might not see the start of the next packet...
   - the empty packets P(0) are most likely sent by the car, so it could be that the TX just replies right away by sending the control packet and therefore I miss them.
 - all 4 frequencies are used
   - but all 4 frequencies show large gaps which seems to indicate that the frequency is changing really slowly
   - I'm not sure what's making the frequency change, 3 options:
     - it changes on a given timing -> this is most likely not the case since in case of interference the TX/RX would lose connectivity for a long time
     - it changes when the car decides in the telemetry frame -> I don't think the car would drive this part
     - it changes when the TX doesn't receive the P(0) = some sort of ack from the car -> my preferred option. I would see something like if no ack is received for X number of packets, then send data on 4 the for frequencies and when the car acks on a frequency stay on it until it happens again. Therefore it doesn't really matter in which order you hop, you just try to get the connection back.
 - the car telemetry might send a real telemetry packet time to time instead of just ack but that's for later... And we will need to understand what's inside anyway.
 - the timing of the packets is really all over the place... I'm thinking to go with 8ms.

Based on these assumptions, I think I could try to code the behavior I've described.
But I could also try to start simple to test it by only sending data on one RF frequency and see if we can get control of the car. I think I might have failed in the past because I tried to send data on all 4 channels all the time, the car was acking some packets but since I was switching channels it kept thinking that the connection was broken so not accepting the few valid control packets.

pascallanger, 2025-02-17:
Using XN297 250K Enhanced/Scrambled

Bind packets
---
Bind packets alternate with normal packets -> most likely take place of a standard packet time to time
Timing ?
RF channels: 45, 59, 52, 67
Address = 01 03 05 07 09
P(15)= 01 F6 96 01 00 81 00 00 00 00 32 A4 00 00 E5
P[0] = packet type, 01 = bind, 08 = normal, 10 = telemetry
P[1] = TXID[0]
P[2] = TXID[1]
P[3] = TXID[2]
P[4] = TXID[3]
P[5] = TXID[4]
P[6] = 00
P[7] = 00
P[8] = 00
P[9] = Gyro 00..C8
P[10] = ST_Trim 00..C8
P[11] = ST_DR 00..C8
P[12] = 00 | 40 TH_REV | 80 ST_REV
P[13] = 00 ?
P[14] = sum P[0..13]

Normal packets
---
Timing 8-9ms
RF channels: 45, 59, 52, 67 when car is off, one channel when car is on
Address = F6 96 01 00 81
P(15)= 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
P[0] = packet type, 01 = bind, 08 = normal, 10 = telemetry
P[1] = ST 00..64..C8
P[2] = TH 00..64..C8
P[3] = Rate 00 low/64 mid/C8 high , CH4 on SG 1604 model
P[4] = Light 00/C8 , CH3 on SG 1604 model
P[5] = 00 -> most likely additional channel
P[6] = 00 -> most likely additional channel
P[7] = 00 -> most likely additional channel
P[8] = 00 -> most likely additional channel
P[9] = Gyro 00..C8
P[10] = ST_Trim 00..C8
P[11] = ST_DR 00..C8
P[12] = 00 | 40 TH_REV | 80 ST_REV
P[13] = 00 -> future flags ?
P[14] = sum P[0..13]

Telemetry
---
RF channel -> same as normal
2 different packets are sent back:
- Empty packets P(0), some sort of ack ?
- Telemetry P(15)= 10 00 05 FF FF B7 00 00 00 00 00 00 00 00 00

Telemetry packets seen: 
P(15)= 10 00 05 FF FF B7 00 00 00 00 00 00 00 00 00
P(15)= 10 00 03 FF FF B8 CF 0A 6C 37 FC 31 D7 8E 58
P(15)= 10 00 03 FF FF B8 CE 0A 6C 37 FC 31 D7 06 58
P[0] = packet type, 01 = bind, 08 = normal, 10 = telemetry
P[1..14] = ???
Looks to be sent every 14-15 normal packets

yarda, 2025-02-17:
I guess the `10 00 03` telemetry packets are from the previous model? Because, IIRC I haven't seen this packets with my model.  I will play with it, I will try to increase the distance from the TRX (maybe there is RX quality/signal strength reported) and will try with battery charged to different levels.  I expect some very simple coding, e.g. one FF for RX quality, second FF for battery level, or just battery depleted flag :). Regarding the `10 00 03` packets, I have no idea and probably I cannot help, because AFAIK I was unable to reproduce them.

pascallanger, 2025-02-17:
For the car telemetry, we will see when we can receive them within the protocol. For now, no need to spend time on this...

pascallanger, 2025-02-17:
I can see in my code that I already tried to send normal packets to only one channel... The only thing the code is not doing is to alternate bind and normal packets. But if I remember the car was replying with P(0) to the normal address meaning it took the bind info.
So still not sure what to try next.

yarda, 2025-02-17:
I will try to build the test code and will give it a try. Is there anything special I need to do for the STM32? Other than installing the board into Arduino IDE?

I will also try to provide the PHY GFSK dumps.

pascallanger, 2025-02-18:
I've pushed my latest code and renamed the protocol from Pinecone to UDIRC. I can't test anything and doubt it will work as it is.

pascallanger, 2025-02-18:
I can't push to github anymore... It is stuck with the message "Pushing to origine. Please hang on"... Since the latest windows patches some stuff aren't working anymore. Not sure what I should do as I can't really find anything useful online.

pascallanger, 2025-02-18:
Using git in command line works but not using the GUI... The repo is now up to date.
Building and uploading is explained here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md
You need to modify the _config.h file (or do your updates in _MyConfig.h) as by default the code is too large to fit in flash. For example disable PPM, and if you are on a MT12 add #define MULTI_SURFACE to only compile surface protocols.
To compile with debug, you have to set in the Arduino IDE Tools->Debug option->Serial/FTDI

aseitz79, 2025-02-18:
One question: As far as I understand, when this here protocol is implemented, also the Pinecone protocol should work (see https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/952) right?

pascallanger, 2025-02-18:
Yes same troublesome protocol...

yarda, 2025-02-18:
I sniffed the GFSK PHY. I can confirm the channel sequence is: `45 59 52 67`. It sends successively three XN297 coded packets (triplets) @ 250 kbit GFSK. Each packet takes cca. 1.5 ms. Between packets there is cca. 3.5 ms pause. Between triplets there is cca. 10.3 ms pause. It sends five triplets then it hops. I.e. it spends cca. 97 ms on each channel. I.e. the full hop cycle takes cca. 400 ms. The TX spectrum is quite nice, not worse than the spectrum on the FCC photos. Now I am writing the PHY decoder, so I will exactly know who sends what and when. It will take some time.

pascallanger, 2025-02-19:
Was the car powered on during your observation?

pascallanger, 2025-02-19:
> I can confirm the channel sequence is: `45 59 52 67`

It matches what I've found and my code:
```
		hopping_frequency[0] = 45;
		hopping_frequency[1] = 59;
		hopping_frequency[2] = 52;
		hopping_frequency[3] = 67;
```
> It sends successively three XN297 coded packets (triplets) @ 250 kbit GFSK. Each packet takes cca. 1.5 ms. Between packets there is cca. 3.5 ms pause. Between triplets there is cca. 10.3 ms pause.

I assume the car was on.
From the TX dumps with car turned off, I can see Transmit_packet_0 -> 5ms -> Transmit_packet_0 -> 16ms -> Transmit_packet_1 -> 5ms -> Transmit_packet_1. So I assume the 3rd, transmit in between was the car.

Dump RX off: |T0 -> 5ms -> |T0 16ms                           -> |T1 -> 5ms -> ...
--|--|--|--
Dump RX off: |T0 -> 1.5+3.5ms -> |T0 1.5+14.5ms                           -> |T1 -> 1.5+3.5ms -> ...
Sniffed RX_on: |T0 -> 1.5+3.5ms -> |T0 1.5+3.5ms -> RX 1.5+10ms -> |T1 -> 1.5+3.5ms -> ...

So your observations match with the prior dumps if the car was on.

yarda, 2025-02-19:
I think I will be able to distinguish the individual parties according to the different RSSI, so it should be clear what's the TRX packet and what's the car packet.

pascallanger, 2025-02-19:
If I turn on the original TX, turn on the RX, turn off the original TX, send with my module packets with the same ID, I can control the car.
So I think there is some sort of challenge response that the RX is waiting before accepting packets.

pascallanger, 2025-02-19:
After spending hours, the RX is finally accepting my normal packets!!! You need to "ack" what the RX is sending. The "ack" is done by sending a packet with no payload P(0).
Still need to work out some details but that's a good step forward.

yarda, 2025-02-20:
> What hardware/software are you using?
> 
USRP / GNURadio

pascallanger, 2025-02-20:
Never had to send an empty payload in enhanced mode. There was a bug in the emulation layer that I've fixed (2 bits were not getting xored).
Usually protocols are happy when the bidirectional communication is established but for this one they haven't tested the successful  handshake based on packets exchange but the if they receive an ack or not instead... Once at least one ack is received the RX is happy and starts to operate.

I don't know how the TX decides to switch from one RF channel to another. I'm wondering if it hops to the next channel when it doesn't receive telemetry for a couple of packets. So if out of 50 telem packets you receive less than let's say 10 you switch.

Finding the telemetry flag that indicates low batt is easy. I'll take care of this.

pascallanger, 2025-02-20:
Using XN297 250K Enhanced/Scrambled

Bind packets
---
Address = 01 03 05 07 09
RF channels: 45, 59, 52, 67
Timing : bind packets alternate with normal packets -> most likely take place of a standard packet time to time
P(15)= 01 F6 96 01 00 81 00 00 00 00 32 A4 00 00 E5
P[0] = packet type, 01 = bind, 08 = normal, 10 = telemetry
P[1] = TXID[0]
P[2] = TXID[1]
P[3] = TXID[2]
P[4] = TXID[3]
P[5] = TXID[4]
P[6] = 00
P[7] = 00
P[8] = 00
P[9] = Gyro 00..C8
P[10] = ST_Trim 00..C8
P[11] = ST_DR 00..C8
P[12] = 00 | 40 TH_REV | 80 ST_REV
P[13] = 00 ?
P[14] = sum P[0..13]

Bind packet from another TX: P(15) = 01 D0 06 00 00 81 00 00 00 69 5F C8 00 00 E8

Normal packets
---
Address = F6 96 01 00 81 / D0 06 00 00 81
RF channels: 45, 59, 52, 67 when car is off, one channel when car is on. How is the hopping mechanism work when there is a conflict?
Timing ?
P(15)= 08 64 50 00 00 00 00 00 00 00 32 A4 00 00 92
P[0] = packet type, 01 = bind, 08 = normal, 10 = telemetry
P[1] = ST 00..64..C8
P[2] = TH 00..64..C8
P[3] = Rate 00 low/64 mid/C8 high , CH4 on SG 1604 model
P[4] = Light 00/C8 , CH3 on SG 1604 model
P[5] = 00 -> most likely additional channel
P[6] = 00 -> most likely additional channel
P[7] = 00 -> most likely additional channel
P[8] = 00 -> most likely additional channel
P[9] = Gyro 00..C8
P[10] = ST_Trim 00..C8
P[11] = ST_DR 00..C8
P[12] = 00 | 40 TH_REV | 80 ST_REV
P[13] = 00 -> future flags ?
P[14] = sum P[0..13]

Telemetry
---
Address-> same as normal
RF channel -> same as normal
Telemetry packets seen:
P(15) = 01 03 03 FF FF B8 CF 0A 6C 37 FC 21 D7 0E 58 -> no ack received on telemetry packets, bind/connection incomplete
P(15)= 10 01 03 FF FF B8 CE 0A 6C 37 FC 31 D7 06 58 -> low battery <6.6V
P(15)= 10 00 03 FF FF B8 CF 0A 6C 37 FC 31 D7 8E 58
P(15)= 10 00 03 FF FF B8 CE 0A 6C 37 FC 31 D7 06 58
P(15)= 10 00 03 FF FF B8 CE 0A 6C 37 FE 31 D7 06 58
P(15)= 10 00 05 FF FF B7 00 00 00 00 00 00 00 00 00

At least one telemetry packet must receive an ack (packet with an empty payload P(0) ) for the receiver to consider the connection established (same goes for the original TX).

P[0] = 01 = bind, 08 = normal, 10 = telemetry
P[1] = 01 = low batt < 6.6V, 03 = not bound/connected
P[2] = 03/05 car type?
P[3] = FF?
P[4] = FF?
P[5] = B7/B8 sub car type?
P[6..14] = seems like a buffer over run in case of car type 3 and a clean buffer on car type 5

MRC3742, 2026-08-19:
I have just acquired a [Pinecone Forest SG-1205 Model](https://usa.banggood.com/SG-PINECONE-FOREST-1205-RTR-1-or-12-2_4G-4WD-RC-Car-SSV-Off-Road-Truck-LED-Light-Rock-Crawler-Full-Proportional-Vehicles-Models-Toys-p-1994054.html?cur_warehouse=USA&ID=6287845&rmmds=search) with this same problematic protocol.

TX ID= C3 E4 04 00 81, Channels 45, 52, 59, 67
I have done numerous captures and found some new information regarding this protocol.
These two payloads both occur with TX ON and looking for the model before RX is powered.
```
C=45 Enhanced pid=0 S=Y A= 01 03 05 07 09 P(15)= 01 C3 E4 04 00 81 00 00 00 64 60 6C 00 00 5D
C=45 Enhanced pid=2 S=Y A= C3 E4 04 00 81 P(15)= 08 64 64 64 00 00 00 00 00 64 60 6C 00 00 64
```

Once  the RX is powered the captures are flooded with P(0)= payloads
`C=45 Enhanced pid=0 S=Y A= C3 E4 04 00 81 P(0)=`

Then I have only captured one with starting packet "02" and not sure if its coming from TX or RX?
This is critical info as F8 and 30 are inserted into the normal packets starting with "08" to complete the bind.
These bytes, F8 and two instances of 30 are also included in the RX telemetry payloads starting with "10".
```
C=59 Enhanced pid=3 S=Y A= C3 E4 04 00 81 P(15)= 02 F8 00 00 30 00 00 00 00 64 60 6C 00 00 5A
C=45 Enhanced pid=2 S=Y A= C3 E4 04 00 81 P(15)= 10 00 00 00 30 00 00 00 00 00 00 F8 00 00 30
```

The normal payload packets starting with "08" change before bind completes to this sequence.
`C=45 Enhanced pid=0 S=Y A= C3 E4 04 00 81 P(15)= 08 64 64 64 00 F8 00 00 30 64 60 6C 00 00 8C`

I have been able to manipulate the code adding P[5]=F8, P[8]=30 and although it still will not bind using MPM, as Pascal wrote above, if I bind the model with the original TX, then power down TX, power up the MPM TX with my updated code, the model will rebind and I now have control.

I am also adding a few captures for review, and if any more are needed I can provide them as well.

[Pinecone_SG-1205_TX-Bind_RX_Bound_02.txt](https://github.com/user-attachments/files/31235749/Pinecone_SG-1205_TX-Bind_RX_Bound_02.txt)

[Pinecone_SG-1205_TX-Bind_RX_Bound_at-each-CH-Start.txt](https://github.com/user-attachments/files/31235752/Pinecone_SG-1205_TX-Bind_RX_Bound_at-each-CH-Start.txt)

Thanks - Rich

pascallanger, 2026-08-20:
I've done many iterations. At one point I was able to control the car. I think at another point I was able to bind it but failed to control. Then got into health issues and it stayed there.
Let me share what I have on my laptop, may be you will find something useful.

yarda, 2026-08-20:
I am also still on it. I had to focus on more prio tasks, but it's still in my backlog :) Hopefully, we will sort it out :)

pascallanger, 2026-08-20:
Attaching the files I was mentioning. They might be full of junk due to loads of trials.

[UDIRC_ccnrf.zip](https://github.com/user-attachments/files/31275392/UDIRC_ccnrf.zip)

MRC3742, 2026-08-21:
@pascallanger 

I have used your latest .ino file as a start and have successfully been able to bind and control the model with my TX ID. 

I am trying to determine how the RX responses "01" and "02" are used to modify the normal packets starting with "08" to get control after bind.
Do you have any captures of your normal packets after bind from your TX ID of A= D0 06 00 00 81 
yarda captures show all Packets[5]. [6], [7], [8], [12], [13] are all 00 before and after bind. 
Mine shows Packets [5], [8] change from 00, 00 to after bind F8, 30.

Thanks - Rich