pascallanger/DIY-Multiprotocol-TX-Module #1055: Module Flash size: 256KB

Mr-Squeeto, 2025-01-07:
Thank you Pascal for your incredible contribution to this project and your continued assistance to others.

Because of its weak transmitter, I currently have one plane that I need to address, a 4 channel Volantex P51, and want to use hardware at hand.
I build this around a FlySky FS-i6S transmitter and a 64KB BluePill STM32 but am having no luck at binding, does this look okay:

Multiprotocol version: 1.3.4.12
Module Flash size: 256KB
EEPROM initialized: 0
Protocol selection switch reads as 1
Using bank 0
Module Id: 1a59e75a
Protocol selected: 48, sub proto 1, rxnum 0, option 0
ID: 1A 59 E7 5A , HOP: 0F 14 46
Using NRF
Proto=V761, nbr_sub=3, Sub=4ch , Opt=0, FS=0, CHMap=0, rfSw=2
Init complete

I am curious about the Module Flash size of 256KB for a 64KB MCU.
Is there a _Config.h setting for this?
Thank you

pascallanger, 2025-01-07:
Everything looks good from a startup perspective.
256KB is what the bluepill's STM32 is reporting... I don't think it has any impact. If the NRF is well connected to the bluepill and you have set the module in bind mode right after the plane has been plugged that should work...

Mr-Squeeto, 2025-01-08:
I will check the NRF wiring.

Thank you. Last request..
Do the lights blink as expected?

https://www.youtube.com/watch?v=pW4d-1_LSx0

_Config.h

#define AETR
..
#define NRF24L01_INSTALLED
..
#define NRF24L01_ENABLE_LOW_POWER
..
#define	FX_NRF24L01_INO
..
#define	V761_NRF24L01_INO
..
#define ENABLE_PPM
..
#define TX_CUSTOM		//Custom
..
#if defined(TX_CUSTOM)
	#define PPM_MAX_100	1900	//	100%
	#define PPM_MIN_100	1100	//	100%
#endif
..
#define MIN_PPM_CHANNELS 4
..
#define MAX_PPM_CHANNELS 10
..
#define NBR_BANKS 1

const PPM_Parameters PPM_prot[14*NBR_BANKS]=	{
#if NBR_BANKS > 0
//******************************       BANK 1       ******************************
//	Switch		Protocol	Sub protocol	RX_Num		Power		Auto Bind		Option		Chan Order
/*	1	*/	{PROTO_V761,	V761_4CH  ,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	2	*/	{PROTO_V761,	V761_4CH  ,	1	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	3	*/	{PROTO_V761,	V761_4CH  ,	2	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	4	*/	{PROTO_V761,	V761_4CH  ,	3	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	5	*/	{PROTO_V761,	V761_4CH  ,	4	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	6	*/	{PROTO_V761,	V761_4CH  ,	5	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	7	*/	{PROTO_V761,	V761_4CH  ,	6	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	8	*/	{PROTO_V761,	V761_4CH  ,	7	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	9	*/	{PROTO_V761,	V761_4CH  ,	8	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	10	*/	{PROTO_V761,	V761_4CH  ,	9	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	11	*/	{PROTO_V761,	V761_4CH  ,	10	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	12	*/	{PROTO_V761,	V761_4CH  ,	11	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	13	*/	{PROTO_FX,	FX620     ,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	14	*/	{PROTO_FX,	FX620     ,	1	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
#endif
..
//my adds:
#define ARDUINO_GENERIC_STM32F103C
#define GENERIC_STM32F1_SERIES
//#define DEBUG_SERIAL

Mr-Squeeto, 2025-01-08:
Thank you again. I should have all that I need.
I have tried holding the bind button (black wire to gnd in the video) on start up but it is the same results.

I contacted the seller about a better transmitter option for this plane and he said that it is based on the Volantex P51.
But maybe it has a slightly different protocol than the V761.

I will check my (rats nest) wiring. I do intend to finally etch a circuit board when I have it all correct.

pascallanger, 2025-01-08:
It's the same protocol. Again there was no bind at startup so for whatever reason your strap is. Not working... You can force the module to initiate a bind automatically at startup by changing the No_autobind  flag in the bank.

closed 2025-02-26 by pascallanger

Mr-Squeeto, 2025-02-26:
Thanks again.
I did have an idea though, could a receiver be coded to collect 2 seconds of transmission to capture the bind signal and then usb send it to file to view on computer?

pascallanger, 2025-02-26:
It's already there and called XN297dump, this will scan and output what it finds on the serial usb connection.
But from our previous exchange your module is not sending any bind packets therefore it can't work. You must request the module to send a bind sequence by pressing the bind button before turning on the module.

Mr-Squeeto, 2025-02-26:
I seem to be able to bind with a PROTO_FX, FX620 so maybe my PROTO_V761, V761_4CH is defective. It has been starting to act up. It is just a cheap toy.
I will dedicate some extra time to this and I will report back if anything comes to fruition.

pascallanger, 2025-02-26:
The FX protocol is autobind. Which means that it will bind even if not ask. This is why it works...
If you don't want to troubleshoot your bind button then just add this line
`BIN_IN_PROGRESS;`
in this file and line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/d800f2c3332ef880d122aeb75a3638a9a94995e0/Multiprotocol/V761_nrf24l01.ino#L324

pascallanger, 2025-02-26:
Try to add the line like I indicated earlier and see if your plane works. You must have the plane on and initialized before turning on the module.