pascallanger/DIY-Multiprotocol-TX-Module #1057: E190 heli protocol

pascallanger, 2025-01-10:
E190 protocol

1M C=50 S=Y A= B3 13 36 DD D9 P(9)= A0 00 00 FD FF F8 F3 55 DC

4 RF channels identified: 50[10] 53[10] 66[10] 73[10]
Keeping only RF channels with more than 6 packets: 50* 53 66 73
--------------------------------
Identifying RF channels order.
Time between CH:50 and CH:50
Time: 49734us
Time: 49739us
Time: 49735us
Time: 49735us
Time between CH:50 and CH:53
Time: 18947us
Time: 12435us
Time: 12431us
Time: 12434us
Time: 12436us
Time: 12437us
Time: 12428us
Time: 12440us
Time between CH:50 and CH:66
Time: 24845us
Time: 24844us
Time: 24849us
Time: 24850us
Time: 24860us
Time: 24859us
Time: 24855us
Time: 24852us
Time: 24855us
Time between CH:50 and CH:73
Time: 37272us
Time: 37278us
Time: 37279us

Channel order:
50:     0us
53: 12428us
66: 24844us
73: 37272us

--------------------------------
Identifying Sticks and features.
P: A0 00 00 FD FF FD F3 55 E1
P: A0 00 00 FD FF F8 F3 55 DC
P: A0 00 00 FB FB FF F6 55 E0
P: A0 00 00 FD FF FF F3 55 E3
P: A0 00 00 FD FF F5 F3 55 D9
P: A0 00 00 FB FB FD F6 55 DE
P: A0 00 00 FB FB F8 F6 55 D9
P: A0 00 00 FB FB FD F6 55 DE
P: A0 00 00 FD D5 F5 DC 55 98
P: A0 00 00 02 63 D8 4B 56 7E
P: A0 00 00 74 9E 83 95 12 DC
P: A0 00 00 E8 00 00 00 03 8B
P: A0 00 00 DB 00 00 00 01 7C
P: A0 00 00 02 00 00 00 01 A3
P: A0 00 00 00 00 00 00 00 A0

Normal mode:

1M C=50 S=Y A= B3 13 36 DD D9 P(9)= A0 00 00 FD FF F8 F3 55 DC

P[0]: 00
    | 80 bind done
	| 20 not telem from the heli
	| 10 Lock
	| 08 H / 00 L
	| 01 Screw pitch (temporary)=long press up
P[1]: 00
	| 08 Land ? only when unlock (toggle) = short press down
	| 20 Mode (toggle)
P[2]: 00
	| 02 Altitude hold
	| 03 Manual
	| 40 Flip (toggle)
P[3]: CH1L A=0..1F4(500)..3E8(1000)
P[4]: CH2L E=0..1F4(500)..3E8(1000)
P[5]: CH3L T=0..1F4(500)..3E8(1000)
P[6]: CH4L R=0..1F4(500)..3E8(1000)
P[7]: CH4H<<6|CH3H<<4|CH2H<<2|CH1H
P[8]: sum(P[1..7])

Sticks low out
P: A0 00 40 E8 00 00 00 03 CB
Sticks low in
P: A0 00 40 00 00 00 E8 C0 88
Sticks low left
P: A0 00 40 00 00 00 00 00 E0

Heli telem in normal mode:
2 packets:
P(9)=	78	0	0	0	50	10	0	0	D8
P(9)=	78	0	0	0	4F	10	0	0	D7
P[4]: BATL?
P[5]: BATH?
Bat voltage=2*BATH:BATL

P(9)=	79	FE	0A	0	0	0	5	0	86
P(9)=	79	0	0B	0	0	0	5	0	89
P[1]: looks like the low part of a counter but not quite
P[2]: looks like the high part of a counter but not quite

Bind:
RX: 76815us C=48 S=Y A= B3 13 36 DD D9 P(9)= B0 00 00 FE FE F7 F2 55 EA
RX: 49681us C=48 S=Y A= B3 13 36 DD D9 P(9)= B0 00 00 FD FE F7 F2 55 E9
RX: 12426us C=48 S=Y A= 00 00 00 00 00 P(9)= B0 00 00 FE FE FA F2 55 ED
RX: 59824us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 76820us C=48 S=Y A= B3 13 36 DD D9 P(9)= B0 00 00 FE FA F2 F2 55 E1
RX: 49704us C=48 S=Y A= B3 13 36 DD D9 P(9)= B0 00 00 FE FA F2 F2 55 E1
RX: 72320us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 126608us C=48 S=Y A= B3 13 36 DD D9 P(9)= B0 00 00 FE FB EF F5 55 E2
RX: 12428us C=48 S=Y A= 00 00 00 00 00 P(9)= B0 00 00 FE FA F2 F2 55 E1
RX: 39987us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19896us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19890us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 56980us C=48 S=Y A= B3 13 36 DD D9 P(9)= B0 00 00 FE FA FA F5 55 EC
RX: 62162us C=48 S=Y A= 00 00 00 00 00 P(9)= B0 00 00 FB FA EF F5 55 DE
RX: 59867us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19885us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E

RX: 69382us C=48 S=Y A= 00 00 00 00 00 P(9)= B0 00 00 FD FF F7 F2 55 EA
RX: 20078us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19879us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 39765us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 119086us C=48 S=Y A= 00 00 00 00 00 P(9)= B0 00 00 FB FB FA F5 55 EA
RX: 20088us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 39781us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19896us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 119119us C=48 S=Y A= 00 00 00 00 00 P(9)= B0 00 00 FE FB EF F5 55 E2
RX: 39987us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19890us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19884us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 119094us C=48 S=Y A= 00 00 00 00 00 P(9)= B0 00 00 FE FB EF F5 55 E2
RX: 39954us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E
RX: 19874us C=48 S=Y A= 00 00 00 00 00 P(9)= 30 00 00 B3 13 36 DD 55 5E

A= 00 00 00 00 00
P: B0 00 00 F7 FB F4 F2 55 DD
P: 30 00 00 B3 13 36 DD 55 5E
P: B0 00 00 F7 FB F4 F2 55 DD
P: 30 00 00 B3 13 36 DD 55 5E
P: B0 00 00 F9 F7 F5 F5 55 DF
P: 30 00 00 B3 13 36 DD 55 5E

Bind packet
P[0]: 00
	| 20 not telem from the heli
	| 10 Lock
	| 08 H / 00 L
	| 01 Screw pitch (temporary)=long press up
P[1]: 00
	| 08 Auto land only set when unlock (toggle) = short press down
	| 20 Mode (toggle)
P[2]: 00
	| 02 Altitude hold
	| 03 Manual
	| 40 Flip (toggle)
P[3]: A[0]
P[4]: A[1]
P[5]: A[2]
P[6]: A[3]
P[7]: CH4H<<6|CH3H<<4|CH2H<<2|CH1H
P[8]: sum(P[1..7])

A[4] is missing from bind... How does the heli find it? scan?
Does the RF frequencies change?
When does the bind end? When the heli replies on normal RF channels?

OB1-OB2, 2025-01-14:
Hey, I'm Left_Drift from RCG. I got the heli and before even trying to fly, I updated the MPM and tried a bind. It seems to work:

- I turned on the heli, and the yellow light flashed fast.
- I pushed "bind" on the TX and the yellow light on the heli turned green - indicating bind-ness.
- I moved the sticks but nothing moved on the heli - am I supposed to have control over the collective/cyclic?

I powered down for now... I want to try flying with the original TX to make sure the heli even operates properly next.

Are there anything specific I need to set up in the model? Right now, I've just setup a simple 4 channel acro model in EdgeTX.

pascallanger, 2025-01-17:
Protocol works but still some to do:
- Add more than 2 TX IDs/RF frequencies -> I need people to do some dumps of other original TXs or send me their unneeded TXs.
- Find where the low batt signal is in the telemetry

closed 2025-01-19 by pascallanger