pascallanger/DIY-Multiprotocol-TX-Module #1058: E010/H36 with JR-NH010R9 boards don't bind with either E010 or E010r5 protocols

lyngklip, 2025-01-15:
I now have a couple of H36 models with boards that says "JR-NH010R9". They don't bind with either E010 or E010r5 protocols. They have altitude hold. I believe they have barometers. They also have empty footprints and the text "wifi" on the circuit boards. They are future proof I suppose. I bought them from Banggood in November/December.

pascallanger, 2025-01-16:
2 options for me to look at this protocol: send me a TX and model at my attention in France, give some money so that I can place a purchase on BG.

pascallanger, 2025-01-19:
Duplicated with issue #1048

closed 2025-01-19 by pascallanger

pascallanger, 2025-01-28:
Please try the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) 1.3.4.25 protocol H36, documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#h36---103