pascallanger/DIY-Multiprotocol-TX-Module #1059: Documentation: Clarifying of which protocols are included in each firmware type?

Patronics, 2025-01-16:
It's really unclear from the website, exactly which protocols are included in each of the firmware releases. Could the [supported protocols list](https://www.multi-module.org/basics/supported-protocols) be updated to clarify which protocols are included in which categories of release?

Also is there any recommended hardware for future-proofing that does support both sets of protocols? The documentation says that the FCC release is "Only available for CC2500 modules, where flash space permits all protocols to be included", but my understanding is that those modules are actually more limited than most, only having 1 of the "4 in 1" chips onboard, is that correct? If so, that wording should also probably be updated to clarify.

Since most of my projects use a dedicated receiver, I assume I'll mostly be covered either way by the "core" protocols category. The fact that the core protocols are included in both air and surface builds should also probably be emphasized more strongly in the documentation, I spent a while trying to figure out which category my receivers would fall into before finding the blurb in the "Release file changes" page that mentioned that detail.

Patronics, 2025-01-16:
A good start on this is covered in the pull request #1034, but I'd argue it's worth listing it properly in the aforementioned table, rather than expecting users to read through and make sense of validate.h in or to determine which protocols are included in each release.

pascallanger, 2025-01-19:
Feel free to submit a pull request with your proposed changes.