pascallanger/DIY-Multiprotocol-TX-Module #1060: DSMR -

IdefixRC, 2025-01-18:
Hello there,

I came across an interesting issue with my MT12 and a Spektrum surface receiver (SR6200A).

I have 2x MT12 and 2x Arrma Vorteks Cars with SR6200A receivers. 
One radio (EdgeTX v2.10.5 with MM v1.3.4.0) connected using the R1-F mode (DSM2SFC - 6 is missing from the FW build as listed [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md?fbclid=IwY2xjawH4PQxleHRuA2FlbQIxMAABHTl5j_nDFiyzNWooDAqCAT9TFvdtiHjN3km_Pi8DolJ088-6bl665_7rhw_aem_kEMziTMMBxVJxfxLsCDlkg#sub_protocol-dsmr_1f)) after a few tries and works although Telemetry comes and goes all the time. 

The 2nd MT12 (same SW version as the other) and Vorteks connect when binding. Telemetry comes up, drops and never comes back up again and when I cycle the power on the receiver, the receiver goes back into binding mode directly when powered back on.

In addition the issue that only receiver number [00] can be used remains also in 1.3.4.0 - see video of the issue from another user here: https://www.youtube.com/watch?v=KLTIEe_VTGY

Any help would be highly appreciated ! Thanks a lot

IdefixRC, 2025-01-18:
Update to my earlier post: I tried different versions of the MM FW and got it to bind with version 1.3.3.14 of the Multi Module FW, higher versions all ad the same issue of connecting but not keeping the bind after a receiver power cycle.

It even works now if I select the [01] receiver number which with the other FW versions resulted in no connection at all.
Maybe this helps to troubleshoot. 
Thanks

pascallanger, 2025-01-19:
I don't have any DSMR receiver to test... Can you narrow down the exact version from which the issue started?

IdefixRC, 2025-01-29:
Hello Pascal,

Here is what I have tested so far:

- Version 1.3.3.14 works
- Version 1.3.3.20 broken

Hope this helps

pascallanger, 2025-01-29:
@benlye forget... I think I found the issue.
I'm pretty sure v1.3.3.20 works fine... It broke on v1.3.3.33 when DSM clone was introduced...
@IdefixRC Please try the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.26 with your DSMR receiver including multiple receiver numbers.

IdefixRC, 2025-01-30:
@pascallanger, thanks for the quick fix.
I have tested v1.3.4.26 with 2 different MT12 and 2 different Spektrum SR6200A receivers (one of the receivers previously only worked with v1.3.3.14 and would not retain the bind after power cycle with the other versions).
Both are working well. I tried multiple binds with model number 00, 01, 02, 03 and all 4 worked fine (connects and stays connected after power cycle).

Only telemetry is still patchy. I might be imagining things but when I bind with Model Number 00 or 01, telemetry drops out sporadically but is semi stable (stays on for 15-30 seconds before dropping). If I bind with number 02 or 03, telemetry comes and goes every few seconds. Not sure how I can help you troubleshoot the telemetry part of the connection.

Thanks !

pascallanger, 2025-01-30:
That's surprising as I'm talking to someone else which has no telemetry drops on whatever model number. He has let his radio and car turned on for 30 minutes without a single telemetry lost message while walking around with the radio in his house to test at different distances.
He is using SR215 and SR315 in his tests, that's the only difference I can see. So the issue seems to be targeted to your RXs model...
Anyway, I've done a slight change. Can you try the latest test version 1.3.4.27?
If it doesn't work better, you'll need to send me on of these receivers or make a donation so I can purchase one.

IdefixRC, 2025-01-30:
Thanks @pascallanger.

I have tested 1.3.4.27 and it has the same issue unfortunately. :-(
I have made a video showing the telemetry issue. You can check it here: https://youtube.com/shorts/rW6Nb3ZMbwI?feature=share

Does the other user use the Radiomaster MT12? Here is another report of a similar telemetry issue (Radiomaster MT12 and a Spektrum SR315 Receiver): https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/992
If he uses a different TX, maybe it is not the receiver but has something to do with the TX?

Another user in the link above states that it works better in low power mode vs normal mode. Here is another user (MT12 and SR6200A) with the same telemetry cut out issues: https://www.arrmaforum.com/threads/dx3s-avc-knob-and-replicating-advanced-avc-settings-with-the-radiomaster-mt12.73559/post-1190707

pascallanger, 2025-01-30:
MT12 or any other radio doesn't change anything...
In the other report, it looks like the SR315 is connected to an ESC with telemetry. Is it your case too? If it is then can you try without connecting this ESC and power the RX with just an external battery?  Looking at your video it looks like your telemetry is stable for 30 seconds then starts to drop and go, that might be the time the ESC takes to send back it's data...

IdefixRC, 2025-01-31:
Hello @pascallanger,

thanks for the continued effort on this one.
Correct, the Receiver is connected to a Spektrum SMART ESC with telemetry from the ESC.
Note that the SSR6200A receiver is a AVC receiver with integrated Gyro and sends telemetry data on it's own as well.

I have done the following:
1) disconnected ESC
2) only left steering connected to the receiver
3) powered the receiver via an external receiver pack
4) rebind receiver with MT12 on 1.3.4.27 - model # 01
5) similar outcome as before with ESC connected

Please see the video here: https://youtube.com/shorts/kMxNn5pEVag

for the 2nd test I used the same setup as above but rebind the receiver on model # 02
Although I believe the telemetry lost/reconnect is way more frequent with this model number I wonder how model numbers can impact the telemetry function? Maybe this helps to narrow it down.

Please see the video here: https://youtube.com/shorts/yFRZZUtY7v8

ken830, 2025-01-31:
1 tested both 1.3.4.26 and 1.3.4.28 on my RadioMaster RM 4-in-1 External Module connected to my MT12. This is connecting to an SR6200A. Still get telemetry cutting out. In the telemetry screen I can see the RSSI will drop from 90 to 0 instantly. With 1.3.4.26, I feel like it's stuck at 0 when it happens, but with 1.3.4.28, it comes back and holds it more steadily. If I'm in the next room, holding the radio in certain positions, I can get telemetry to stay connected for quite some time (~30s to 1m). Many times (but not always) I seem to be able to bring it back by just wrapping my hand around the external antenna. When telemetry comes back online, the RSSI jumps from a steady 0 to 90 or so...

Dan0111, 2025-01-31:
Hi,

I have exactly the same problem. I can reproduce it with the Spektrum SR6200A and the SR6100AT. Both are telemetry receivers with integrated AVC. With an SR515, which I also have here, it works absolutely without problems and the telemetry also remains constantly active. So it seems to be somehow due to the AVC RX.

pascallanger, 2025-01-31:
I have no idea where to look at. I have reports that SR215, SR315 and SR515 are working fine. I've used a SR215 during the dev of this protocol so that's what I would expect.
At this stage and without getting my hands on a SR6xxxx I don't see what I can do...

Dan0111, 2025-01-31:
Which country do you live in? I hope there is someone who can send you one to test.

pascallanger, 2025-01-31:
I'm located in France.

IdefixRC, 2025-01-31:
@pascallanger, happy to chip in for you to purchase one.
@Dan0111 & @ken830, you guys willing to chip in? 
If you post a Paypal payme/fund me link here we can see who else is willing to support?

Probably not the cheapest but Amazon France seems to have them here for 92 Euro with free shipping: 
https://www.amazon.fr/Spektrum-Radio-Accessoires-SPMSR6200A-Multicolore/dp/B08LF25NQG/ref=sr_1_42_mod_primary_new

Horizon Europe is cheaper but not sure about the shipping costs: 
https://www.horizonhobby.de/fr_DE/product/sr6200a-dsmr-6-channel-avc-surface-receiver/SPMSR6200A.html

ken830, 2025-01-31:
@pascallanger : Absolutely.

Dan0111, 2025-01-31:
I am from Germany. Shipping to France and back to me is 30€. I have an SR6200A here that I don't need at the moment. Would that be an option? And when you're done @pascallanger  will you send it back to me?

pascallanger, 2025-01-31:
I'm wondering if it would be better if I purchase this 2nd hand DX3+SR6200A so that I have a real DSM radio to play with too that I can look into on top of the receiver. I'm thinking the DX3 is a telemetry capable radio of some sort since it has some SMART battery level indicator but let me know if that's not the case since I don't really know the Spektrum world.
65€ + some shipping: [image]

IdefixRC, 2025-01-31:
That might be a good idea.
The DX3 is telemetry capable. With a bluetooth module you can view the data on the spektrum app (I assume it has a serial interface to connect to the BT module which might make it easy to read the telemetry data).
Do send us your paypal link / fund me page and I will chip in.

From the Spektrum website: 
It takes only a few simple, affordable upgrades for your DX3 Smart to provide many of the same capabilities as radios that cost much more. Just download the free Spektrum™ Dashboard™ app to your mobile device. Add the BT2000 Bluetooth Module (sold separately) to your transmitter. Then, with no additional sensors or modules, you’ll be able to use the transmitter’s Smart ESC compatibility to receive important telemetry data in real-time, right at your fingertips — and can also set up warning alarms and enjoy new programming options!

pascallanger, 2025-01-31:
My paypal account is my email address pascal _ langer @ yahoo . fr (remove the spaces)

IdefixRC, 2025-01-31:
@pascallanger, sent you 40 Euro to contribute.
Hopefully it works out.

Dan0111, 2025-01-31:
I have also sent you @pascallanger  40€. Good luck and thank you very much!

ken830, 2025-02-01:
I just sent $40... Just want to say thank you for your efforts so far and thank you in advance for your continued efforts.

ken830, 2025-02-01:
@pascallanger : while we wait for you to get the Spektrum transmitter and receiver, is there anything we can do on our end to help? I have the DX3 and SR6200A and an oscilloscope with logic analyzer. I don't have an SDR though. Have my eye on the HackRF with the Portapack H4M.

pascallanger, 2025-02-01:
You could open the dx3 and look at the RF modulator. I think it would be good to do a dump there if we can.

jslatten, 2025-02-01:
The DX3 transmitter is going to be difficult to see and use the telemetry as compared to a dx5c or dx6c transmitter as it only has limited telemetry through Spektrum's android app...and the android app is pretty crappy. It wouldn't work on the latest android version. Luckily I had a old phone with android 11 where it would function. 

Also, I was the one who submitted the other issue about inconsistent telemetry. I ended up learning that I can get telemetry fairly consistent, albeit slower/less frequent updates than the spektrum transmitter, when the transmitter and receiver are at a farther distance away from each other. It's almost as if the transmitter -and multi-module are experiencing RF overload or something from the telemetry transmission. In any case....make sure you are at least 10-15 feet (3-5 meters) away from your RC vehicle and see if it improves the consistency. I would love for it to work up close but outside of the less frequent updates the telemetry is usable most of the time. My only other gripe with the Spektrum receiver and Radiomaster Mt12 / Multi-module combo  is range seems to be about 2/3 or 1/2 of my Spektrum transmitter.

jslatten, 2025-02-02:
Also...I unfortunately caught this thread too late as I have a couple of DX3 transmitters sitting on my shelf that I don't use. I could have offered one up for you.

pascallanger, 2025-02-06:
Still waiting for the delivery of the RX + DX3... Should be there Saturday.
I don't care if the DX3 is doing a good job at decoding the telemetry, I only need it to receive it.
Also for DSMR I only have a couple of IDs/RF frequencies. Do you have a RX or can send me a RX that still have a CYRF6936 that I can connect to the SPI?

IdefixRC, 2025-02-20:
Hi @pascallanger did the TX/RX arrive in the meantime and did you manage to take a look yet? Keen to hear what the issue might have been.
Thanks !

pascallanger, 2025-02-20:
@IdefixRC I've received the parcel but no time to look into yet. I'm working on multiple protocols at the same time and my personal life also getting in the middle so please be patient ;-)

IdefixRC, 2025-02-20:
No worries @pascallanger. Looking forward to hearing what you find. Thanks

pascallanger, 2025-02-26:
Please try the [latest test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.41

ken830, 2025-02-27:
@pascallanger : Thanks for the update. I was looking forward to it since I saw you cleaning up a lot of the issues last night. I have just the SR6200A that came with my Kraton 6s EXB v2 and just tried the v1.3.4.41 with my MT12+4-in1 and I still get telemetry cutting out constantly. Were you ever able to duplicate the issue with the existing/older firmware? There must be something different between your setup and mine.

I do notice it being slightly different. When turning on the transmitter and receiver next to each other, I don't get any telemetry at all whereas before, it would just cut in and out. Now it's just nothing. But when I walk away, it would start to come in and out... and it feels like a lot more frequently compared to before. Also, if I hold my hand on the antenna, it instantly gets telemetry, but still not very stable at all.

pascallanger, 2025-02-27:
That's surprising and disappointing...
I was having the same issue has reported on it but it did go away after some tuning. I've only tested on my table so the TX and RX are 50-60cm away from each other with not a single loss.

ken830, 2025-02-27:
I want to hear about others' experience to see if I'm doing something wrong. I tried enabling the "low power" option but it didn't seem to do much of anything. I didn't think rebinding would make a difference, so I didn't do that. Not sure what else I can try.

IdefixRC, 2025-02-27:
I will give it a try on the weekend and report back.
Thanks for the effort so far @pascallanger

ken830, 2025-02-28:
> > I want to hear about others' experience to see if I'm doing something wrong. I tried enabling the "low power" option but it didn't seem to do much of anything. I didn't think rebinding would make a difference, so I didn't do that. Not sure what else I can try.
> 
> Try to rebind with different RX numbers. I haven't selected one in particular but may be the RX doesn't like some of them... There are only 20 in total so it shouldn't be too long for you to test them all.

I just spent some time testing rebinding with different Rx numbers. Could you clarify why you say there are only 20, when it allows me to select anything between 00 and 63? I didn't try all 64, but it was originally on 00 and I tried a bunch, like 02, 03, 04, then I skipped around to like 20 and 25 and then jumped straight to 63. I'm not 100% sure, but it felt like they do feel a bit different from 00, but it could be my imagination because it still doesn't really work very well. Even when I get telemetry, I don't see most of the sensor values updating.

Looking at the MDL Telemetry screen, I see sensors 1 through 40 discovered... I seems like I can usually get sensor 15 (TRSS) coming through (the asterisk flashing). Sometimes, I would only get 15 through 19. Sometimes I would only get 15 through 25. There was one time (I forget which Rx number I was on) where I got 26 (GYRX), 27 (GYRY), & 28 (GYRZ) refreshing at a rapid rate and it was pretty steady, which was great, but I got nothing else.. Getting these partial telemetry usually happens when I walk away some distance after binding. I usually get nothing (or just 15 (TRSS) when I'm next to the receiver.

Is this a clue?

jslatten, 2025-03-03:
I tested firmware 1.3.4.42 with my MT12 / Internal 4 in 1 multi-module and SR315. The new firmware seems to be working pretty well. Although while telemetry now appears to work up close and is updating very frequently, telemetry seems like it may have a bit less maximum range now and it cuts out when my RC car gets maybe ~15-20 feet (~5-7 meters ) or so away....then there is a black out of telemetry for a little bit until my car gets closer. This is not a huge deal as you probably can get by with that range at a track or while bashing around the yard. The new firmware also seems to fix binding to multiple separate IDs. I've only tested 2 ids (ID 0, and 1) but both work perfectly. While maybe the firmware could use a bit of fine tuning (if possible), it is otherwise working pretty well now. I just gave a quick glance at all of the telemetry available and  it seems to be inline with what is available with the Spektrum transmitter.

Awesome job @pascallanger! It appears you've gotten it pretty darn close in terms of functionality as compared to the official Spektrum DX6C transmitter. Your hard work is greatly appreciated.

ken830, 2025-03-03:
@jslatten : Hmm.. Wonder if your experience is different from mine or we just have different expectations? I'm testing indoors right now and your 15 feet is like the next room over, which is kind of when I start to see things change too, although I never see many sensor updates (5 or so at a time). I guess one difference is that you have internal 4-in-1 and I have an external 4-in-1. And  yours is the SR315 (no AVC) and mine SR6200A (with AVC).

Are you seeing all the telemetry values come through in the MDL->Telemetry? How many sensors are you reading?

IdefixRC, 2025-03-03:
Sorry, could not test on the weekend due to last min commitments. Have blocked time for coming Saturday and will update.

ken830, 2025-03-03:
I mounted a magic arm on my controller to clip my phone to get a video of what I'm seeing. Sorry for the slightly shaky video. Even with stabilization turned off, the Pixel still seems to try to stabilize. And sorry for the vertical framing, it was the best way to capture the screen and buttons.

I started the video next to the receiver and then walked around my house. Depending on where I stand and orientation of the transmitter, I will get various sensors. I don't walk more than 15-20 feet away.

You can see that only a fraction of the telemetry sensors come through at any given time. And most of them are not updating regularly (they have a bracket, which appears to indicate stale values). Even so, I can sometimes get enough updates (flickering between brackets and no brackets) for the values to be updated at a fairly rapid rate, like the gyro it G sensors when I shake the vehicle (as I show toward the end of the video). But when I get those, none of the other sensors are updating at the time.

[https://youtu.be/3OOv6xPhX4o](https://youtu.be/3OOv6xPhX4o)

jslatten, 2025-03-06:
> [@jslatten](https://github.com/jslatten) : Hmm.. Wonder if your experience is different from mine or we just have different expectations? I'm testing indoors right now and your 15 feet is like the [...]

I'm pretty sure all of the telemetry metrics are coming through. As I mentioned above...I think there are still some range differences with telemetry and probably a bit less range on controls, but overall I think its much better than it was previously especially since telemetry seems to be fairly consistent now and you can now program multiple DSM receivers since the programmable RX ID is now working as expected.

ken830, 2025-03-06:
> I'm pretty sure all of the telemetry metrics are coming through. As I mentioned above...I think there are still some range differences with telemetry and probably a bit less range on controls, but [...]

Did you see my video? I have a very different experience from what you're describing.

jslatten, 2025-03-07:
Sorry, The video wasn't fully loading for me previously, however I was able to view it now. I had my alerts turned off, but after turning them on I see what you are saying now. When alerts are turned on, the transmitter continues to annoy me with the "Telemetry Recovered" and "Telemetry Lost" messages over and over. Just like in your video. There is no way you could drive with the alerts turned on. On the other hand, when alerts are off it just appears to update somewhat consistently, however you can tell it is actually missing some updates of the telemetry values as there are some occasional gaps of updates. It wasn't so apparent until combined with the annoying voice alerts.

ken830, 2025-03-07:
When sensor values aren't updated, they get a bracket (I assume to indicate stale data). Updates come in with an * next to the value. I only see a subset of sensors coming through at any given time, and most values have very very long periods of no updates -- It appears to be infinitely long, unless I re-position the transmitter relative to the receiver, which would get me some other set of sensor values, but none of the rest.

IdefixRC, 2025-03-08:
@pascallanger, could you please repost the link to the latest test version? I tried to access via the link above but get a nginx GW timeout error on multiple PCs (tried via VPN as well). Thanks

ken830, 2025-03-08:
I'm seeing the gateway timeout error too. But I also found the test builds (including the DSMR test build) here: [](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/actions?query=branch%3Amaster++is%3Asuccess++event%3Apush)

IdefixRC, 2025-03-08:
> I'm seeing the gateway timeout error too. But I also found the test builds (including the DSMR test build) here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/actions?query=branch%3Amaster++is%3Asuccess++event%3Apush

Awesome. thanks

IdefixRC, 2025-03-09:
Hello @pascallanger,

finally managed to test version mm-stm-serial-aetr-sfc-v1.3.4.41.bin on the MT12 with internal multi module and SR6200A receiver.
Unfortunately I do not have better news compared to the rest.

- Telemetry connection seems to be established and stays for a few seconds before it drops
- However even while telemetry seems to be established the discovery only yields 5 sensors
- After the telemetry drops and comes back the radio discovers additional sensors
- Looking at how they update however it seems that only sensor 1 get's frequent updates, 2-5 get them sporadic and 6+ do not seem to get them at all or at a much lower rate

I have made a video showing the connection, sensor discovery and update frequency.

IdefixRC, 2025-03-24:
Hello @pascallanger,

Did you manage to investigate further?

Thanks

pascallanger, 2025-04-02:
I checked again today. Testing on my dev platform worked without any issues. So I decided to do a video to show you guys that everything works on my side. For this I downloaded the GitHub builds like you do and installed them on my internal module. The issue of telemetry in/out was as expected with v1.3.4.31, I then flashed v1.3.4.49 and was surprised that it doesn't change anything since the telemetry still goes in/out. So now, I have something to troubleshoot for, somehow when I'm in debug mode the telemetry is working fine (not on v1.3.4.31 but on v1.3.4.49) but not using the standard builds...

IdefixRC, 2025-04-02:
Thanks for the update and for keeping on it @pascallanger. Sounds like a solid lead and looking forward to hearing what you find.
Thanks

pascallanger, 2025-04-02:
Try the test version v1.3.4.50. I know why I hate this DSM protocol... It's just too tight on timing...

IdefixRC, 2025-04-03:
Woooohooooo @pascallanger, Well done !! It seems to work now 🥳🥳

Did a quick test:
- flash V1.3.4.50 to the MT12 internal module
- started the car
- Telemetry stays on
- All sensors discover on the first try
- Sensors are updated without interruption
- Test within the same room (did not test range yet) and zero disconnect

Video: https://youtube.com/shorts/sR8OfqWGK6A?feature=share

You might have solved it. Thanks a lot !!

ken830, 2025-04-04:
Thank you @pascallanger !

I just tested it indoors and it seems to work! This is so great and I'm excited to get some custom telemetry screens created for my Kraton 6s!

closed 2025-04-28 by pascallanger