pascallanger/DIY-Multiprotocol-TX-Module #1063: 1:43 Rc Drift Car is supported ?

LilTeo14, 2025-01-22:
Pictures:

[image]

[image]

Transmitter Board:

[image]

Receiver Board:

[image]

Aliexpress data:
Brand Name: SMODELF
Model Number: JBL-4303

If it is not supported i already have debug firmware to dump over the air and have a analyzer too.

Thanks for help.

pascallanger, 2025-01-22:
I don't think it's supported but feel free to try all the SFC protocols and sub-protocols.

If you have a logic analyzer that would be perfect. Connect it on these 3 pins + GND:

[image]

LilTeo14, 2025-01-23:
Here is the data from logic analyzer.

https://usmcl-my.sharepoint.com/:u:/g/personal/mateo_morales_usm_cl/EUXc4VDJHHRKo9rgKDj9ESsBLZ2tDzrWaFPluHNe8_kKFw?e=fYhWQ6

Ch0: CSN, Ch1:Csk, Ch2: Data.

M-i-c-h-a-e-l1, 2025-01-23:
Hello,

I have the same car, but with a gyro installed. The Transmitter has an extra potentiometer to change the sensitivity of the gyro.

[image]

[image]

[image]

I am willing to test the protocol on my RM 4in1 radio.

Kind regards
M

LilTeo14, 2025-01-23:
The TX keep blinking until the car is power on.

"Bindmode" Tx is the whole time in bind mode "Led blinking"
"Bindmode+Bind" Tx start in bind mode and after 1 second aproximated the car is turned on and the Led stop blinking

https://usmcl-my.sharepoint.com/:f:/g/personal/mateo_morales_usm_cl/EmGWskJlxtdFsW5wWM53tFQBy3OHnQC4ilQJs_frqeEJxw?e=MbbkKq

pascallanger, 2025-01-23:
It looks good but I need at least Bindmode to start from the TX power on (start the capture before turning on) to see the init of the XN297.
Please also do a dump of the normal mode from the TX power on. If the car needs to be turned for the TX to stop blinking then do a dump as well of this part.

M-i-c-h-a-e-l1, 2025-01-23:
Hello,

The Transmitter also has the following functions:

1. Steereing left - right (and trims)
2. Forward and reverse and brake lights
3. Lights on/off
4. Three levels of speed/power: 20%, 50% and 100%. The speed/power levels are selected with the arrow up/down on the transmitter. If 20% is selected a green LED lights up, 50% orange LED and 100% red LED. I don't think that this is necessary to implement in the protocol as this can be done with a 3 position switch on the radio.
5. Gyro sensitivety (for cars with a gyro)

Kind regards
M

pascallanger, 2025-01-23:
I'll need a different capture for every single channel and feature. So one stuff moving at a time so that it is ready to identify.

LilTeo14, 2025-01-23:
"BindedFromTxPowerOn" I turned on the car, start recording, turned on radio.
"BindModeFromTxPowerOn" Start recording, turned on radio with car off.
Both files uploaded to the same folder!

pascallanger, 2025-01-24:
I've started to look at it but I need these other dumps to understand the protocol, one file per dump:
 - turn on in normal mode with car off
 - turn on in normal mode car on
 - normal mode steering slow move from left to right (making sure trims are centered first)
 - normal mode throttle slow move from reverse to forward
 - lights
 - speed settings 20%, 50% and 100%
 - gyro
 - everything else you can think of

pascallanger, 2025-01-24:
Protocol uses XN297 1Mb Scrambled

Bind
---
Bind is in 3 steps
**Step 1**
Tx sends packets and listens for an answer
TX Addr A7 07 57 A7 26
RF: 7, 39, 69 = 0x07, 0x27, 0x45
Packets TX every 12700µs
RF=7   P(8)=CB 03 A5 00 00 00 00 CF
RF=39 P(8)=CB 03 A5 00 00 00 00 EF
RF=69 P(8)=CB 03 A5 00 00 00 00 0D
....

Packet
P[0] = ?
...
P[7] = next RF mixed with CRC?

**Step 2**
RX from car
P(7)=CB 03 A5 9D 05 A2 68

Packet
P[0] = same as TX
P[1] = same as TX
P[2] = same as TX
P[3] = RXID[0]
P[4] = RXID[1]
P[5] = RXID[2]
P[6] = crc?

**Step 3**
Tx sends packets
TX Addr A7 9D 05 A2 26 -> A7 RXID[0] RXID[1] RXID[2] 26

RF: 35, 45, 75 = 0x23, 0x2D, 0x4B
Packets TX every 2408µs sometimes 3990µs for a packet

RF=35 P(8)=CB 03 A5 00 00 00 00 EB
RF=45 P(8)=CB 03 A5 00 00 00 00 F5
RF=75 P(8)=CB 03 A5 00 00 00 00 13
...

Packet
P[0] = ?
...
P[7] = next RF?

LilTeo14, 2025-01-24:
Hello there!

First of all, thank you so much for your support and guidance. I want to clarify a few points regarding the dumps you requested:

- My radio automatically turns on in bind mode, so there isn’t really a separate “normal mode” when powering it on.

- Because I was out today, I’ll carry out the tests tomorrow to gather the data you need.

- “Turn on in normal mode car on” corresponds to what I did in the file BindedFromTxPowerOn.

- “Turn on in normal mode with car off” corresponds to what I did in BindModeFromTxPowerOn.

For the remaining recordings, do you want me to start capturing data before powering on the radio, or after the radio is already on and bound to the car? Also, is there a recommended length for each recording, or should I record as much as possible?

Thank you again for all your help. I’m really grateful for your time and expertise, and I look forward to getting these dumps done.

Best regards,
Mateo.

pascallanger, 2025-01-25:
To determine the channels and features, just do a capture when the radio is already bound to the car (no need to start from power on).
For the length, it all depends of the action. When it's a channel the capture should cover the full move of the channel. For example for steering the capture should have a slow move from mid->left->right->mid which will be quite long but for lights which is a button I just need a short capture where you press the button one time.

Do you have more than one car? if yes then a capture “Turn on in normal mode car on” of every car would be useful. 

We'll see if we can get to the bottom of this protocol but there are many unknowns right now. The most annoying is the last byte of the packet which looks like to have a crc and next channel which calculation is un known yet and could prevent the car to accept packets. The second is the RF channels of normal mode which might be based on the TX ID and RX ID (car) so locking it down to only your set... But I'm getting ahead of myself, let's see the captures and go from there.

M-i-c-h-a-e-l1, 2025-01-25:
Hello,

I noticed the following today when driving with 2 cars: Sometimes when the trasnmitter A and car A are switched on and I switch on car B (and not the transmitter B), the two cars can be driven simultaneously by transmitter A. This already happened twice. I would need to do some tests to reproduce this consistently. No binding of car B is necessary.

Instead of soldering wires to the XNX297LBN could I also use a [SOIC8 SOP8 test clip](https://www.amazon.de/WINGONEER-SOIC8-EEPROM-Circuit-Programmierung-Adapter-SOP8-Test-Clip/dp/B012VSGQ0Q/ref=sr_1_2?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2HFISLQ1VN7DC&dib=eyJ2IjoiMSJ9.1f8QZ4oywar2_MC4cAjFDW5hloCMADy9ygIuzT_7Ajuj9OZnReDwFJo2naU7T-H5FshBoE2bWZHGRBcM7HfoCvRUWt3mLsRVqXVgNKa9QV-IpIdSS37tRw4BX9f7Di3KoLchaitEz6FMUnpgZJZEYMOfMg1IOQcwLCp9GOqcuWYbmeC5jg8rFCXGE2btawkwsVXJdLWI-avc5bLarYvyHhRUReu60EOp1SCc3dBlZ1f_oTG8g7XPU1046HXoLGgaxFgMF4XUBKuKyC_x5quP5d5bJI-nXdAsyInjSatwHFhPKxJSBOjSUDAC_bN4UetuvtAMArQjF8S5w8RHOADXMeqHW785zNDu1dVQV3pQQxTWrc5v34gv8ukGhN5iu65YLK-dhCTjdxjjxtn-7NnGg6c0yXtzmC6hcNYWrPM1asAd631-puEwskZ1XDUTpwcT.e0QjxGnrSAinqh7Fh2HW-69PN1EJj7-FeRNHKkq1_KY&dib_tag=se&keywords=SOIC8+SOP8+Test+clip&qid=1737828526&sprefix=soic8+sop8+test+clip%2Caps%2C81&sr=8-2) to make some dumps? I have never done this before, so no idea if this will be a success ...

LilTeo14, 2025-01-25:
I believe it’s possible, and I actually think it’s a great idea! However, I’m not an expert in this area. You would just need a logic analyzer. I use this one:

[Logic Analyzer on Amazon](https://amzn.eu/d/bu8DK6m)

It’s not the best quality, but it works. Of course, a higher-quality analyzer would be preferable if possible.

LilTeo14, 2025-01-25:
> I've started to look at it but I need these other dumps to "possibly" understand the protocol, one file per dump:
> 
> * turn on in normal mode with car off
> * turn on in normal mode car on
> * normal mode steering slow move from left to right (making sure trims are centered first)
> * normal mode throttle slow move from reverse to forward
> * lights
> * speed settings 20%, 50% and 100%
> * gyro
> * everything else you can think of
> 
> Please preconfigure the SPI analyzer like this before the capture: [image]

I've completed the tests and added the dumps to the OneDrive folder.

For the "lights" test, I toggled the switch 4 times. I also added an additional test for the hazards since there’s a button that activates or deactivates them. For this test, I pressed the button 4 times as well. Let me know if you need anything else!

pascallanger, 2025-01-25:
Based on your captures, I've done the easy part of decoding the normal packets content for the different channels and switches. I've updated my previous post https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1063#issuecomment-2612928112 with the information and will continue to update it with new findings. 
I don't think I'll have the time tomorrow, but the next step is to figure out the crc.

pascallanger, 2025-01-26:
> I noticed the following today when driving with 2 cars: Sometimes when the trasnmitter A and car A are switched on and I switch on car B (and not the transmitter B), the two cars can be driven [...]

This shouldn't be possible with what I see on the none gyro TXs/cars since the address used to transmit is based on the car ID. So the car only listens to its own address nothing else and during the bind the car learns the TX ID, so even if 2 TXs are powered on when you power the car only one will be able to control the car.
Gyro cars might use a different protocol as there is no obvious space for an extra channel. I mean there are some free bits in P[4], P[5] and P[6] but it won't follow their current scheme on how they transmit channels.

M-i-c-h-a-e-l1, 2025-01-26:
Hello,

I hope that I will be able to do some dumps on the cars with the gyros. I have the logic analyser, but I am waiting for the test clip.

Could you please let me know if the following settings are ok for Logic 1.2.14 to do the dumps?

[image]

How do I connect the logic analyser correctly to the pins of the XN297LBW: MOSI (= Pin ?), MISO (=Pin ?), Clock (=Pin ?) and Enable (= Pin ?).

Please be patient with me, steep learning curve (I hope).

Kind regards
M

pascallanger, 2025-01-26:
> Could you please let me know if the following settings are ok for Logic 1.2.14 to do the dumps?

yes

> How do I connect the logic analyser correctly to the pins of the XN297LBW: MOSI (= Pin ?), MISO (=Pin ?), Clock (=Pin ?) and Enable (= Pin ?).

[image]
MOSI = DATA, MISO not connected, Clock =SCK, Enable = CSN and don't forget to connect GND to the Bat-. Anyway don't worry, it's easy to find out how you've plugged it, so really connect the 3 pins in green and GND.

pascallanger, 2025-01-26:
I've found a way to calculate the crc. It works for all the packets sent by the TX but the same calculation doesn't work on the packet received by the car (strangely). There is a constant in the crc which I don't know if it changes based on the TX ID or something else but 0x55 looks like a good init for the calculation.
Still completely unknown, the RF channels use for normal packets...

Anyway, I can code the protocol and get it to work with @LilTeo14's car. But don't expect too much as I fear it will only work with that TX ID/RF and car for now...
When there will be more TX and cars we might be able to find out more things.

pascallanger, 2025-01-27:
You are lucky! A friend has one of this car (without gyro) which he dropped on my desk this morning.
I've written the protocol using what I knew so far and some of his radio/car details. Tests were conclusive, driving the car around with my TX16s.
There was still one issue like said previously which was the RF channels in normal mode. I've changed the TX ID which is easy as I'm "owning" it but it had no impact on the RF channels so the conclusion is that the car/RX ID is the one that "decides" the RF channels. So I had to write the code of a car receiver, to push some car IDs to the original TX and get back the RF frequencies by scanning. After a few hours, I've been able to figure out the link between the RX_ID and the 3 RF channels of the normal mode. 
So basically any car can be controlled from a Multi module which can use any TX ID.

Please try the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.24 SFC and report. The doc is [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#jiabaile---102).
Couple of things:
- The car remembers the TX ID so in theory you don't need to bind it every time. But I haven't implemented it so you need to bind each time which is not difficult with the bind on channel feature.
- It might work or not on gyro cars.
- The ST_TRIM channel is useless in my opinion, better use the radio trims. I've left it just in case the gyro cars need it.

M-i-c-h-a-e-l1, 2025-01-27:
Dear Pascal,

Great news and many thanks for your work! I will test it on my gyro car this week and I will let you know the reslts.

Kind regards
M

pascallanger, 2025-01-28:
@LilTeo14 Have you tested?

LilTeo14, 2025-01-28:
> @LilTeo14 Have you tested?

Hello!
I was away for a couple of days, but in just a few hours I’ll be able to sit down and test the protocol. Thank you so much for your patience, and I’ll be sure to keep you updated on any progress I make.

LilTeo14, 2025-01-28:
> You are lucky! A friend has one of this car (without gyro) which he dropped on my desk this morning. I've written the protocol using what I knew so far and some of his radio/car details. Tests were conclusive, driving the car around with my TX16s. There was still one issue like said previously which was the RF channels in normal mode. I've changed the TX ID which is easy as I'm "owning" it but it had no impact on the RF channels so the conclusion is that the car/RX ID is the one that "decides" the RF channels. So I had to write the code of a car receiver, to push some car IDs to the original TX and get back the RF frequencies by scanning. After a few hours, I've been able to figure out the link between the RX_ID and the 3 RF channels of the normal mode. So basically any car can be controlled from a Multi module which can use any TX ID.
> 
> Please try the protocol JIABAILE included in the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) v1.3.4.24 SFC and report. The doc is [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#jiabaile---102). Couple of things:
> 
> * The steering gives more wheel deflection than the original TX.
> * The car remembers the TX ID so in theory you don't need to bind it every time. But I haven't implemented it... So you need to bind each time which is not difficult with the "Bind on channel" feature.
> * It might work or not on gyro cars.
> * The ST_TRIM channel is useless in my opinion, better use the radio trim. I've left it just in case the gyro cars need it.

When I selected the model, I had the car turned on, and the radio automatically bound with the car. After that, every time I turn on the car, it connects automatically to the radio, even if I turn the radio off and back on. As for the channels, everything works perfectly. I tested all the features: speeds, lights, indicators, steering, and throttle, and everything works flawlessly.

pascallanger, 2025-01-28:
The thing I haven't implemented is that the module sends bind packets when the protocol is selected (power on or a bind request using either bind through UI or bind on channel). I could save the car ID in "eeprom" and therefore not launch any bind unless explicitly requested which could be useful if there are multiple people with the same cars around and prevent to bind everything around you. Let me know if this is worth it. Also I'm sending bind requests until I receive an answer from the car. I might want to only do that for 1 sec to make sure to not bind everything around too.

Otherwise that's good that everything works with your module ID and car ID!

Waiting to hear more from @M-i-c-h-a-e-l1 and his gyro cars.

pascallanger, 2025-01-29:
I've implemented a 2 second bind timeout and the storage of the car ID. So bind packets are only sent/needed when you bind a new car. You must use a different RX number per car, otherwise you just overwrite the previously saved car ID.

Please try the [test version](https://downloads.multi-module.org/latest-test/) 1.3.4.25 SFC. [Documentation](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#jiabaile---102) updated.

M-i-c-h-a-e-l1, 2025-01-29:
Dear Pascal, dear LilTeo,

I tried both versions 1.3.4.24 SFC and 1.3.4.25 SFC and it is not possible to bind the radio to the gyro cars. I tried different receiver IDs, no binding.

The firmware tested was mm-stm-serial-aetr-sfc-v1.3.4.25.bin on my RM Zorro 4 in 1.

Kind regards
Michael

pascallanger, 2025-01-29:
I'm not too surprised since like I said the protocol doesn't really allow for the gyro channel.
I need either a dump over the air or a capture of the SPI bus to see what has changed (hopefully not much).

M-i-c-h-a-e-l1, 2025-01-29:
This is the hardware inside of the transmitter of the gyro cars:

[image]

It seems to be different from the non-gyro cars.

Kind regards
Michael

pascallanger, 2025-01-29:
There is no SPI bus to sniff since the PAN chip is an all in one integrating micro controller and RF.
You need to use capture over the air. This is done by connecting serially the multi module to a PC and load a debug firmware. If you have an external module, it's easy since you just need to leave the USB cabled plug, if that's an internal module you need to solder a serial converter to the module board. Is that something you can do?

M-i-c-h-a-e-l1, 2025-01-29:
Dear Pascal,

I do not have an external module and unfortunetely I can not solder to the internal module :-(

If someone else is willing to to the dump and you develop a new firmware I am of course ready to test.

Kind regards
Michael

pascallanger, 2025-01-29:
Looking at AliExpress a car with gyro costs 40€. If you are willing to participate to at least half of the cost, I can purchase one and reverse the protocol.

M-i-c-h-a-e-l1, 2025-01-29:
Dear Pascal,

I have sent you an email.

Kind regards
Michael

pascallanger, 2025-02-05:
I've received the car with gyro. It's a totally different protocol from a first look...

pascallanger, 2025-02-05:
XN297 1M Scrambled

Normal packets
---
RF: 18, 27, 53
Timing: 8205µs
Address (4) = 7D 82 28 C2

P(8)= 7F 80 02 FF 50 00 00 AF
P[0] = THR 68..80..A0
P[1] = ST 50..80..B0
P[2] = TH Rate 0->L, 1->M, 2->H
P[3] = Gyro FF..00 FF=off, 0=max
P[4] = 50 ??
          |04 light off
          |01 trim R (momentary)
          |02 trim L (momentary)
P[5] = 00 ??
P[6] = 00 ??
P[7] = check

pascallanger, 2025-02-06:
Ok, I think the protocol for gyro is now final.
@M-i-c-h-a-e-l1  Please try the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) 1.3.4.30 following the [documentation](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-gyro---1).
@LilTeo14 Please also test this version as I heavily touched the code of none gyro and I can't test it. There should be no change, just to check that I haven't broken it...

M-i-c-h-a-e-l1, 2025-02-06:
I have also noticed that on some occesions there is a strange behaviour when braking. One car has this more often, the other car only had this once. The only thing that can be done is to bind the car again to the transmitter and then usually then the braking behaviour changes. There are no settings available I know of which can change this.

Kind regards
Michael

pascallanger, 2025-02-06:
Bind the car: I've done that at least a hundred times to reverse the protocol and it had no effects...

Have you tried your cars?

M-i-c-h-a-e-l1, 2025-02-06:
I have made some tests and everything works as expected with a RM Zorro:

Car binds immediately to the transmitter.

CH1, steering,  pot gimbal -100 to 100 -> ok
CH2, throttle, pot gimbal -100 to 100 -> ok
CH3, gyroscope sensitivity S2 pot -100 to 100 -> ok
CH4, steering trim not tested
CH5, speed, 3-position switch, -100, 0, 100 ->ok
CH 6, light, 2-position switch -100 and 100-> ok
Ch7, light flash, momentary switch, press switch -100, release switch 100  -> ok (not a documented or possible function with the original tx for the gyor car)

All channels are moving in the right direction no servo reverse programming on the radion needed.

For the original and the RM Zorro this behavouor is the same:

Forward - reverse: Froward - brake lights on and stop, then only reverse and lights are off
Reverse - forward: Reverse - forward immediately, no stop and spinning tyres :-) even with rubber tyres. 

Kind regards
Michael

pascallanger, 2025-02-06:
Great closing this issue then.

closed 2025-02-06 by pascallanger