pascallanger/DIY-Multiprotocol-TX-Module #1067: 1.3.4.0 Surface brance and SLT.

CDRaff, 2025-01-29:
First off I'm sorry if this is not the correct forum for this question; please feel free to delete if it isn't. That being said:

Is SLT not supported in the 1.3.4.0 surface branch, or am I missing it buried somewhere?

pascallanger, 2025-01-29:
Not sure what you mean. The SLT protocol is part of the SFC build at least on the [latest test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) 1.3.4.25 .

CDRaff, 2025-01-29:
Thank you for getting me the file; realize I'm pretty dumb and new at the whole test build stuff so bare with me here, but where would I go to look for test files in the future? I have been looking at the "Releases" tab here and checking the "downloads.multi-module.org" site neither of which is apparently showing me that there are in fact pre-release builds.

closed 2025-01-29 by CDRaff