pascallanger/DIY-Multiprotocol-TX-Module #1068: XK K170 UH-60L Black Hawk

pascallanger, 2025-01-30:
XN297 1Mb Scrambled

Bind
---

Normal packets
---
Address: 54 A6 AB 25 24
RF channels: 60,70,80
Timing: 8085µs
Packet
P(12)= 00 00 AB 7E 80 80 80 00 40 00 AA AA
P[0] = 00?
P[1] = 00
        | 04 Gyro calibration (momentary)
        | 08 Unk long press second top right button (momentary)
        | 10 Unk short press second top right button (toggle)
        | 40 Unk short press second top left button (momentary)
        | 80 Unk long press second top left button (momentary)
P[2] = AB? This value keeps changing when touching any button... Debug?
P[3] = THR 00->2A..7E..AF->B4->BA->CF..DD strange behavior
P[4] = AIL 28..80..DC
P[5] = ELE 22..80..DC
P[6] = RUD DB..80..26
P[7] = Ail trim 20..00..E0
P[8] = 40 Low/ 80 Med/ C0 High
        | 04 Take-off/Landing (momentary)
P[9] = Ele E0..00..20
P[10] = AA ?
P[11] = AA ?

pascallanger, 2025-01-30:
Variation of the Hontai protocol

pascallanger, 2025-01-30:
Protocol implemented

closed 2025-01-30 by pascallanger

yakacm, 2025-07-23:
> That's the heli behavior, nothing I can do.
> It's in the doc "autobind protocol": https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#hontai---26
> Use the bind on channel feature to ease the bind.

It literally takes 2 seconds to hit bind on the radio, probably a lot quicker than the time it took to write this email, lol.

 

Thanks

Kind Regards

Andy Mackay.

Dietrich-L, 2025-07-23:
@Pascal Bind on channel did the trick. I was not aware of this feature, so another lesson learned.
Thanks a lot for the quick help
Dietrich