pascallanger/DIY-Multiprotocol-TX-Module #1069: Park10 branded (wltoys) a160 J3-cub skylark

bad-andy-rc, 2025-01-31:
Recently picked up a Park10 branded version of WLToys' a160 j3-cub rc plane. I would love to bind to it using TX16s 4-in-1. I tried latest stable release v.1.3.4.0, and test release v1.3.4.27 under all protocols known to have worked with prior WLToy's versions. No luck, this board is different. It says P710x on it. 

[image]

[image]

pascallanger, 2025-01-31:
Which type of multi module do you have? External by any chance?

bad-andy-rc, 2025-01-31:
It’s internal. Radiomaster TX16s 4-in-1.

pascallanger, 2025-01-31:
Do you have a 3.3V FTDI/serial converter that you can connect on your internal module? It requires to open the radio but the pins to connect to on the module are accessible.

pascallanger, 2025-01-31:
That's the 2 pins which need to be connected to use the debug mode on your PC: [image]

bad-andy-rc, 2025-01-31:
I’m sorry I do not have one of these. 

I work primarily in Mac OS. (But run windows under parallels) When I plug in the transmitter to USB it offers an option for “USB Serial (VCP)” but I don’t suppose that helps?

bad-andy-rc, 2025-01-31:
> A 3.3V USB serial adapter is only a few euros/dollars and most likely available in a few days wherever you are.
> image.png (view on web) <https://github.com/user-attachments/assets/be442f7c-91f2-4a2a-8852-35ba462e68a0>

Something like this one would work? https://www.amazon.com/gp/product/B0BJKCSZZW/ref=ox_sc_act_title_1?smid=A1YZW40LYQY3L1&psc=1

pascallanger, 2025-01-31:
Yes perfect

bad-andy-rc, 2025-02-04:
Hi Pascal, I’ve received the adaptor - I haven’t done this before so please bear with me - I want to clarify I’ve connected the adaptor as you requested. The pins however are too large to fit the tx16s 4-in-1 holes you’ve ID'd. I can probably work around that just wanted to make sure that’s what you need?

Can I run debug mode as you requested… on a Mac? Or will I need windows?

￼

bad-andy-rc, 2025-02-04:
> You have to switch the voltage to 3.3V on your USB to serial adapter using the strap, right now it's on 5V on the picture.
> The connection looks good. You need to connect both GND together and connect the module TX to the USB serial RX.
> Yes the pins won't fit but you could remove the pin and solder the wire directly on the module. You can then leave this connector in the radio if we ever need it for another model.
> You can use your Mac to do the dumps. What you need a serial terminal to display what the module sends. There are many like Arduino has one built in if you already have it installed or use putty which works on Mac: https://www.puttygen.com/download-putty

I have downloaded STM32CubeProgrammer. Will this be suitable? 

I’ve made the connections. You mentioned "connect the module TX to the USB serial RX.” — Do you mean TX from the 4-in1 to RX as marked on the serial adaptor? Or TX to TX? (The white wire is TX to TX currently, is that correct?)

￼

pascallanger, 2025-02-04:
The picture is wrong...

Module|Serial
---|---
GND|GND
TX|RX

I don't know what the STM32CubeProgrammer can do but I don't think it will do it. Install putty if you don't need Arduino.
If you follow this: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_XN297Ldump.md you should see the module booting when powering on the radio with a model that has your internal module enabled for Multi.

bad-andy-rc, 2025-02-04:
I’ve corrected the connections and I’m using a PC now to make things easier. Have putty running, but not seeing anything. Something about debug mode needs to be uploaded to the adaptor? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_Debug.md

Do I need to follow this? If so what software tool is in the pics?

pascallanger, 2025-02-04:
The software in the pic is Arduino. But you can take any serial software...
For the module you need to install the latest debug firmware ftdi debug which is in this zip file : https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip

bad-andy-rc, 2025-02-04:
Ahh ok, this one? mm-stm-xn297dump-ftdidebug-v1.3.4.28 
Just copy onto the SD and flash like usual?

bad-andy-rc, 2025-02-05:
I’m attempting to flash the multi-module with that firmware and receiving an error…

NEEDS FILE
Stm-opentx-noinv

￼

pascallanger, 2025-02-05:
Please test this file: [multi-stm-ftdi-1.3.4.28.bin.zip](https://github.com/user-attachments/files/18672469/multi-stm-ftdi-1.3.4.28.bin.zip)

bad-andy-rc, 2025-02-05:
> Please test this firmware: multi-stm-ftdi-1.3.4.28.bin.zip <https://github.com/user-attachments/files/18672469/multi-stm-ftdi-1.3.4.28.bin.zip>

Ok thank firmware loaded ok. Thanks. I’ve got putty running but don’t see anything during boot. Do I need to enable something elsewhere?

bad-andy-rc, 2025-02-05:
> Have you configured putty to open the serial port of your USB converter (COM port that appears when you plug it) and set the baudrate to 115200?

Yes, it does connect on COM4, and I’ve tried a few baud rates. But I never see anything on the screen.

pascallanger, 2025-02-05:
Try this version: [multi-stm-ftdi-1.3.4.28a.zip](https://github.com/user-attachments/files/18674677/multi-stm-ftdi-1.3.4.28a.zip)

bad-andy-rc, 2025-02-05:
Nothing with that firmware tone either.

pascallanger, 2025-02-05:
That's strange...
Can you take a picture of your setup?
The module should write some outputs when you select a model which is using it or boot the radio with a model using the module.

bad-andy-rc, 2025-02-05:
Sure, here’s some… I know it’s not clean/soldered (No way I could solder safely at that size) but I do see jitter and characters if I mess with the terminals so I know the wires are connected definitely beyond the radio.
￼￼￼

pascallanger, 2025-02-06:
That's really strange. Let me reproduce it when I'll be home this WE and get back to you.

bad-andy-rc, 2025-02-06:
Absolutely, thank you. FWIW, when I have the serial convertor connected to PC, it shows an always red light. I had it disconnected from PC last night but still attached to radio, working on other planes and was noticing various lights coming on and off. Green ones in the center, other red ones - and they would change when doing different radio tasks. I took that to mean the adaptor might be seeing some traffic from the radio?

pascallanger, 2025-02-06:
I would assume yes since the module sends something each time you change from one protocol to another.
What happens when you open putty and select a model with the LEDs changing? Don't hesitate to change from one model to another using a different protocol.

benlye, 2025-02-07:
@pascallanger I've been working on a board definition for internal serial debugging. I modded one of my radios so I could connect an FTDI adapter to the internal module for testing. I confirm that neither of the builds you posted generates serial debug output.

However, using the new board definition I compiled this binary, which does provide serial output.

@bad-andy-rc, give this binary a go. You should see readable serial output with baud set to 115200. If you don't get anything with this file then you should take a look at your wiring and connections.

[multi-stm-ftdi-1.3.4.30-ben.zip](https://github.com/user-attachments/files/18704940/multi-stm-ftdi-1.3.4.30-ben.zip)

pascallanger, 2025-02-07:
@benlye thanks!!!

bad-andy-rc, 2025-02-07:
@benlye I am getting output. Nice work! Now that this is achieved what output can I provide to help?

pascallanger, 2025-02-07:
Few questions:
 - How do you put the TX in bind mode?
   - Is it in bind each time you power it up?
   - Do you exit bind mode when you go up and down on throttle? Something else?
   - Does the TX needs to have the plane powered on to exit bind?
 - Does the TX need the plane to be on for the led to change its pattern after power on?

For the dumps, it depends a little on the above but you can do this anyway:
 - power on the TX in bind mode
 - create a new model, select the protocol XN297Dump and subprotocol auto
 - You should see on the output the module scanning all frequencies and bitrate. Let it do its thing. When it finds something it will start to analyze. When it finishes to look at the RF part, it will switch to identify features. At this stage, save the content of everything that has appeared and store it in a text file that you post here.
For the second dump, you do the same but in normal mode (exit bind). Just reload the model, it will start scanning again... Save the content and call it normal.txt.
At this stage we should see what bind and normal packets look like. The next step is to identify what the packet content is. When the module states identify sticks and features, move slowly the throttle down->up->down, save that to a throttle.txt file. Repeat the action with aileron, rudder, elevator and any other switches/buttons/special features that the plane has every time in a different file.

bad-andy-rc, 2025-02-07:
> How do you put the TX in bind mode?
> Is it in bind each time you power it up? YES
> Do you exit bind mode when you go up and down on throttle? Something else? Plane and stock remote bind without anything required. Plane flashes red led during bind mode
> Does the TX needs to have the plane powered on to exit bind? YES
> Does the TX need the plane to be on for the led to change its pattern after power on? YES, tx led flashes until plane is on and bind occurs
I performed the instructions through the first dump, attached here.  I will perform the rest this evening.

Thanks!
￼

> Few questions:
>
> How do you put the TX in bind mode?
> Is it in bind each time you power it up?
> Do you exit bind mode when you go up and down on throttle? Something else?
> Does the TX needs to have the [...]

Multiprotocol version: 1.3.4.30
Module Flash size: 128KB
EEPROM initialized: 0
Protocol selection switch reads as 0
Using bank 0
Module Id: d33ea96a
Init complete
Normal telem 0,1
Protocol selected: 1, sub proto 0, rxnum 0, option 0
Protocol selected: 1, sub proto 0, rxnum 0, option 0
Protocol selected: 63, sub proto 0, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=7, Sub=250Kbps, Opt=8, FS=0, CHMap=0, rfSw=2
Channel=0,0x00
Protocol selected: 63, sub proto 3, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=7, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=2M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69

Packet detected: bitrate=Using NRF
250K C=69 S=Y A= CC CC CC CC CC P(9)= 9C BB CC DD 84 24 20 00 97

--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69
RX on channel: 69, Time:     0us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 49448us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 52415us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 52378us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 52419us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 52387us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 52393us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 52414us P: 9C BB CC DD 84 24 20 00 97
RX on channel: 69, Time: 52394us P: 9C BB CC DD 84 24 20 00 97
Trying RF channel: ,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84

1 RF channels identified: 69[9]
Keeping only RF channels with more than 6 packets: 69*
--------------------------------
Identifying RF channels order.
Time between CH:69 and CH:69
Time: 52390us
Time: 49036us
Time: 55320us
Time: 52369us

Channel order:
69:     0us
69: 49036us

--------------------------------
Identifying Sticks and features.
P: 9C BB CC DD 84 24 20 00 97

pascallanger, 2025-02-07:
It's a variant of the XK2 protocol.
The dump you've sent is what the plane sends to the TX.
The plane is currently waiting for an answer from the TX... So like if your TX is not turned on or not in bind mode or...

Extract from my XK2 protocol notes:
> Bind
> ----
> Plane sends these packets:
> RX:     0us C=71 S=Y A= CC CC CC CC CC P(9)= 9C BB CC DD 38 12 10 00 19
> P[0] = 9C bind phase 1
> P[1] = Dummy TX_ID
> P[2] = Dummy TX_ID
> P[3] = Dummy TX_ID
> P[4] = RX_ID[0]
> P[5] = RX_ID[1]
> P[6] = RX_ID[2]
> P[7] = 00
> P[8] = sum(P[0..7]) + BF

The bind RF frequency is different since yours is 69 instead of 71
In your case the plane ID is 84 24 20
The checksum calculation is also different P[8] = sum(P[0..7]) + CF. They've just added 10 so may be it's +10 everywhere?

Normally the TX answers with its ID (replacing the dummy bytes) and the plane acks before switching to normal packets.
Since they've changed the bind RF frequency and the checksum, I'm wondering how much they've changed the protocol... Without all the dumps it's hard to answer.

As you can see you must be extra clear on which state the TX is, which state the plane is and so on when doing the dumps. Usually the LED on the TX and plane changes to indicate the different phases so use this to report.

bad-andy-rc, 2025-02-07:
Here’s what I’m getting now. On the TX16s I am loading another model (FMS) then loading this model. I’ve attached the output…

￼

> It's a variant of the XK2 protocol.
> The dump you've sent is what the plane sends to the TX.
> The plane is currently waiting for an answer from the TX... So like if your TX is not turned on or not in [...]

Bad frame RX
Multiprotocol version: 1.3.4.30
Module Flash size: 128KB
EEPROM initialized: 0
Protocol selection switch reads as 0
Using bank 0
Module Id: d33ea96a
Init complete
Normal telem 0,1
Protocol selected: 28, sub proto 0, rxnum 4, option 128
Bad frame RX
Multiprotocol version: 1.3.4.30
Module Flash size: 128KB
EEPROM initialized: 0
Protocol selection switch reads as 0
Using bank 0
Module Id: d33ea96a
Init complete
Normal telem 0,1
Protocol selected: 14, sub proto 0, rxnum 0, option 0
Bad frame RX
Multiprotocol version: 1.3.4.30
Module Flash size: 128KB
EEPROM initialized: 0
Protocol selection switch reads as 0
Using bank 0
Module Id: d33ea96a
Init complete
Normal telem 0,1
Protocol selected: 63, sub proto 3, rxnum 5, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=7, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=2M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=2M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=2M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=2M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32

pascallanger, 2025-02-07:
If you have powered on only the TX, yes you get nothing... This is what I've explained you... The TX doesn't send anything until it "hears" from the plane. The plane is talking first.
Power on both the TX and the plane. Make sure you have control from the original TX and send the requested dumps.
For the other bind phases we will see later...

bad-andy-rc, 2025-02-08:
Ok understood. So this time I powered up the tx16s, the plane, then the original controller. This is the output I received. I tried to be systematic in using ailerons, then elevator, then rudder but not sure how/if that registered here. This plane activates the gyro only after hitting throttle. I included outputs both with gyro activated and prior. Don’t know if that helps. Let me know what else you’d like me to try. Thanks!

￼￼

> If you have powered on only the TX, yes you get nothing... This is what I've explained you... The TX doesn't send anything until it "hears" from the plane. The plane is talking first.
> Power on both [...]

Multiprotocol version: 1.3.4.30
Module Flash size: 128KB
EEPROM initialized: 0
Protocol selection switch reads as 0
Using bank 0
Module Id: d33ea96a
Init complete
Normal telem 0,1
Protocol selected: 0, sub proto 0, rxnum 0, option 255
Protocol selected: 63, sub proto 3, rxnum 5, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=7, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64

Packet detected: bitrate=Using NRF
250K C=64 S=Y A= E8 25 3B CC CC P(9)= E8 25 3B 00 00 00 00 00 24

--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66
RX on channel: 66, Time:     0us P: 32 32 00 32 A0 40 00 C8 6D
RX on channel: 66, Time:  3856us P: 32 32 00 32 A0 40 01 C8 6E
RX on channel: 66, Time: 29699us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 66, Time:  6005us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 66, Time: 21088us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 66, Time: 32463us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 66, Time: 32581us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 66, Time: 24373us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 66, Time: 69453us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 66, Time:  4041us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 66, Time: 28236us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 66, Time:  6240us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 66, Time: 14990us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 66, Time: 41151us P: 32 32 00 32 C1 48 01 C8 97
Trying RF channel: ,67
RX on channel: 67, Time:     0us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  3392us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  3356us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time: 10929us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 67, Time:  2529us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  6726us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  5436us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 67, Time:  3667us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 67, Time:  7723us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  2854us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 67, Time:  5876us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 67, Time: 15298us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 67, Time:  5656us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 67, Time:  3948us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  3375us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  6694us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time: 16443us P: 32 32 00 32 C1 48 00 C8 96
RX on channel: 67, Time:  4023us P: 32 32 00 32 C1 48 01 C8 97
RX on channel: 67, Time:  3125us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  6715us P: E8 25 3B 00 00 00 00 00 24
RX on channel: 67, Time:  5182us P: 32 32 00 32 C1 48 00 C8 96
Trying RF channel: ,68,69,70
RX on channel: 70, Time:     0us P: 32 32 00 32 A0 48 01 C8 76
RX on channel: 70, Time: 11011us P: 32 32 00 32 E0 48 00 C8 B5
RX on channel: 70, Time: 23462us P: 32 32 00 32 E0 48 01 C8 B6
[... 1956 lines cut ...]
P: 32 32 00 32 A0 40 01 C8 6E
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 E0 40 01 C8 AE
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 20 40 00 C8 ED
P: 32 32 00 32 20 40 01 C8 EE
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 20 40 01 C8 EE
P: 32 32 00 32 60 40 00 C8 2D
P: 32 32 00 32 60 40 01 C8 2E
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 A0 40 00 C8 6D
P: 32 32 00 32 A0 40 01 C8 6E
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 E0 40 00 C8 AD
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 E0 40 01 C8 AE
P: 32 32 00 32 E0 40 00 C8 AD
P: 32 32 00 32 E0 40 01 C8 AE
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 20 40 00 C8 ED
P: 32 32 00 32 20 40 01 C8 EE
P: E8 25 3B 00 00 00 00 00 24
P: 32 32 00 32 60 40 00 C8 2D
P:

pascallanger, 2025-02-08:
Sorry but you are not following the instructions neither answering the questions... What you are providing so far is not useful and does not allow me to work on it.
So let's try again. You MUST follow each line at the letter and provide either a text file or an answer.

Here is the process you need to follow for the dumps below:
1. Power on the plane
2. Power on the original TX
3. Make sure that you can control the plane
4. Turn off the plane
5. At this stage the original TX is still powered on and transmitting -> do not power off the original TX

Once the above is done:
1. Relaunch a dump (by switching model or sub_protocol)
2. Wait for the text "Identifying Sticks and features" to appear
3. Move extremely slowly Aileron from mid->right->mid->left->mid
4. Copy the output in a text file and save it with the name Aileron.txt
5. Repeat step 1 to 4 replacing Aileron by Throttle, Elevator, Rudder
6. Repeat step 1 to 4 replacing Aileron by 4 slow presses of each button on the original TX, example press slowly 4 times the 3D/6G button and save the content in the file 3D6G.txt
7. Turn off the TX
8. Post all the text files here

bad-andy-rc, 2025-02-08:
I am trying my best to perform the steps as you describe. The expected outcome you describe, is not what is actually happening, so it is difficult to attain the data you are asking for. I answered these questions before, I don’t know what other questions you are referring to:

> How do you put the TX in bind mode?
> Is it in bind each time you power it up? YES
> Do you exit bind mode when you go up and down on throttle? Something else? Plane and stock remote bind without anything required. Plane flashes red led during bind mode
> Does the TX needs to have the plane powered on to exit bind? YES
> Does the TX need the plane to be on for the led to change its pattern after power on? YES, tx led flashes until plane is on and bind occurs
> Here is the process you need to follow for the dumps below:
> 
> Power on the plane
> Power on the original TX
> Make sure that you can control the plane
> Turn off the plane
> At this stage the original TX is still powered on and transmitting -> do not power off the original TX (TX starts to beep when plane turns off)
> Once the above is done:
> 
> Relaunch a dump (by switching model or sub_protocol) This is what output looks like. Often the output turns into endless scrolling of those bytes. If I turn off the plane, the scrolling stops, the TX beeps and nothing happens in putty. Occasionally this process stops at “Identifying sticks and features” but no data appears when moving the sticks. I’ve attached both outputs but this is all I seem to be able to get following these steps. 
￼￼

pascallanger, 2025-02-09:
I have a hard time to believe that when the plane is bound and surfaces move when you touch the TX, that if you shutdown the plane the TX stops sending anything on any RF channels.
This would be catastrophic since it would mean that if the plane goes too far or something blocking the signal and the TX doesn't receive the plane anymore it will stop sending anything, which means that you just lose the plane even on a temporary event...
Are you sure that you are really following:
1. Power on the plane
2. Power on the original TX
3. Make sure that you can control the plane
4. Turn off the plane and do not power off the original TX
6. Launch a dump using the sub protocol Auto
7. Launch a dump using the sub protocol 1M and RF frequency=-1

pascallanger, 2025-02-09:
I've tried to code the protocol but there are too many unknowns to hope that it will work right away and even if it does don't fly with it...
That said try the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) 1.3.4.31 protocol XK2/P10.

bad-andy-rc, 2025-02-09:
Maybe this will be easier if you can witness what is happening.  I will upload video of the steps being performed exactly. Please let me know if you want to try anything differently. I will try your new f/w shortly.

pascallanger, 2025-02-09:
Ok for the features, you need to do some dumps:

1. Power on the plane
2. Power on the original TX
3. Make sure that you can control the plane
4. Turn off the plane and do not power off the original TX
5. Launch a dump using XN297DP sub protocol Auto
6. Wait for the text "Identifying Sticks and features" to appear
7. Press one button 4 times on the original TX with a PAUSE in between each press.
8. Save the content in a file with the name of the feature.
9. Repeat steps 5 to 8 for all the buttons/features (not the trims but things like 3D/6G, rate, hover, light, ...)
10. Post all the files

I can't even see on your manual how to switch the lights on the plane despite the fact that there is a light icon on the LCD. You might need to long press some buttons to find this "hidden" feature.

bad-andy-rc, 2025-02-10:
> Ok for the features, you need to do some dumps:
>
> Power on the plane
> Power on the original TX
> Make sure that you can control the plane
> Turn off the plane and do not power off the original TX
> Launch a dump using XN297DP sub protocol Auto
> Wait for the text "Identifying Sticks and features" to appear
> Press one button 4 times on the original TX with a PAUSE in between each press.
> Save the content in a file with the name of the feature.
> Repeat steps 5 to 8 for all the buttons/features (not the trims but things like 3D/6G, rate, hover, light, ...)
> Post all the files
> I can't even see on your manual how to switch the lights on the plane despite the fact that there is a light icon on the LCD. You might need to long press some buttons to find this "hidden" feature.

This isn’t working. No matter what I try I get endless packet scrolling. The videos in my previous reply show this very clearly. Only occasionally has it scanned and stopped at ID’ing sticks…. but then no data appears when moving the sticks. Believe me, I am doing everything you say to the letter and this is the result. It is very frustrating because I am spending a lot of time not able to produce the result you are asking for. I am performing your steps exactly. I am trying other approaches too, everything I can think of…  Perhaps the f/w needs a tweak? I am using ben’s f/w since that was the only one that output anything.

I don’t know about the lights, there are extra ports on a satellite pcb for them I think. I will try to find some lights to investigate that function.

bad-andy-rc, 2025-02-10:
> I'm struggling to understand the issue. You've already provided the same kind of outputs I need in this post #1069 (comment) <https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1069#issuecomment-2644416670> as an example. I'm not asking more than that.
> What I really need is on all the lines that you see scrolling like this:
> First, the lines starting with P: E8 25... can be ignored. When you press a button you should see some values moving in the P: 32 lines. This is what I really need: when I press the button X this value changes momentary or permanently from Y to Z when I press it again it goes from Z to Y.

I tried a few times today. One time the packets were not scrolling as fast so I tried to get the data. I don’t know if this is valid or the slower scrolling just indicated an issue, but maybe the files have something of worth. I don’t know. I’ll try more later but with the endless super fast scrolling this is quite challenging. 

￼

pascallanger, 2025-02-11:
I don't understand why the fast scrolling is challenging... I was not asking you to look at the data on the output specifically, let it scroll quickly press the button four times and save the full putty output then restart putty to clear the previous data and repeat...

Anyway, I've looked at your files and all the flags look to be correct in the test version you tried out.
The channels are setup like this: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-p10---1
You need to test and really look at what the plane is doing after you power on the motor at least once to enable the gyro and all the compensations. For example going from 6G to 3D can be identified since 6G will move the elevator to put the fuselage level where in 3D the elevator will stay flat and only move to counteract sudden movements.

bad-andy-rc, 2025-02-11:
￼

Ok I did more dumps as you requested. The plane boots into 6g mode. Every test was performed after I hit the throttle once the plane was started/bound. This is required to activate the gyro/compensations.

Thanks

pascallanger, 2025-02-12:
The protocol is final from my perspective and has said yesterday I haven't touched anything since the [test version](https://nightly.link/pascallanger/DIY-Multiprotocol-TX-Module/workflows/main/master/multi-test-build.zip) from 3 days ago as everything I see in the dumps are already in there.
Channels setup: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#xk2---99

pascallanger, 2025-02-12:
For bind, the protocol doc and the X4 sub protocol : "The plane does not need to be bound each time if it is powered on **after** the radio/protocol is on.". Since both protocols are similar I would expect that it works the same. Basically if the plane is bound and controlled, turn it off and turn it back on, it should reconnect immediately. If not then I can modify the doc.

pascallanger, 2025-02-18:
No feedback so I consider this protocol done.

closed 2025-02-18 by pascallanger

bad-andy-rc, 2025-02-19:
> For bind, the protocol doc and the X4 sub protocol : "The plane does not need to be bound each time if it is powered on **after** the radio/protocol is on.". Since both protocols are similar I would [...]

Yes If the TX has been bound to the plane, as long as it remains powered on it will rebind if plane is powered off/on. However once the TX is powered off, you'd need to rebind manually. 

I was able to test the CH8 - Lights function and it indeed works. +100 turns on 5v at those additional ports on the remote pcb. So that's cool, adding lights to this model would be fairly easy.

pascallanger, 2025-02-19:
> Yes If the TX has been bound to the plane, as long as it remains powered on it will rebind if plane is powered off/on. However once the TX is powered off, you'd need to rebind manually.

Let me recap:
If the TX is on, you can turn off the plane then on and the plane will reconnect, correct?
If the TX is off then turned on (waiting a few sec for the model to load and the multi module to start) and only then you turn on the plane, it doesn't reconnect?

pascallanger, 2025-02-19:
It doesn't make sense... The plane has no way to know that the TX has been turned off. This is why if you turn on the TX, make sure the model is loaded, the module has started and it is not in bind mode then and only then power on the plane, it should just reconnect.

bad-andy-rc, 2025-02-19:
Yeah that doesn’t happen. I have to rebind manually every time I power on the TX.

pascallanger, 2025-02-19:
Have you checked "Bind on channel" by any chance?

bad-andy-rc, 2025-02-19:
> If it's checked that's why the plane doesn't reconnect automatically... The module in this case is not doing anything and waiting that you either bind manually or that you flip a switch on CH16.
> But in the case of that plane it's not needed... Just power on the radio before the plane (as it should) and the plane will just connect fine.

"Just power on the radio before the plane (as it should) and the plane will just connect fine.” WRONG. ….Please stop saying this because it is WRONG. 

I have tested many, many, many times. Please read what I wrote about the behavior - which is 100% accurate as to what is happening.

pascallanger, 2025-02-19:
It doesn't work because you have checked "Bind on channel" which should not be checked for this model...
So sorry but you are WRONG here.

bad-andy-rc, 2025-02-19:
"I leave bind on channel unchecked.” Was what I last wrote on that message. It is NOT CHECKED. I’m not trying to argue but you are either not believing me, or reading my messages correctly.