pascallanger/DIY-Multiprotocol-TX-Module #1070: irx4 lite module telemetry

Danmcmahon1, 2025-02-02:
I have a FRsky X20s that has had all the latest updates and Lua scripts installed
When I bind to my Draco, I am seeing 2 issues
First the temp in Fahrenheit is about 30 degrees low
And the rpms are reading about 6-10 times faster then what it appears to be

I tried to attach a video here but it will not let me

pascallanger, 2025-02-03:
You are not even telling which protocol/subprotocol you are using...

Danmcmahon1, 2025-02-04:
under the info tab in my radio - the external module is called multimodule 1.3.4.0
under the RF system tab the protocol is set to spectrum DSMX-2F
on the widget screen I have the units set to Fahrenheit  - do I need to change this somewhere else?
on the widget screen for the ESC RPM i have the following settings which were factory default
range 0r/m to 250000r/m
Mode is set to prop
average is set to 1 sample
blades is set to 1 - should it be 4?

Thanks for your help

closed 2025-02-04 by pascallanger