pascallanger/DIY-Multiprotocol-TX-Module #1076: Flysky 2A-BS protocol

yinglong, 2025-02-20:
Flysky has a protocol named ‘2A-BS’, which comes with a range of receivers (https://www.flysky-cn.com/receivers), I would like to know if this protocol can be supported or not.

pascallanger, 2025-02-21:
Is it the same as this one: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/930

yinglong, 2025-02-21:
> Is it the same as this one: [#930](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/930)

Not sure. According to flysky website, there's a separate product line featuring the ANT protocol, which shares no product in common with the 2A-BS line. However, something weird I found is that my FMS fcx18 model came with a flysky MG41 transmitter, which runs ANT protocol (https://www.flysky-cn.com/mg41specifications). But inside the vehicle I found the receiver has a 2A-BS label on it(https://www.rcgroups.com/forums/showatt.php?attachmentid=17930001&d=1704654517)

CGPro856, 2025-08-20:
[image]

pascallanger, 2025-08-21:
That looks to be an all in one RF chip. Nothing to put an analyzer on... Could they even manufacture their own chip since it starts with FS like FlySky?
Might want to try XN297dump or SDR.