pascallanger/DIY-Multiprotocol-TX-Module #1077: Unable to update the lastest firmware to MT12

Cvajra, 2025-02-21:
hey there. I have tried to update the latest firmware to my Radiomaster mt12 4in1 in EdgeTX Buddy but it keeps showing this:

FYI = I had turned off my mt12 and plugged USB cable from it to my laptob(windows 11).

[image]
[image]

let me know if you have any solution!!!

pascallanger, 2025-02-21:
You need to update your driver using Zadig.

pascallanger, 2025-02-21:
You want to flash the radio right.
The multi module update is done through the sd card.

pascallanger, 2025-02-21:
That's not really related to the multi module but the radio itself so I'll close this issue.

Cvajra, 2025-02-21:
I had tried via Zadig, but in Edge tx buddy it still shows the same thing.

pascallanger, 2025-02-21:
My guess is that you haven't selected the correct driver or you haven't applied it. Anyway this is the wrong place to post as this is an EdgeTX issue not Multi module.